zephyrproject-rtos/zephyr #26519: samples: net: sockets: dumb_http_server: instabllity on nucleo_f767zi

ABOSTM, 2020-06-29:
**Describe the bug**
Running samples/net/sockets/dumb_http_server on nucleo_f767zi board provides unstable result.
Sometimes web page is properly dispalyed on client browser, sometimes not (page unreachable).
Reset is required to switch from working to not working status, ... or reverse.
Console log looks normal, just missing connection information about connection which is printed when successful
```
Connection #0 from 192.168.0.27
Connection from 192.168.0.27 closed
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f767zi samples/net/sockets/dumb_http_server
2. west flash
3. open the web page from any client browser
4. sometimes, web page is unreachable

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.3.0-637-gce7dec138a6a  ***
Single-threaded dumb HTTP server waits for a connection on port 8080...
[00:00:02.774,000] <inf> net_config: Initializing network
[00:00:02.774,000] <inf> net_config: Waiting interface 0x200209e0 to be up...
[00:00:03.264,000] <inf> net_config: Interface 0x200209e0 coming up
[00:00:03.264,000] <inf> net_config: IPv4 address: 192.168.0.14
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain : gnuarmemb
 - Commit SHA : f4896adaa86cef3bfb4c2580e1d880f0718da01a

pfalcon, 2020-06-30:
Is this reproducible with BOARD=qemu_x86? If not, that's nucleo_f767zi specific issue, I don't have that board, so wouldn't be able to look into that. Unassigning from myself.

pfalcon, 2020-06-30:
@ABOSTM: Besides testing other boards, please also test other samples, to confirm the title of this ticket. Next in queue would be big_http_download.

ABOSTM, 2020-07-01:
@pfalcon,
I tested nucleo_f429zi and don't face the instability.
I also tested sample big_http_download on nucleo_f767zi and I face also some instability (sometimes, unable to get IP address from DHCP server).
So it looks like issue is specific to nucleo_f767zi, and common to several samples.
I will continue investigations.

ABOSTM, 2020-07-07:
I figure out that I used a nucleo_f767zi with soc cut-A (Device marking A) which has some ethernet restrictions:
https://www.st.com/content/ccc/resource/technical/document/errata_sheet/group0/23/a6/11/0b/30/24/46/a5/DM00257543/files/DM00257543.pdf/jcr:content/translations/en.DM00257543.pdf

Using a more recent board, with cut-Z, the instability vanished.

closed 2020-07-07 by ABOSTM

pfalcon, 2020-07-07:
@ABOSTM, Very interesting, thanks for recording this situation and related links, may be useful to other folks facing a similar situation.

Nukersson, 2020-07-07:
> I figure out that I used a nucleo_f767zi with soc cut-A (Device marking A) which has some ethernet restrictions: [...]

Maybe adding short warning in `nucleo_f767zi` board doc would be not bad?

PR #26726 (2020-07-10): boards: arm: nucleo_f767zi: ethernet restriction on cut-A
nucleo_f767zi with soc cut-A (Device marking A) has some ethernet restrictions.
Use of cut-Z is advised.

fixes #26519
commit e09fcca8 (2020-07-10): boards: arm: nucleo_f767zi: ethernet restriction on cut-A

nucleo_f767zi with soc cut-A (Device marking A) has some ethernet
restrictions.
Use of cut-Z is adviced.

fixes #26519