zephyrproject-rtos/zephyr #26524: Problem with hci_uart and L2CAP CoC connections

jjamesson, 2020-06-29:
Hi, Zephyr community!

I am experiencing connection issues between an Android 10 device and the hci_uart example running on an nrf PCA10040 evaluation kit. The problem arises when I use L2CAP channel connection from the android app side, as Bluez states that the PSM value used to create the connection is incorrect. 

**Btmon log:**
\> ACL Data RX: Handle 0 flags 0x02 dlen 18
> LE L2CAP: LE Connection Request (0x14) ident 2 len 10
>> PSM: 1 (0x0001)
> Source CID: 83
> MTU: 65535
> MPS: 1021
> Credits: 65535

< ACL Data TX: Handle 0 flags 0x00 dlen 18
> LE L2CAP: LE Connection Response (0x15) ident 2 len 10
>> Destination CID: 0
> MTU: 0
> MPS: 0
> Credits: 0
> Result: Connection refused - PSM not supported (0x0002)

I have tried several PSM values to solve this, but without luck. I'm starting to wonder if Zephyr supports L2CAP CoC connections, or is my Zephyr configuration wrong, or is this a Bluez issue?

With regards,
James

jhedberg, 2020-07-01:
`hci_uart` only provides an HCI transport between a host (e.g. BlueZ) and the Zephyr controller. It's completely agnostic to what traffic flows over this transport on higher layers like L2CAP (i.e. hci_uart knows absolutely nothing about L2CAP). So I don't think the issue can be with the hci_uart application itself, rather most likely something in BlueZ or Android.

jjamesson, 2020-07-01:
Thanks @jhedberg,
I saw no documentation on Bluez about assigning my own dynamic PSMs, so I will try rebuilding my own driver in an embedded device and let you know if this can be bypassed or not.

Will keep you updated.

WR,

jjamesson, 2020-07-07:
Hi, Zephyr community and @jhedberg ,

After digging around for information about Bluez, I was unable to assign dynamic PSMs myself, so I wrote a custom HCI driver on an embedded system. Thus I was able to generate custom PSMs as I liked and I got it to work, so this is likely a Bluez limitation.

Thanks for thinking along. This issue can now be closed.

WR,

jhedberg, 2020-07-07:
@jjamesson IIRC BlueZ gives you dynamic PSMs when you set the PSM to 0 in the struct sockaddr_l2

Anyway, I'll close this then since it's not a Zephyr issue.

closed 2020-07-07 by jhedberg