zephyrproject-rtos/zephyr #26527: Couldn't flash image into mimxrt1050_evk board using west flash command.

chen-png, 2020-06-30:
**Describe the bug**
when running west flash command to flash test image into mimxrt1050_evk board, it  failed with "_TargetHalt: CPU halted" error.

but when i used another new mimxrt1050_evk board, it was ok to flash. maybe this a hardware issue, but how to reset this board.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mimxrt1050_evk -p auto samples/hello_world/
2. west flash
3. See error

**Logs and console output**
SEGGER J-Link Commander V6.50b (Compiled Sep  6 2019 17:46:52)
DLL version V6.50b, compiled Sep  6 2019 17:46:40

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link OpenSDA 2 compiled May 27 2019 11:01:03
Hardware version: V1.00
S/N: 621000000
VTref=3.300V
Target connection not established yet but required for command.
Device "MCIMXRT1052" selected.

**Connecting to target via SWD
InitTarget() start
InitTarget()
_TargetHalt: CPU halted
InitTarget() end
InitTarget() start
InitTarget()**
_TargetHalt: CPU halted
InitTarget() end
InitTarget() start
InitTarget()
_TargetHalt: CPU halted
InitTarget() end
InitTarget() start
InitTarget()
_TargetHalt: CPU halted
InitTarget() end
Cannot connect to target.

Target connection not established yet but required for command.
Device "MCIMXRT1052" selected.

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.3
 - Commit ID: 9c65a17743a3

MaureenHelm, 2020-07-28:
This can happen if the boot header gets corrupted. Can you try adjusting the SW7 switches to prevent booting from Hyperflash, reset the board, then try programming it again? If that succeeds, you'll then need to adjust the SW7 switches back to their Hyperflash configuration and reset again. See the troubleshooting section [here](https://mcuoneclipse.com/2017/12/16/mcuxpresso-ide-v10-1-0-with-i-mx-rt1052-crossover-processor/) for more details.

chen-png, 2020-07-30:
i will have a try

chen-png, 2020-08-04:
according to the guideline, adjusted SW7 to 1-ON 2-ON 3-ON 4-OFF, but it still reported the same error

dleach02, 2020-08-12:
@chen-png, when you connect the board via the usb, does the green LED next to the 5v power connector and switch come on?

chen-png, 2020-08-14:
it has broken three mimxrt1050_evk boards totally, so changed its priority to high.

dleach02, 2020-08-14:
@chen-png, can you answer the question on whether the green LED next to the 5v power connector switch comes on?

dleach02, 2020-08-14:
@chen-png, in your environment, are you connecting and disconnecting these boards from the test environment fairly regularly? I'm also asking @hakehuang to get in touch with you so we can figure out what is going on.

hakehuang, 2020-08-15:
@chen-png  can you ship your broken board to me?

chen-png, 2020-08-17:
> @chen-png, can you answer the question on whether the green LED next to the 5v power connector switch comes on?

no. i tested it with a good board, the green led near to the 5v power connector should be lighted, but for that broken board, it didn't.

chen-png, 2020-08-17:
> can you ship your broken board to me?

yes, i replied it in your email, thanks.

dleach02, 2020-08-19:
This does not look like a bug with Zephyr and instead is a hardware issue with the boards that occurred during usage. The affected boards have been sent to NXP to diagnose/debug.

I recommend closing this issue and updating it later with what ever information NXP is able to get from the boards.

closed 2020-08-24 by dleach02