zephyrproject-rtos/zephyr #26529: How to support Nordic ble5.0 on Android 7.0？

zhaoyangyang12, 2020-06-30:
On Android 7.0 system, is there any Nordic interface firmware that can scan, pair, connect, send and receive data and files?About docking Nordic ble 5.0 on Android system, you can share some ideas

cvinayak, 2020-07-01:
Do you need a connectivity solution to an Android based system? and/or do you need an Android application?
Could you please elaborate on your requirement?

> About docking Nordic ble 5.0 on Android system

Docking?

zhaoyangyang12, 2020-07-03:
We want to support Nordic 52840 on Android system through the relevant software provided by you. For example, Bluetooth on mobile phones, we want to replace Bluetooth on mobile phones with 52840. The hardware Android UART / USB is connected with Nordic UART / USB. Through the standard bluedroid protocol stack on Android, Nordic 52840 can be called. What kind of interface firmware should Nordic 52840 provide? What we're going to do is similar to a Bluetooth adapter.

zhaoyangyang12, 2020-07-03:
> Do you need a connectivity solution to an Android based system? and/or do you need an Android application?
> Could you please elaborate on your requirement?
> 
> > About docking Nordic ble 5.0 on Android system
> 
> Docking?

We want to support Nordic 52840 on Android system through the relevant software provided by you. For example, Bluetooth on mobile phones, we want to replace Bluetooth on mobile phones with 52840. The hardware Android UART / USB is connected with Nordic UART / USB. Through the standard bluedroid protocol stack on Android, Nordic 52840 can be called. What kind of interface firmware should Nordic 52840 provide? What we're going to do is similar to a Bluetooth adapter.

cvinayak, 2020-07-08:
nRF52840 is Bluetooth Low Energy single mode only. I will assume you will need to port/fix  android Bluetooth subsystem to use Bluetooth standard complaint HCI commands for LE controller.

carlescufi, 2020-07-08:
@zhaoyangyang12 I am really not sure the Android Bluetooth stack supports single-mode BLE adapters, but you can try by building the `samples/bluetooth/hci_usb` sample, that will give you what you need.

closed 2020-07-08 by carlescufi