zephyrproject-rtos/zephyr #26533: Support newlib for Aarch64 architecture

vladimir-podbrezsky, 2020-06-30:
Bug description
The function "z_bss_zero" does not clear BSS section when using newlib for Aarch64 architecture.
Actually the function "memset" invoked from z_bss_zero clears only first 128 bytes of BSS section and returns. The size of BSS section is 0x72BC according to map file
                0x000000000040C000                __bss_start = .
                0x00000000004132bc                __bss_end = ALIGN (0x4)

At the same image it works perfectly when using minlib. 

To Reproduce
Steps to reproduce the behavior:

1. Use flag "CONFIG_NEWLIB_LIBC=y" in prj.conf file in samples/hello_world    
2. Build image for board "bcm958402m2_a72"
3. Set break point to function "z_bss_zero"

Expected behavior
It is expected that whole BSS section is zeroed 

Environment 

1. Build of the image runs on Ubuntu 18.04.3
2. Toolchain: Zephyr SDK  0.11.3
3. Zephyr version 2.3.0 rc1

stephanosio, 2020-07-01:
cc @Abhishek-brcm @sandeepbrcm

stephanosio, 2020-07-01:
> Actually the function "memset" invoked from z_bss_zero clears only first 128 bytes of BSS section and returns.

I tested this on `qemu_cortex_a53` (because I do not have `bcm958402m2_a72` hardware) and was unable to reproduce this.

I intentionally set `__bss_start:__bss_end` memory region to garbage values in `z_arm64_prep_c` before calling `z_bss_zero` because everything is already initialised to 0 in QEMU; I could confirm that the newlib `memset` function successfully initialises the entire BSS region defined by `__bss_start:__bss_end` to 0.

https://asciinema.org/a/x48Ca0pFQd3oAks34uxZMSqB7

Abhishek-brcm, 2020-07-08:
I am able to reproduce the mentioned newlib issue.
@stephanosio can you point to memset implementation used in zephyr SDK aarch64 newlib?

andrewboie, 2020-07-09:
> I intentionally set `__bss_start:__bss_end` memory region to garbage values in `z_arm64_prep_c` before calling `z_bss_zero` because everything is already initialised to 0 in QEMU

You cannot rely on this on real hardware, I strongly suggest you don't do this even if it might be redundant in a certain emulator.

Abhishek-brcm, 2020-07-09:
I was able to figure out this issue to some extent.
I see that [dc zva, addr](https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libc/machine/aarch64/memset.S;h=103e3f8bb0f20a5d02578f2379620687eae10a52;hb=HEAD#l177) (data cache zero by address range) instruction is casing data abort.  I think this could possibly be because D-cache and MMU are not enabled yet.

sandeepbrcm, 2020-07-10:
> I was able to figure out this issue to some extent.
> I see that [dc zva, addr](https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libc/machine/aarch64/memset.S;h=103e3f8bb0f20a5d02578f2379620687eae10a52;hb=HEAD#l177) (data cache zero by address range) instruction is causing data abort. I think this could possibly be because D-cache and MMU are not enabled yet.

yes you need `MMU` on with valid map before using `DC ZVA`

closed 2020-11-04 by carlocaione