zephyrproject-rtos/zephyr #26536: The CONFIG_BT_L2CAP_RX_MTU setting are not reflected correctly in the build

andung07, 2020-06-30:
**Describe the bug**
The `CONFIG_BT_L2CAP_TX_MTU` and `CONFIG_BT_L2CAP_RX_MTU` setting are not reflected correctly in the build. The resulting negotiated mtu capped to 68 bytes.

**To Reproduce**
Steps to reproduce the behavior:
1. Set `CONFIG_BT_L2CAP_TX_MTU` and `CONFIG_BT_L2CAP_RX_MTU` to 255
2. Create a ble service with read and notify characteristic
3. Create notification with payload at least 70 bytes long
4. Open NRF connect android app, go to setting and select `Connectivity` and enable `Auto request maximum MTU`. This will negotiate MTU to 517 bytes.
5. Scan, and connect to the ble device then in the characteristic created in 2, enable notification on that service.

**Expected behavior**
The negotiated MTU should be 255 `min(server_mtu, client_mtu)`, `min(255, 517)` and the notification is triggered successfully. 

**Impact**
*showstopper*

**Logs and console output**
```
[00:00:09.949,859] <dbg> bt_att.find_info_cb: handle 0x0017
[00:00:09.964,996] <dbg> bt_att.att_rsp_sent: conn 0x200019e4 att 0x200026f0
[00:00:09.965,026] <dbg> bt_att.att_pdu_sent: conn 0x200019e4 att 0x200026f0
[00:00:09.972,167] <dbg> bt_att.bt_att_recv: Received ATT code 0x02 len 2
[00:00:09.972,198] <dbg> bt_att.att_mtu_req: Client MTU 517
[00:00:09.972,229] <dbg> bt_att.att_mtu_req: Server MTU 68
[00:00:09.972,229] <dbg> bt_att.att_mtu_req: Negotiated MTU 68
[00:00:09.987,121] <dbg> bt_att.att_rsp_sent: conn 0x200019e4 att 0x200026f0
[00:00:09.987,152] <dbg> bt_att.att_pdu_sent: conn 0x200019e4 att 0x200026f0
[00:01:04.166,870] <wrn> bt_att: ATT MTU exceeded, max 68, wanted 77
[00:01:04.166,870] <wrn> bt_gatt: No buffer available to send notification
```

**Environment (please complete the following information):**
 - OS: Mac
 - PlatformIo
 - 2.2

**Additional context**
In `subsys/bluetooth/host/att_internal.h` line 12
```
#if BT_L2CAP_RX_MTU < CONFIG_BT_L2CAP_TX_MTU
#define BT_ATT_MTU BT_L2CAP_RX_MTU
#else
#define BT_ATT_MTU CONFIG_BT_L2CAP_TX_MTU
#endif
```

while the `BT_L2CAP_RX_MTU` is defined in `subsys/bluetooth/host/l2cap_internal.h` line 196.
```
#if defined(CONFIG_BT_HCI_ACL_FLOW_CONTROL)
#define BT_L2CAP_RX_MTU CONFIG_BT_L2CAP_RX_MTU
#else
#define BT_L2CAP_RX_MTU (CONFIG_BT_RX_BUF_LEN - \
			 BT_HCI_ACL_HDR_SIZE - BT_L2CAP_HDR_SIZE)
#endif
```

As the snippet indicates, unless the `CONFIG_BT_HCI_ACL_FLOW_CONTROL` is set to `y`, the `BT_L2CAP_RX_MTU` will be set to that following calculation. Which I believe resulting in 68 by default. 

My solution is to set `CONFIG_BT_HCI_ACL_FLOW_CONTROL` to `y` then the `server_mtu` value is calculated correctly. The documentation of this config doesn't mention anything about how it is affecting the MTU negotiation.

**Suggestion**
If this is intended, please update the documentation. If not, please address this MTU configuration so that less coupled with that guard.

joerchan, 2020-07-13:
Related: https://github.com/zephyrproject-rtos/zephyr/pull/26447

joerchan, 2020-07-14:
@jhedberg @carlescufi 
Any opinions on what to do about this issue?

The L2CAP RX MTU is set from two different config variables depending on if BT_HCI_ACL_FLOW_CONTROL is enabled or not.
I would think removing this dependency and always using the BT_L2CAP_RX_MTU would be best. But that might result in config
issues in existing applications.

Definition of BT_L2CAP_RX_MTU has depends on BT_HCI_ACL_FLOW_CONTROL
```
config BT_L2CAP_RX_MTU
	int "Maximum supported L2CAP MTU for incoming data"
	default 200 if BT_BREDR
	default BT_EATT_RX_MTU if BT_EATT
	default 65 if BT_SMP
	default 23
	range 70 1300 if BT_EATT
	range 65 1300 if BT_SMP
	range 23 1300
	depends on BT_HCI_ACL_FLOW_CONTROL
	help
```

joerchan, 2020-07-15:
@andung07 Don't you get something like 
```
warning: BT_L2CAP_RX_MTU (defined at subsys/bluetooth/host/Kconfig.l2cap:8) was assigned the value
'69' but got the value ''
````

andung07, 2020-07-15:
@joerchan  No, no such warning appears. Here is the complete log:
```
Processing solemetrix_dwm1001 (platform: nordicnrf52@4.3.0; board: solemetrix_dwm1001; framework: zephyr)
------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/solemetrix_dwm1001.html
PLATFORM: Nordic nRF52 4.3.0 > SoleMetrix DWM1001
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-zephyr 2.20300.200623 (2.3.0) 
 - framework-zephyr-canopennode 0.0.0-alpha+sha.5c6b0566d5 
 - framework-zephyr-civetweb 0.0.0-alpha+sha.99129c5efc 
 - framework-zephyr-cmsis 0.0.0-alpha+sha.542b2296e6 
 - framework-zephyr-fatfs 0.0.0-alpha+sha.9ee6b9b951 
 - framework-zephyr-hal-nordic 0.0.0-alpha+sha.742d8fb839 
 - framework-zephyr-libmetal 0.0.0-alpha+sha.3c3c9ec83b 
 - framework-zephyr-littlefs 0.0.0-alpha+sha.0aefdda69d 
 - framework-zephyr-loramac-node 0.0.0-alpha+sha.29e516ec58 
 - framework-zephyr-lvgl 0.0.0-alpha+sha.74fc2e753a 
 - framework-zephyr-mbedtls 0.0.0-alpha+sha.4bf099f125 
 - framework-zephyr-mcuboot 0.0.0-alpha+sha.e88113bbeb 
 - framework-zephyr-mcumgr 0.0.0-alpha+sha.5885efb7ca 
 - framework-zephyr-mipi-sys-t 0.0.0-alpha+sha.957d46bc3c 
 - framework-zephyr-open-amp 0.0.0-alpha+sha.724f7e2a45 
 - framework-zephyr-openthread 0.0.0-alpha+sha.a83d18cf18 
 - framework-zephyr-segger 0.0.0-alpha+sha.6fcf61606d 
 - framework-zephyr-tinycbor 0.0.0-alpha+sha.40daca97b4 
 - framework-zephyr-tinycrypt 0.0.0-alpha+sha.3e9a49d267 
 - framework-zephyr-trusted-firmware-m 0.0.0-alpha+sha.7de2daa196 
 - tool-cmake 3.16.4 
 - tool-dtc 1.4.7 
 - tool-gperf 3.1.0 
 - tool-ninja 1.9.0 
 - tool-sreccat 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)
Reading CMake configuration...
Parsing system calls...
Generating syscall files
Generating KObject files...
Validating driver...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <gms_packet>
|   |-- <gms_tools>
|-- <mpu6050_ext>
|   |-- <gms_packet>
|   |   |-- <gms_tools>
Building in release mode
Compiling .pio/build/solemetrix_dwm1001/zephyr_prebuilt/zephyr/misc/empty_file.c.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_battery.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_command.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_entrypoint.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_misc.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_public.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_replies.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_sampler.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_sender.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_service.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_settings.o
Compiling .pio/build/solemetrix_dwm1001/src/gms_uwb.o
Compiling .pio/build/solemetrix_dwm1001/zephyr_final/zephyr/misc/empty_file.c.o
Compiling .pio/build/solemetrix_dwm1001/offsets/zephyr/arch/arm/core/offsets/offsets.c.o
Compiling .pio/build/solemetrix_dwm1001/lib9c8/gms_tools/gms_tools.o
Generating header file with offsets .pio/build/solemetrix_dwm1001/zephyr/include/generated/offsets.h
Archiving .pio/build/solemetrix_dwm1001/zephyr/liboffsets.a
Indexing .pio/build/solemetrix_dwm1001/zephyr/liboffsets.a
[... 297 lines cut ...]
CONFIG_FLASH_MAP=y
CONFIG_NVS=y
CONFIG_SETTINGS=y

CONFIG_ADC=y
CONFIG_SPI=y
CONFIG_I2C=y
CONFIG_GPIO=y
CONFIG_SENSOR=y
CONFIG_MPU6050=y
CONFIG_MPU6050_TRIGGER_NONE=y
CONFIG_MPU6050_ACCEL_FS=8
CONFIG_MPU6050_GYRO_FS=2000
CONFIG_LIS2DH=y
CONFIG_M24M02=y

CONFIG_DW1000=y
CONFIG_DW1000_LOG_LEVEL=3

CONFIG_OPENOCD_SUPPORT=y
CONFIG_LOG=y
CONFIG_LOG_DEFAULT_LEVEL=4

CONFIG_DEVICE_POWER_MANAGEMENT=y
```

joerchan, 2020-07-15:
@andung07 Are you sure that that is a clean build?
```
➜ west build tests/bluetooth/shell -- -DCONFIG_BT_L2CAP_RX_MTU=250
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/joakim/repos/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/joakim/repos/zephyrproject/zephyr/tests/bluetooth/shell
-- Zephyr version: 2.3.99 (/home/joakim/repos/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter 
-- Board: nrf52840dk_nrf52840
-- Found toolchain: zephyr (/opt/zephyr-sdk-0.11.1)
-- Found west: /home/joakim/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /opt/zephyr-sdk-0.11.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/joakim/repos/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: /home/joakim/repos/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/joakim/repos/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h

warning: BT_L2CAP_RX_MTU (defined at subsys/bluetooth/host/Kconfig.l2cap:8) was assigned the value
'250' but got the value ''. Check these unsatisfied dependencies: BT_HCI_ACL_FLOW_CONTROL (=n). See
...
```
This is how it looks to me, it appears your output is different, what kind of build environment is that?

andung07, 2020-07-16:
@joerchan oh, that is platformio build. And yes, when I use west, it exactly gives me that warning.

carlescufi, 2020-07-17:
> The L2CAP RX MTU is set from two different config variables depending on if BT_HCI_ACL_FLOW_CONTROL is enabled or not.
> I would think removing this dependency and always using the BT_L2CAP_RX_MTU would be best. But that might result in config
> issues in existing applications.

I would tend to align with this. In the PR we need a note in the Stable API changes section of the release notes explaining what has changed and what needs to be done by the application to adapt.
It makes no sense that the L2CAP RX MTU depends on the ACL flow control, they just have nothing to do. I am not sure why we did this in the first place, @jhedberg ?

jhedberg, 2020-07-17:
@carles it's because the existence of the `acl_in_pool` buffer pool in `hci_core.c` depends on ACL flow control, and the BT_L2CAP_RX_MTU is used for determining the size of the buffers in it. If ACL flow control is disabled the normal RX pool is used and BT_L2CAP_RX_MTU is (AFAIK) of no use.

joerchan, 2020-07-17:
BT_L2CAP_RX_MTU sets the ATT_MTU:
```
#if BT_L2CAP_RX_MTU < CONFIG_BT_L2CAP_TX_MTU
#define BT_ATT_MTU BT_L2CAP_RX_MTU
#else
#define BT_ATT_MTU CONFIG_BT_L2CAP_TX_MTU
#endif
```
And the L2CAP_MPS with `CONFIG_BT_L2CAP_DYNAMIC_CHANNELS`
```
#define L2CAP_MAX_LE_MPS	BT_L2CAP_RX_MTU
/* For now use MPS - SDU length to disable segmentation */
#define L2CAP_MAX_LE_MTU	(L2CAP_MAX_LE_MPS - 2)
...
	/* Use existing MTU if defined */
	if (!chan->rx.mtu) {
		chan->rx.mtu = L2CAP_MAX_LE_MTU;
	}
```

jhedberg, 2020-07-18:
@joerchan I think those dependencies have appeared later, and it seems Kconfig hasn't kept up with the actual code changes.

joerchan, 2020-07-20:
@jhedberg Thank you for clarifying, I have updated the existing PR regarding L2CAP RX MTU for the SMP_SVR sample to include setting this from CONFIG_BT_l2CAP_RX_MTU always.

PR #26447 (2020-08-06): Bluetooth: host: Update configuration of L2CAP RX MTU setting
Remove dependency on CONFIG_BT_ACL_FLOW_CONTROL and use
CONFIG_BT_L2CAP_RX_MTU always to set L2CAP RX MTU.
The ATT MTU is set from two different KConfig options depending
on if CONFIG_BT_ACL_FLOW_CONTROL is enabled, which makes it
a confusing option and hard to provide a conf file that
supports multiple board configurations.

This changes the behavior when CONFIG_BT_ACL_FLOW_CONTROL and
CONFIG_BT_BUF_RX_LEN was used to set the L2CAP RX MTU, and by
extension the ATT MTU.

Fixes: #26536

Fix smp_svr Bluetooth configuration for when BT [... cut]
closed 2020-08-06 by carlescufi