zephyrproject-rtos/zephyr #26551: sam0 devicetree failing to compile

sslupsky, 2020-06-30:
**Describe the bug**
@galak When this commit of yours was merged: 8357005ba12f2a10d080d7bd40405c4e796c77cf

I am observing a build error now:
"devicetree error: clock-names property in /soc/sercom@42001800 in arduino_mkr_1310.dts.pre.tmp has 2 strings, expected 1 strings"

Is there a way to clean this up ?

galak, 2020-06-30:
@sslupsky can you give the command line or something to show the build error.

sslupsky, 2020-06-30:
@galak That (above) is exactly the error I copied from the terminal output.  Here are a few more lines:

```
devicetree error: clock-names property in /soc/sercom@42001800 in arduino_mkr_1310.dts.pre.tmp has 2 strings, expected 1 strings
CMake Error at /Users/stevenslupsky/Documents/source/zephyrproject/zephyr/cmake/dts.cmake:229 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /Users/stevenslupsky/Documents/source/zephyrproject/zephyr/cmake/app/boilerplate.cmake:514 (include)
  CMakeLists.txt:27 (include)
```

sslupsky, 2020-06-30:
@galak This was the build command:
```
west build -p auto -b arduino_mkr_1310 /Users/stevenslupsky/Documents/source/zephyrproject/dock1310 -- -DCMAKE_EXPORT_COMPILE_COMMANDS=1
```

sslupsky, 2020-06-30:
@galak Here are (what I think are the offending lines) from the dts.pre.tmp file:
```
&sercom0 {
 interrupts = <9 0>;
 clocks = <&gclk 0x14>, <&pm 0x20 2>;
 clock-names = "GCLK", "PM";
};
&sercom1 {
 interrupts = <10 0>;
 clocks = <&gclk 0x15>, <&pm 0x20 3>;
 clock-names = "GCLK", "PM";
};
&sercom2 {
 interrupts = <11 0>;
 clocks = <&gclk 0x16>, <&pm 0x20 4>;
 clock-names = "GCLK", "PM";
};
&sercom3 {
 interrupts = <12 0>;
 clocks = <&gclk 0x17>, <&pm 0x20 5>;
 clock-names = "GCLK", "PM";
};
&sercom4 {
 interrupts = <13 0>;
 clocks = <&gclk 0x18>, <&pm 0x20 6>;
 clock-names = "GCLK", "PM";
};
&sercom5 {
 interrupts = <14 0>;
 clocks = <&gclk 0x19>, <&pm 0x20 7>;
 clock-names = "GCLK", "PM";
};
```

sslupsky, 2020-06-30:
Appears to me like devicetree is saying we have multiple strings (which we do). 

The yaml definition file does not appear to define the property as an array?

sslupsky, 2020-06-30:
base.yaml appears to define the property:
```
    clock-names:
        type: string-array
        required: false
        description: name of each clock
```
For whatever reason, devicetree is not accepting multiple strings.

galak, 2020-06-30:
`arduino_mkr_1310` doesn't look to be a board in upstream.  Can you reproduce this with an upstream board?

sslupsky, 2020-06-30:
The mkr_1310 is based on the arduino_zero board in the tree.

In my overlay, I do define my own gclk's.  But, these do not appear to overlap with what you have done or use the clock-names property so I do not think that would be the issue here.

```
	gclks {
		compatible = "atmel,sam0-gclk";
		#clock-cells = <0>;

		gclk0: gclk0 {
			clock-generator = <0>;
			clock-frequency = <48000000>;
			#clock-cells = <0>;
			label = "GCLK0";
		};

		gclk1: gclk1 {
			clock-generator = <1>;
			clock-frequency = <32768>;
			#clock-cells = <0>;
			label = "GCLK1";
		};

		gclk2: gclk2 {
			clock-generator = <2>;
			clock-frequency = <32768>;
			#clock-cells = <0>;
			label = "GCLK2";
		};

		gclk3: gclk3 {
			clock-generator = <3>;
			clock-frequency = <8000000>;
			#clock-cells = <0>;
			label = "GCLK3";
		};

		gclk4: gclk4 {
			clock-generator = <4>;
			clock-frequency = <32768>;
			#clock-cells = <0>;
			label = "GCLK4";
		};
	};
```

sslupsky, 2020-06-30:
Here is a copy of what I am using for the arduino_mkr_1310.dts:
```
/*
 * Copyright (c) 2020 Scanimetrics Inc.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

/*
 *
 *  See this guide for device tree info
 *  https://docs.zephyrproject.org/latest/guides/dts/index.html
 *
 *  The device tree specification is here:
 *  https://www.devicetree.org/specifications/
 *
 */

/dts-v1/;
#include <atmel/samd21.dtsi>
#include <dt-bindings/gpio/gpio.h>

/ {
	model = "Arduino MKR 1310";
	compatible = "arduino,mkr_1310", "atmel,samd21g18a", "atmel,samd21";

	chosen {
		zephyr,console = &sercom5;
		zephyr,shell-uart = &sercom5;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,code-partition = &code_partition;
	};

	/* These aliases are provided for compatibility with samples */
	aliases {
		led0 = &led0;
	};

	leds {
		compatible = "gpio-leds";
		led0: led_0 {
			gpios = <&porta 20 GPIO_ACTIVE_HIGH>;
			label = "Yellow";
		};
	};
};

&cpu0 {
	clock-frequency = <48000000>;
};

//  SERCOM0 is "Wire"
&sercom0 {
	status = "okay";
	compatible = "atmel,sam0-i2c";
	//  see zephyr/include/dt-bindings/i2c/i2c.h for device tree frequency defines
	clock-frequency = <I2C_BITRATE_STANDARD>;
	#address-cells = <1>;
	#size-cells = <0>;
	// dma = <0>;
	dmas = <&dmac 0 0xb>, <&dmac 0 0xa>;
	dma-names = "tx", "rx";
	// interrupts = <9 0>;

	/* enable this when there is an ecc508 driver */
	// ecc0: ecc508@c0 {
	// 	compatible = "microchip,ecc508";
	// 	label = "ECC508";
	// 	reg = <0xc0>;
	// };
};

//  SERCOM1 is "SPI"
&sercom1 {
	status = "okay";
	compatible = "atmel,sam0-spi";
	#address-cells = <1>;
	#size-cells = <0>;
	dopo = <0>;
	dipo = <3>;
	// rxdma = 0;
	// txdma = 0;
	// interrupts = <10 0>;
};

//  SERCOM2 is "SPI1"
&sercom2 {
	status = "okay";
	compatible = "atmel,sam0-spi";
	#address-cells = <1>;
	#size-cells = <0>;
	dopo = <0>;
	dipo = <3>;
	cs-gpios = <&portb 8 GPIO_ACTIVE_LOW>, <&porta 14 GPIO_ACTIVE_LOW>;
	// interrupts = <11 0>;

	w25q16: w25q16@0 {
		compatible = "jedec,spi-nor";
		label = "W25Q16";
		jedec-id = [ef 40 15];
		size = <0x200000>;
		spi-max-frequency = <48000000>;
		reg = <0>;
		has-be32k;
		has-dpd;
		t-enter-dpd = <3000>;
		t-exit-dpd = <3000>;
	};
};

//  SERCOM4 is "SerialLoRa" / "Serial2"
//  Note: SERCOM4 and SERCOM2 conflict since they are configured
//  to share pins PA12 and PA15
&sercom4 {
	status = "okay";
	compatible = "atmel,sam0-uart";
	current-speed = <19200>;
	rxpo = <3>;
	txpo = <0>;
	// interrupts = <13 0>;
};

//  SERCOM5 is "Serial1"
&sercom5 {
	status = "okay";
	compatible = "atmel,sam0-uart";
	#address-cells = <1>;
	#size-cells = <0>;
	current-speed = <115200>;
	rxpo = <3>;
	txpo = <2>;
	// interrupts = <14 0>;
};

/*
 *  See this link for additional info on flash partitions
 *  https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/guides/dts/flash_partitions.inc
 *
 *  Note, for some reason, code is not place automatically in the code partition
 *  To do so, you need to ensure that the base address is set in your project config
 *  For example:
 *    CONFIG_FLASH_BASE_ADDRESS=0x2000
 *
 */

&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "sam-ba";
			reg = <0x00000000 0x2000>;
			read-only;
		};

		code_partition: partition@2000 {
			label = "code";
			reg = <0x2000 0x3a000>;
			read-only;
		};

		/*
		 * The final 16 KiB is reserved for the application
		 * and is used by file system if enabled.
		 */
		storage_partition: partition@3c000 {
			label = "storage";
			reg = <0x0003c000 0x00004000>;
		};
	};
};

&usb0 {
	status = "okay";
};
```

sslupsky, 2020-06-30:
I also noticed you deleted the "dma" property in the i2c bindings.  Has the i2c driver been updated to use the new "dmas" property?

sslupsky, 2020-06-30:
This is the full version of the overlay file I use:
```
/*
 * Copyright (c) 2020 Scanimetrics Inc.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <dt-bindings/gpio/gpio.h>

/ {
	debugs {
		compatible = "gpio";
		//  Use SENSOR1_EN (A5)
		debug0: debug_0 {
			gpios = <&porta 6 GPIO_ACTIVE_HIGH>;
			label = "DEBUG0";
		};
		//  Use SENSOR2_EN (A2)
		debug1: debug_1 {
			gpios = <&portb 3 GPIO_ACTIVE_HIGH>;
			label = "DEBUG1";
		};
	};

	aliases {
		debug0 = &debug0;
		debug1 = &debug1;

		/* MKR WAN 1310 only has 1 LED */
		led-mono = &led0;
	};

	gclks {
		compatible = "atmel,sam0-gclk";
		#clock-cells = <0>;

		gclk0: gclk0 {
			clock-generator = <0>;
			clock-frequency = <48000000>;
			#clock-cells = <0>;
			label = "GCLK0";
		};

		gclk1: gclk1 {
			clock-generator = <1>;
			clock-frequency = <32768>;
			#clock-cells = <0>;
			label = "GCLK1";
		};

		gclk2: gclk2 {
			clock-generator = <2>;
			clock-frequency = <32768>;
			#clock-cells = <0>;
			label = "GCLK2";
		};

		gclk3: gclk3 {
			clock-generator = <3>;
			clock-frequency = <8000000>;
			#clock-cells = <0>;
			label = "GCLK3";
		};

		gclk4: gclk4 {
			clock-generator = <4>;
			clock-frequency = <32768>;
			#clock-cells = <0>;
			label = "GCLK4";
		};
	};
};

&rtc {
	status = "okay";
	/*  use GCLK4 (xosc32k) as the source for the rtc  */
	/*  RTC clock frequency is defined in soc.h  */
	clock-generator = <4>;
	prescaler = <0>;
	/* RTC uses highest priority interrupt */
	interrupts = <3 0>;
};

&adc {
	status = "okay";
	/* main clock = 48 MHz */
	// gclk = <0>;
	/* prescaler = DIV32 --> Sample rate = 48 / 32 = 1.5 Msps */
	// prescaler = <32>;
	// interrupts = <19 0>;
};

&sercom4 {
	clocks = <&gclk3>;

	lora0 {
		status = "okay";
		compatible = "murata,lora";
		label = "cmwx1zzabz";
		mdm-reset-gpios = <&porta 27 GPIO_ACTIVE_LOW>;		// murata module input  MCU_NRST	(schematic RF_NRES)
		mdm-boot0-gpios = <&portb 9 GPIO_ACTIVE_HIGH>;		// murata module input  BOOT0		(schematic BOOT0)
		mdm-rf-ssn-gpios = <&porta 14 GPIO_ACTIVE_LOW>;		// murata module input  PB12		(schematic RF_SSN)
		mdm-rf-irq-gpios = <&porta 28 GPIO_ACTIVE_HIGH>;	// murata module output PA4		(schematic RF_IRQ)
		// mdm-sck-gpios = <&porta 13 GPIO_ACTIVE_HIGH>;	// murata module output PB13		(schematic SCK)
		// mdm-miso-rx-gpios = <&porta 15 GPIO_ACTIVE_HIGH>;	// murata module output PB14		(schematic MISO_RX)
		// mdm-mosi-tx-gpios = <&porta 12 GPIO_ACTIVE_HIGH>;	// murata module output PB15		(schematic MOSI_TX)
	};
};

&sercom1 {
	cs-gpios = <&porta 11 GPIO_ACTIVE_LOW>, <&portb 10 GPIO_ACTIVE_LOW>;
	// interrupts = <11 0>;

	flash1: flash@0 {
		compatible = "jedec,spi-nand";
		label = "SPINAND0";
		jedec-id = [98 cd];
		size = <0x20000000>;
		spi-max-frequency = <12000000>;
		reg = <0>;
		hold-gpios = <&portb 23 GPIO_ACTIVE_LOW>;
		spi-transfer-mode = <3>;
	};

        sdhc0: sdhc@1 {
                compatible = "zephyr,mmc-spi-slot";
                reg = <1>;
                status = "okay";
                label = "SDHC0";
                spi-max-frequency = <12000000>;
        };
};

&flash1 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		external_partition: partition@0 {
			label = "external";
			reg = <0x00000000 0x20000000>;
		};
	};

};

&sercom0 {
	clock-frequency = <I2C_BITRATE_FAST>;

	ads1115: ads1115@48 {
		compatible = "ti,ads1115";
		reg = <0x48>;
		label = "ADS1115";
		#io-channel-cells = <4>;
		alert-gpios = <&porta 21 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
	};

	tmp116: tmp116@49 {
		compatible = "ti,tmp116";
		reg = <0x49>;
		label = "TMP116";
	};

	bq24195: bq24195@6B {
		compatible = "ti,bq24195";
		reg = <0x6B>;
		label = "PMIC";
	};
};

/*  SERCOM3 is used for board to board communication  */
/*  The dock is the master  */
/*  The ISB is the slave and emulates an AT24 EEPROM  */
&sercom3 {
	status = "okay";
	compatible = "atmel,sam0-i2c";
	#address-cells = <1>;
	#size-cells = <0>;
	clock-frequency = <I2C_BITRATE_FAST>;

	eeprom@54 {
		compatible = "atmel,at24";
		reg = <0x54>;
		label = "EEPROM_SLAVE_0";
		size = <256>;
		pagesize = <16>;
		address-width = <8>;
		timeout = <5>;
	};
};

&sercom5 {
	clocks = <&gclk0>;
};
```

sslupsky, 2020-07-01:
I think I see the problem.  Somehow devicetree is correlating the clocks property and the clock-names property.  The default settings in samd21.dtsi show something like:
```
	interrupts = <13 0>;
	clocks = <&gclk 0x18>, <&pm 0x20 6>;
	clock-names = "GCLK", "PM";
```

So, if you have one phandle assigned to the clocks property, then you can only have one string assigned to the clock-names property.

In my case, I was using the clocks property to set the gclk and there was only one phandle.  So, that conflicted with the "default" setting that has two clock-names strings.

Why are there two phandles assigned to the clocks property in the default samd21.dtsi?

sslupsky, 2020-07-01:
I am having another problem now with the adc.
```
/Users/stevenslupsky/Documents/source/zephyrproject/dock1310/prj.conf:3: warning: attempt to assign the value 'y' to the undefined symbol ADC_0
```

In my prj.conf I have
```
CONFIG_ADC_0=y
```

That worked fine when I was building in 2.2 but now it causes the error shown above.  The label is "ADC_0" so this should be fine.  Did something change with the ADC configuration?

nzmichaelh, 2020-07-01:
> Why are there two phandles assigned to the clocks property in the default samd21.dtsi?

Hi @sslupsky peripherals on the SAMD21 need both the GCLK and the PM (power management) to be set up before they work, which is why each entry has two clocks.  You also need to supply a name for each clock.

nzmichaelh, 2020-07-01:
> That worked fine when I was building in 2.2 but now it causes the error shown above. The label is "ADC_0" so this should be fine. Did something change with the ADC configuration?

I don't know specifically, but in general Zephyr uses DeviceTree to enable modules these days.  I wouldn't be surprised if the KConfig way is a legacy way that has been deleted.

You'll probably need a 

```
&adc {
    status = "okay";
}
```

in your DeviceTree file and a `CONFIG_ADC=y` in your defconfig.

Also, consider opening separate bugs for separate issues.  You're more likely to get a response from, say, someone who knows the ADC configuration in an ADC specific bug.

nzmichaelh, 2020-07-01:
Huh, `adc` is already enabled in the base DeviceTree files so there's nothing to do there.

galak, 2020-07-01:
> In my case, I was using the clocks property to set the gclk and there was only one phandle. So, that conflicted with the "default" setting that has two clock-names strings.
> 
> Why are there two phandles assigned to the clocks property in the default samd21.dtsi?

The two clocks are there for cases like:

```
drivers/counter/counter_sam0_tc32.c:	.pm_apbcmask = BIT(DT_INST_CLOCKS_CELL_BY_NAME(n, pm, bit)),	\
drivers/i2c/i2c_sam0.c:	.pm_apbcmask = BIT(DT_INST_CLOCKS_CELL_BY_NAME(n, pm, bit)),	\
drivers/serial/uart_sam0.c:	.pm_apbcmask = BIT(DT_INST_CLOCKS_CELL_BY_NAME(n, pm, bit)),	\
drivers/spi/spi_sam0.c:	.pm_apbcmask = BIT(DT_INST_CLOCKS_CELL_BY_NAME(n, pm, bit)),	\
```

galak, 2020-07-01:
> In my prj.conf I have
> 
> ```
> CONFIG_ADC_0=y
> ```
> 
> That worked fine when I was building in 2.2 but now it causes the error shown above. The label is "ADC_0" so this should be fine. Did something change with the ADC configuration?

We removed per instance enables from Kconfig and have this completely driven by devicetree now.

sslupsky, 2020-07-01:
@galak @nzmichaelh Thanks for the notes on the adc change.  Apologies for bringing it up here but I've been experiencing "issues" with the 2.3 update and was hoping it was a quick fix.  I did simply remove the enable from the prj.conf and that appeared to fix the build issue.  Though, I am not sure that the ADC is yet working because now there are about a zillion compile errors because of the data type changes.  

Where is the change to the ADC configuration documented in the release notes?  When I checked last night, I could not find anything that explained the change.

On the issue of the data types, why was the data type change done with a point release?  Should that have been deferred to 3.0?  It is turning out to be a major change that requires extensive changes to existing code.

galak, 2020-07-01:
> On the issue of the data types, why was the data type change done with a point release? Should that have been deferred to 3.0? It is turning out to be a major change that requires extensive changes to existing code.

There is a config option you can enable to support the old data types: `CONFIG_DEPRECATED_ZEPHYR_INT_TYPES`.

sslupsky, 2020-07-01:
Speaking of the release notes, where are the changes to the clocks documented?  I could not find anything in the 2.3 release notes regarding this either.

galak, 2020-07-01:
As for the release notes, I don't think there is anything in them related to these two items.  Agreed they should have been documented there.

sslupsky, 2020-07-01:
Yeah, I found the config option, for some reason I am still getting a zillion errors.  The problem is compounding because I am using the terminal inside vscode so it has a limited scroll back.  So, the only errors that I see are the copious errors that are output by the log defines failing to compile properly.  Which brings up another point, there should be a way to suppress all the "extra" errors from those macros.  It really is a huge problem when building that a type error leads to a billion errors because of the log macros.  <end of rant>

sslupsky, 2020-07-01:
BTW, I like ideas for the changes to the clock configuration for sam0.  If you had not already done so, I was about to submit a PR for the work I had already done on this.  Thank you.

closed 2020-07-01 by sslupsky