zephyrproject-rtos/zephyr #26568: tests: net: socket: af_packet: is ethernet cable now mandatory to run this test ?

ABOSTM, 2020-07-01:
**Question**
Test tests/net/socket/af_packet/
used to be successful on stm32 boards (nucleo_f429zi, nucleo_f746zg for example), on test bench.
Since merged of PR  #26339  (specially commit net: conn: Handle multi interface AF_PACKET recv() properly)
the test is now failed.

Note: On test bench, there is no ethernet cable connected to the boards.
But the test used to be successful without cable.

Now test is failed, but if I plug an ethernet cable, it is passed.

Does it means that it is now mandatory to plug ethernet cable to the boards to pass this test ?

ABOSTM, 2020-07-01:
^^  @jukkar

jukkar, 2020-07-10:
No, connecting an ethernet cable should not be needed. There is now some side effect when running the test in the stm32 boards. I am currently on vacation without proper hw so cannot pinpoint immediately what should be changed in the test.

galak, 2020-07-29:
Was able to run the test on `disco_l475_iot1` but seeing it hang on `frdm_k64f`

PR #27252 (2020-07-30): tests: net: af_packet: Fix running in real hw
The tests were failing when run on those boards that had
physical Ethernet support. In that case there were three Ethernet
interfaces, two simulated ones for testing, and the third one
was provided by the board. This extra Ethernet interface caused
the test to fail as the test was sending data to wrong interface.

Fixes #27238
Fixes #26568
commit 7d90812f (2020-07-30): tests: net: af_packet: Fix running in real hw

The tests were failing when run on those boards that had
physical Ethernet support. In that case there were three Ethernet
interfaces, two simulated ones for testing, and the third one
was provided by the board. This extra Ethernet interface caused
the test to fail as the test was sending data to wrong interface.

Fixes #27238
Fixes #26568
closed 2020-07-30 by carlescufi