zephyrproject-rtos/zephyr #26569: cmake: remove '-ffreestanding' instead of adding '-fno-freestanding'

hakonfam, 2020-07-01:
Currently you might end up with both `-ffreestanding` and `-fno-freestanding` defined.

To test, build any unit test for native_posix board.

```
ccache /usr/bin/gcc 
# ...
-ffreestanding 
# ...
-fno-freestanding 
# ...
```

Instead, simply ensure that `-ffreestanding` is not added in the first place, if it should not be there.

This looks messy, and brakes some compilers (e.g. `clang-tidy`'s gcc front-end).

tejlmand, 2020-07-01:
Not really a bug, as this is allowed by gcc, but as a cleanup could benefit `clang-tidy` and also looks nicer it is still a valid suggestion.
Changed to `enhancement`

tejlmand, 2020-07-01:
have you used `ZEPHYR_TOOLCHAIN_VARIANT=llvm` ?
https://docs.zephyrproject.org/latest/getting_started/toolchain_host.html?highlight=llvm

aaronemassey, 2022-05-12:
If you use clangd. You can also add the following to your `.clangd` file:

```
CompileFlags:                     # Tweak the parse settings
  Remove: [-fno-freestanding]
```

So it doesn't complain about unrecognized argument error.

nordicjm, 2024-11-19:
Cannot replicate with zephyr 4.0 using `native_sim` so seems to have been fixed, feel free to re-open if still an issue

closed 2024-11-19 by nordicjm