zephyrproject-rtos/zephyr #26584: Multicast emission is only possible for ipv4 starting with 224.

elsinkior, 2020-07-01:
**Describe the bug**
ipv4 multicast emission only possible for @ starting with 224. The following modifications could be applied so that the L2 layer replaces the MAC address of the interface by a multicast MAC address for all multicast addresses.  

1) Open the file $ ZEPHYR_BASE / zephyr / subsys / net / l2 / ethernet / ethernet.c
2) Redefine the content of the following static functions:

static inline bool ethernet_ipv4_dst_is_broadcast_or_mcast(struct net_pkt *pkt)
{
    if (net_ipv4_is_addr_bcast(net_pkt_iface(pkt), &NET_IPV4_HDR(pkt)->dst) ||
        net_ipv4_is_addr_mcast(&(NET_IPV4_HDR(pkt)->dst)))
    {
        return true;
    }
    return false;
}

static bool ethernet_fill_in_dst_on_ipv4_mcast(struct net_pkt *pkt, struct net_eth_addr *dst)
{
    if (net_ipv4_is_addr_mcast(&(NET_IPV4_HDR(pkt)->dst))) 
    {
        dst->addr[0] = 0x01;
        dst->addr[1] = 0x00;
        dst->addr[2] = 0x5e;
        dst->addr[3] = NET_IPV4_HDR(pkt)->dst.s4_addr[1];
        dst->addr[4] = NET_IPV4_HDR(pkt)->dst.s4_addr[2];
        dst->addr[5] = NET_IPV4_HDR(pkt)->dst.s4_addr[3];
        dst->addr[3] &= 0x7f;
        return true;
    }
    return false;
}

**To Reproduce**
NA

**Expected behavior**
Multicast mac addresses should be used at ethernet level instead of unicast address for all multicast ipv4 addresses.

**Impact**
NA

**Logs and console output**
NA

**Environment (please complete the following information):**
 - OS: zephyr v2.3.99 - board native_posix or mimxrt1064_evk
 - Toolchain: zephyr sdk v0.11.3
 - SHA: 979e124e21fb05cfedf8ed12078983d40e2f0b6c

**Additional context**
NA

jukkar, 2020-07-20:
@elsinkior you proposed change looks good, could you send a PR for that?

jukkar, 2020-09-29:
The upstream code looks to work ok. It checks 224 value directly instead of using the function. I can refactor the code a bit to use the function. Lowering the priority.

PR #28760 (2020-09-30): net: ethernet: Check IPv4 multicast pkt using util function
Instead of directly checking the multicast IPv4 address, use
the net_ipv4_is_addr_mcast() utility function.

Fixes #26584
commit ccea4d32 (2020-09-30): net: ethernet: Check IPv4 multicast pkt using util function

Instead of directly checking the multicast IPv4 address, use
the net_ipv4_is_addr_mcast() utility function.

Fixes #26584
closed 2020-09-30 by jukkar