zephyrproject-rtos/zephyr #26585: IPv4 multicast datagrams can't be received for mimxrt1046_evk board (missing ethernet API)

elsinkior, 2020-07-01:
**Describe the bug**
In order to support UDP ipv4 multicast reception, it is necessary to modify the imx6 ethernet driver API to join a multicast group.
For imx6, it is the file eth_mcux.c, which provides the configuration function set_config to add or remove from a multicast group the interface.

1) Open the file $ ZEPHYR_BASE / zephyr / drivers / ethernet / eth_mcux.c
2) the following static function could be defined

```
static int set_config (struct device * dev,
                           enum ethernet_config_type type,
                           const struct ethernet_config * config)
{
    int result = -1;
    switch (type)
    {
    ETHERNET_CONFIG_TYPE_FILTER box:
    {
        switch (config-> filter.type)
        {
        ETHERNET_FILTER_TYPE_SRC_MAC_ADDRESS box:
        {
            / * Adds or leaves the ENET device to a multicast group. * /
            uint8_t multicastMacAddr [6];
            for (uint32_t i = 0; i <6; i ++)
            {
                multicastMacAddr [i] = config-> filter.mac_address.addr [i];
                if (config-> filter.set)
                {
                    printk ("\ r \ nENET_AddMulticastGroup \ r \ n");
                    ENET_AddMulticastGroup (ENET, multicastMacAddr);
                }
                else
                {
                    printk ("\ r \ nENET_LeaveMulticastGroup \ r \ n");
                    ENET_LeaveMulticastGroup (ENET, multicastMacAddr);
                }
                result = 0;
            }
        }
        break;
        default:
        {
            // Nothing
        }
        break;
        }
        break;
        default:
        {
            // Nothing
        }
        break;
    }
    }
    return result;
}
```
3) ... and added to the api by modifying the api_funcs structure:

```
static const struct ethernet_api api_funcs = {
    .iface_api.init = eth_iface_init,
    .get_capabilities = eth_mcux_get_capabilities,
    .send = eth_tx,
    .set_config = set_config, / * Adding the set config function to the API * /
};
```
**To Reproduce**
Try to receive UDP ipv4 multicast datagrams.

**Environment (please complete the following information):**
 - OS: linux
 - Toolchain: zephyr sdk v0.1.13
 - board: mimxrt1064_evk
 - SHA: 979e124e21fb05cfedf8ed12078983d40e2f0b6c

jukkar, 2020-07-10:
For IPv6 there is `net_if_mcast_mon_register()` that can be used to register a function that will be called if a multicast address is added/removed to the network interface. That function should have IPv4 address support and then we could enhance the IPv4 support in the driver side. I am not convinced that we should have this implemented via `set_config()`.

agansari, 2020-07-10:
@jukkar i've looked at the driver and adding the driver part is not difficult we have to re-purpose `net_if_mcast_cb` and remove ipv6 configuration, also re-purposing net_if_mcast_mon_register() is not difficult.
However I need to ask you don't we need to add some new ipv4 functions/events/handles that handle joining/leaving ethernet multicast groups? I.e. networking stack support.

jukkar, 2020-07-20:
> However I need to ask you don't we need to add some new ipv4 functions/events/handles that handle joining/leaving ethernet multicast groups? I.e. networking stack support.

Yes, we need to join/leave multicast group when IPv4 address is added/removed to/from network interface. Sorry for not being more clear in my words above. So this would need to be implemented, basically the functionality should be similar as with IPv6.

dleach02, 2020-08-13:
Is this still a problem with tcp2?

jukkar, 2020-08-14:
> On the issue of adding join/leave API, is this something that is generically needed across all drivers?

Probably yes, typically (at least for IPv6 mcast msg) the driver needs to program the Ethernet controller so that it can receive the multicast packets.

> Also, if we add the .set_config would that be enough to close this ticket?

We have an API where a driver can register a function to be called when net stack joins/leaves a multicast group. Right now that API supports only IPv6 addresses. That API would need to be extended to support also IPv4 addresses. See for example `drivers/ethernet/eth_mcux.c`and search for `net_if_mcast_mon_register()` to see how it is used.

But this is only part of the story as also network stack would need to support IPv4 multicast handling similar way as what have been done for IPv6. Right now there is an API to add a IPv4 multicast address to the network interface `net_if_ipv4_maddr_add(()` but it does not trigger a call to multicast monitoring which would inform the driver that the multicast group needs to be joined.
Changes to the code to support this are not very big, it is just that someone needs to join the dots to make this work and test it properly.

dleach02, 2020-08-14:
@jukkar, If I'm understanding you then it seems like this isn't a platform specific bug?

jukkar, 2020-08-17:
@dleach02 First of all, this is not a bug, we are just missing a feature. And second, this is not a platform specific issue.

dleach02, 2020-08-17:
Cool... so @jukkar we can convert this to an enhancement request then?

jukkar, 2020-08-18:
> we can convert this to an enhancement request then?

Yep, done.

PR #38501 (2021-09-29): net: if: Add IPv4 support to multicast monitor
Make multicast group join/leave monitor support both IPv6 and IPv4 addresses.

Fixes #26585
commit 7926657b (2021-09-29): net: if: Add IPv4 support to multicast monitor

Make multicast group join/leave monitor support both IPv6 and IPv4
addresses.

Fixes #26585
closed 2021-09-29 by cfriedt