zephyrproject-rtos/zephyr #26600: net.util test is broken on MPU-enabled ARM platforms

stephanosio, 2020-07-02:
**Describe the bug**
The `tests/net/utils/net.util` test fails on the MPU-enabled ARM platforms.

Known failing platforms:
* `mps2_an385` (QEMU)
* `mps2_an521` (QEMU)
* `sam_e70_xplained`
* `frdm_k64f`

**To Reproduce**
Build and run `tests/net/utils/net.util` on any of the aformentioned known failing platforms.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.3.0-666-g2a706d9f2096  ***
Running test suite test_utils_fn
===================================================================
START - test_net_addr
START - test_ipv4_pton_1
Failed to verify �
failed
START - test_ipv4_pton_2
Failed to verify �
failed
START - test_ipv4_pton_3
Failed to verify �
failed
START - test_ipv4_pton_4
Failed to verify ����255.255.255
failed
START - test_ipv4_pton_5
passed
START - test_ipv4_pton_6
Failed to verify
failed
START - test_ipv4_pton_7
Failed to verify
failed
START - test_ipv4_pton_8
Failed to verify
failed
START - test_ipv6_pton_1
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x2000286c
E: r0/a1:  0x00000002  r1/a2:  0x20000d3c  r2/a3:  0x00000000
E: r3/a4:  0x2000286c r12/ip:  0x0000edef r14/lr:  0x00000955
E:  xpsr:  0xa1000000
E: Faulting instruction address (r15/pc): 0x000043a6
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20002310 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.3
 - Commit SHA: 2a706d9f209648533d952a1d1d439bccf22cabd3

stephanosio, 2020-07-02:
cc @jukkar

stephanosio, 2020-07-02:
TODO: Bisect

chrta, 2020-07-07:
When i executed sanitiycheck with undefined behavior sanitizer, i got some alignment issues from some network tests, maybe this is related. See https://github.com/zephyrproject-rtos/zephyr/pull/26674 for the addition of sanitycheck.

ioannisg, 2020-07-21:
I cannot reproduce on latest master, on any of the aforementioned platforms (btw, the test seems to be excluded on an521, due to lack of HW support).

I do reproduce this on 2a706d9f209648533d952a1d1d439bccf22cabd3.

ioannisg, 2020-07-21:
Git bisect shows that the test was fixed in 937482b4a7c111695a9d5677df14ef2947366174

ioannisg, 2020-07-21:
Closing this @MaureenHelm @stephanosio , as I cannot reproduce after commit 937482b on FRDM or SAM E70, pls, re-open if you see issues.

closed 2020-07-21 by ioannisg