zephyrproject-rtos/zephyr #26607: STM32F0 nucleo PWM output not working

nandra, 2020-07-02:
Use basic sample blinky_led to verify stm32f0 nucleo pwm output. Enable timer2 and pwm nodes in devicetree. Despite of that fact there  is no pwm output on any pin. I think that pinmux is maybe wrong but cannot see even pinmux defines for timer. Does anyone try to run pwm on stm32f0 nucleo board to share? Thanks.

erwango, 2020-07-02:
@nandra, there are 3 STM32F0 based nucleo boards ported in Zephyr, can you specify the one you're using ?
(Support is faster when we have precise information on the issue.)
Can you try to adapt this for your board and have a new try: https://github.com/zephyrproject-rtos/zephyr/pull/22277 ?

nandra, 2020-07-02:
@erwango I'm using nucleo_f091rc board. I have adaptions done and driver is working just I think no signal is muxed properly thus  I don't have any output. Thanks.

FRASTM, 2020-07-02:
for example 
TIM2 CH2 output is available on PA1 (alternate 2)
TIM2 CH2 output is available on PB3 (alternate 2)
but the STM32F0 pinmux.c does not reference any pin for TMR 2
Both Pins are available on the nucleo board ARDuino connectors

nandra, 2020-07-02:
@FRASTM ok thanks I'll add it to pinmux and retest. Thanks.

erwango, 2020-07-03:
@nandra please get back to us when you have new results

nandra, 2020-07-06:
This is my changes:
```
diff --git a/boards/arm/nucleo_f091rc/nucleo_f091rc.dts b/boards/arm/nucleo_f091rc/nucleo_f091rc.dts
index 124b3b1..4b37e06 100644
--- a/boards/arm/nucleo_f091rc/nucleo_f091rc.dts
+++ b/boards/arm/nucleo_f091rc/nucleo_f091rc.dts
@@ -93,3 +93,14 @@
 &adc1 {
 	status = "okay";
 };
+
+&timers2 {
+	status = "okay";
+};
+
+&pwm {
+	status = "okay";
+};
+
+
+
diff --git a/boards/arm/nucleo_f091rc/pinmux.c b/boards/arm/nucleo_f091rc/pinmux.c
index 3cdc008..b1eb632 100644
--- a/boards/arm/nucleo_f091rc/pinmux.c
+++ b/boards/arm/nucleo_f091rc/pinmux.c
@@ -43,6 +43,10 @@ static const struct pin_config pinconf[] = {
 #ifdef CONFIG_ADC_1
 	{STM32_PIN_PA0, STM32F0_PINMUX_FUNC_PA0_ADC_IN0},
 #endif /* CONFIG_ADC_1 */
+
+
+{STM32_PIN_PA1, STM32_PINMUX_ALT_FUNC_2}
+
 };
 
 static int pinmux_stm32_init(struct device *port)
diff --git a/dts/arm/st/f0/stm32f091.dtsi b/dts/arm/st/f0/stm32f091.dtsi
index 919f861..14215e6 100644
--- a/dts/arm/st/f0/stm32f091.dtsi
+++ b/dts/arm/st/f0/stm32f091.dtsi
@@ -21,7 +21,7 @@
 			status = "disabled";
 			label = "TIMERS_2";
 
-			pwm {
+			pwm: pwm_2 {
 				compatible = "st,stm32-pwm";
 				status = "disabled";
 				st,prescaler = <0>;
diff --git a/samples/basic/fade_led/src/main.c b/samples/basic/fade_led/src/main.c
index df14b62..d1def19 100644
--- a/samples/basic/fade_led/src/main.c
+++ b/samples/basic/fade_led/src/main.c
@@ -14,9 +14,9 @@
 #include <sys/printk.h>
 #include <device.h>
 #include <drivers/pwm.h>
-
+/*
 #if defined(DT_ALIAS_PWM_LED0_PWMS_CONTROLLER) && defined(DT_ALIAS_PWM_LED0_PWMS_CHANNEL)
-/* get the defines from dt (based on alias 'pwm-led0') */
+get the defines from dt (based on alias 'pwm-led0') 
 #define PWM_DRIVER	DT_ALIAS_PWM_LED0_PWMS_CONTROLLER
 #define PWM_CHANNEL	DT_ALIAS_PWM_LED0_PWMS_CHANNEL
 #ifdef DT_ALIAS_PWM_LED0_PWMS_FLAGS
@@ -27,7 +27,7 @@
 #else
 #error "Choose supported PWM driver"
 #endif
-
+*/
 /*
  * 50 is flicker fusion threshold. Modulated light will be perceived
  * as steady by our eyes when blinking rate is at least 50.
@@ -35,7 +35,7 @@
 #define PERIOD (USEC_PER_SEC / 50U)
 
 /* in micro second */
-#define FADESTEP	2000
+#define FADESTEP	2500
 
 void main(void)
 {
@@ -45,21 +45,22 @@ void main(void)
 
 	printk("PWM demo app-fade LED\n");
 
-	pwm_dev = device_get_binding(PWM_DRIVER);
+	pwm_dev = device_get_binding("PWM_2");
 	if (!pwm_dev) {
-		printk("Cannot find %s!\n", PWM_DRIVER);
+		printk("Cannot find %s!\n", "PWM_2");
 		return;
 	}
 
+int err;
 	while (1) {
-		if (pwm_pin_set_usec(pwm_dev, PWM_CHANNEL,
-					PERIOD, pulse_width, PWM_FLAGS)) {
-			printk("pwm pin set fails\n");
+		if (err = pwm_pin_set_usec(pwm_dev, 1,
+					PERIOD, pulse_width, 0)) {
+			printk("pwm pin set fails, %x\n", err);
 			return;
 		}
-
 		if (dir) {
 			if (pulse_width < FADESTEP) {
+		                printk("stop %x \t", pulse_width);
 				dir = 0U;
 				pulse_width = 0U;
 			} else {
@@ -69,11 +70,13 @@ void main(void)
 			pulse_width += FADESTEP;
 
 			if (pulse_width >= PERIOD) {
+		                printk("stopF %d \t", pulse_width);
 				dir = 1U;
 				pulse_width = PERIOD;
 			}
 		}
 
 		k_sleep(K_SECONDS(1));
+		printk("pulse_width:  %d \n", pulse_width);
 	}
 }
```

but unfortunately cannot get output on any of above mentioned pins. Am I missing still something? Thanks.

FRASTM, 2020-07-06:
what is the PA1 for TIM2_ CH2 configuration : 
Pull up or Pull down ?  
and its speed : STM32_OSPEEDR_HIGH_SPEED  or STM32_OSPEEDR_VERY_HIGH_SPEED

nandra, 2020-07-06:
I just set: ```+{STM32_PIN_PA1, STM32_PINMUX_ALT_FUNC_2}``` alternate function 2. Do you suggest to use pullup + high speed options? Thanks.

FRASTM, 2020-07-06:
yes, if you could  try a combination of STM32_PUSHPULL_PULLUP |  STM32_OSPEEDR_HIGH_SPEED
or STM32_PUSHPULL_PULL_DOWN |  STM32_OSPEEDR_HIGH_SPEED

nandra, 2020-07-07:
@FRASTM I've tried:
```
{STM32_PIN_PB3, STM32_PINMUX_ALT_FUNC_2 | STM32_PUSHPULL_PULLUP | STM32_OSPEEDR_HIGH_SPEED},
```
but unfortunately still nothing.

FRASTM, 2020-07-07:
why not PA1 
{STM32_PIN_PA1, STM32_PINMUX_ALT_FUNC_2| STM32_PUSHPULL_PULLUP | STM32_OSPEEDR_HIGH_SPEED | }

nandra, 2020-07-09:
OK this seems to do the trick. Can I create pull request with related changes to have PWM working on F0 nucleo? Thanks.

erwango, 2020-07-09:
> OK this seems to do the trick. Can I create pull request with related changes to have PWM working on F0 nucleo? Thanks.

Sure, go ahead!

erwango, 2020-07-23:
@nandra if you're satisfied with the proposal can you close the point

erwango, 2020-09-11:
Closing the issue, as last feedback was positive and no update since then

closed 2020-09-11 by erwango