zephyrproject-rtos/zephyr #26617: devicetree: sam0 gclk

sslupsky, 2020-07-02:
@galak @nzmichaelh I have been looking at the changes to the sam0 gclk framework.  I am curious if you plan to add some additional clock-cells or otherwise expand the configuration?  At the moment, the atmel,samd2x-gclk.yaml compat defines one clock cell for the clkctrl_id.

How do you propose to handle the other components of the gclk module (genctrl and gendiv)?  I have noticed that a few compatibles allow for assignment of a "clock-generator" (the sam0-rtc compat) or "gclk" (sam0-adc compat).  However, other elements of the gclk config are missing.

I created a compatible for the sam0 gclk configuration a while ago.  Here is a copy of that compatible.  This compatible lets you define the configuration of a particular gclk<n>.  Please let me know if you think this would be useful and I should submit this as a PR.

```
/*
 * Copyright (c) 2020 Scanimetrics Inc.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

description: ATMEL GCLK (Generic Clock Controller)

compatible: "atmel,sam0-gclk"

include: [clock-controller.yaml, base.yaml]

child-binding:
    description:  GCLK child node
    properties:
        clock-generator:
            type: int
            required: false
            description: generic clock generator

        clock-source:
            type: int
            required: false
            description: generic clock generator source oscillator

        clock-standby:
            type: int
            required: false
            description: generic clock generator run in standby

        clock-divsel:
            type: int
            required: false
            description: generic clock generator divide selection

        clock-div:
            type: int
            required: false
            description: generic clock generator division factor

        clock-frequency:
            type: int
            description: output clock frequency (Hz)
            required: true

        label:
            required: true
            type: string
            description: Human readable string describing the device (used by Zephyr for API name)

```

nzmichaelh, 2020-07-02:
Hi @sslupsky I haven't thought about this area.  You need a `gclk` in DeviceTree so that other peripherals can reference it, which (I assume) is why that was added.  Apart from that the clocks are all configured explicitly in the common soc_samd2x.c code.

If you want, prototype up what a DeviceTree driven version of soc_samd2x.c looks like.  You'll also need to address constants like `SOC_ATMEL_SAM0_MCK_FREQ_HZ` in soc.h.

sslupsky, 2020-07-02:
Yeah, this is the problem with the way the clocks are currently configured, all the clocks are hard coded.  I think it is preferred if they were configurable to some degree in the devicetree.

In my use case, to minimize power consumption I need to use the OSC8M for certain peripherals and the XOSC32K for others and the OSCULP32K for yet others.  Further, some clocks need to turn off during sleep, others need to continue to run during sleep (RUNSTDBY) and yet others need to only run when being used (ONDEMAND).  What I am finding is that you have to go and modify every device driver you use to ensure the clocks are configured properly ... which creates a large amount of overhead because now you need to maintain the changes to the drivers.

The most important things I see at the moment that cannot be done with devicetree are:

1. setting the correct clock generator connected to the peripheral
2. setting the correct divider for the clock generator (and hence the frequency)
3. setting the source oscillator for the generator
4. setting the RUNSTDBY and ONDEMAND bits for the generator
5. setting the RUNSTDBY and ONDEMAND bits for the source oscillator

closed 2020-09-08 by sslupsky