zephyrproject-rtos/zephyr #26619: tests/unit/rbtree fails

andrewboie, 2020-07-02:
**Describe the bug**
```
Running test suite test_rbtree
===================================================================
START - test_rbtree_spam
Checking trees built from 2 nodes...
Checking trees built from 3 nodes...
Checking trees built from 5 nodes...
Checking trees built from 7 nodes...
Checking trees built from 14 nodes...
Checking trees built from 19 nodes...
Checking trees built from 24 nodes...
Checking trees built from 29 nodes...
Checking trees built from 53 nodes...
Checking trees built from 69 nodes...
Checking trees built from 89 nodes...
Checking trees built from 157 nodes...
Checking trees built from 220 nodes...
Checking trees built from 253 nodes...
Checking trees built from 256 nodes...
 PASS - test_rbtree_spam
===================================================================
START - test_rbtree_container
 PASS - test_rbtree_container
===================================================================
START - test_check_rbtree_perf

    Assertion failed at /home/apboie/projects/zephyr2/zephyr/tests/unit/rbtree/main.c:375: test_check_rbtree_perf: get_max_min not within log_N/2 +/- 0.5

    Aborted at unit test function
 FAIL - test_check_rbtree_perf
===================================================================
Test suite test_rbtree failed.
===================================================================
PROJECT EXECUTION FAILED
```

**To Reproduce**
Run the test with platform "unit_testing"

**Expected behavior**
Passing test

**Additional context**
I think this test is not properly written. The case is trying to prove that the data structure has O(log2(n)) asymptotic performance. It is doing this by using system time. This is NOT a good way to do this. The right way to do this is to run the scenario with various values of N, and capture the number of operations performed for each run, and show that it is proportional to whatever O() we are trying to prove.

andrewboie, 2020-07-02:
Set to "High" as this is currently breaking CI

andrewboie, 2020-07-02:
@Zhaoningx I need you to revert any test case changes which try to validate O() performance using a timer.

stephanosio, 2020-07-06:
Closing since #26620 is merged

closed 2020-07-06 by stephanosio