zephyrproject-rtos/zephyr #26621: System can't recover after assertion failed in kernel/mem_domain.c

maksimmasalski, 2020-07-02:
**Describe the bug**
I wrote tests for the mem_domain and find out that assertion in line 171 of the `zephyr/kernel/mem_domain.c` makes system unrecoverable after assertion was executed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/zephyrproject-rtos/zephyr/pull/25535
2. Setup my branch
3. Run the memory protection
`west build -t run -p -b qemu_x86 tests/kernel/mem_protect/mem_protect/`
4. See error
5. If you will place `test_mem_part_assert_add_overmax` to run the last one (in that case run it after `test_create_new_invalid_prio_thread_from_user`) in file `tests/kernel/mem_protect/mem_protect/src/main.`c Then test `test_mem_part_assert_add_overmax` will be passed successfully.

**Expected behavior**
I expect that after kernel assertion in user application next parts of the application's code must continue running without any errors

**Impact**
Testing vulnerabilities of the memory protection in Zephyr

**Logs and console output**
```
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../include/spinlock.h:92
        Recursive spinlock 0x0013237c
E: EAX: 0x00000008, EBX: 0x001320a0, ECX: 0x0000000a, EDX: 0x0013237c
E: ESI: 0x00131f80, EDI: 0x00000246, EBP: 0x001862f0, ESP: 0x001862d4
E: EFLAGS: 0x00000097 CS: 0x0008 CR3: 0x00131f60
E: call trace:
E: EIP: 0x0010377b
E:      0x0010a806 (0x131f80)
E:      0x0010a8a4 (0x1320a0)
E:      0x00105d5a (0x1320a0)
E:      0x00101484 (0x10f3f5)
E:      0x00105ea1 (0x18f12c)
E:      0x00106817 (0x1863e4)
E:      0x00103115 (0x0)
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x001321e0 (main)
Caught system error -- reason 4 0
fatal error was unexpected, aborting
FAILED: zephyr/CMakeFiles/run
cd /home/maksim/zephyrproject/zephyr/build && /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 9 -cpu qemu32,+nx,+pae -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -icount shift=5,align=off,sleep=off -rtc clock=vm -kernel /home/maksim/zephyrproject/zephyr/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/maksim/zephyrproject/zephyr/build --target run
```

**Environment (please complete the following information):**
 - OS: Lunix Fedora
 - Toolchain Zephyr SDK 0.11.3
 - maksimmasalski:14_6_add_new_memory_tests
My branch already rebased and contains new master changes from the upstream

maksimmasalski, 2020-07-03:
@andrewboie Could you take a look on that bug issue?

maksimmasalski, 2020-09-07:
Still necessary to have solution for that.

andrewboie, 2020-09-09:
This isn't a bug, it's just currently designed to catch issues using assertions and not return values.
This is a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/24609

closed 2020-09-09 by andrewboie