zephyrproject-rtos/zephyr #26624: Noridc52840 hci_usb bug on linux when " discoverable on " by bluetoothctl

zhaoyangyang12, 2020-07-03:
Hardware: Nordic 52840dk, I compile HCI_USB, and burn to noric52840dk, and plug the hardware into Linux system. Debugging HCI_USB with Bluetoothctl , found that "discoverable on" has been executed, Nordic Bluetooth is not in the discovered mode. I didn't find a device using Android or apple.Obviously, the HCI command corresponding to "discoverable on" has error. Please check it and look forward to your answer.

zhaoyangyang12, 2020-07-08:
hi，Have you verified it?

cvinayak, 2020-07-08:
Zephyr Bluetooth is Low Energy single mode only.

Please provide details as in the issue template (when creating the this issue), details like how to reproduce the issue, command line or tools used etc.

carlescufi, 2020-07-08:
As described before, you are likely not going to be able to make Android work with a Zephyr-based controller because Zephyr's controller only supports Bluetooth Low Energy, and not BR/EDR (or classical) Bluetooth.

closed 2020-07-08 by carlescufi

zhaoyangyang12, 2020-07-13:
> Zephyr Bluetooth is Low Energy single mode only.
> 
> Please provide details as in the issue template (when creating the this issue), details like how to reproduce the issue, command line or tools used etc.

The following is the log I debugged on the virtual machine. After executing the discoverable on command, it can't be scanned by mobile phone or computer.Normal Bluetooth devices do the following operations, can work。

root@zyy-virtual-machine:~# lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 2fe3:000b  //usb device
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
root@zyy-virtual-machine:~# hciconfig 
hci0:	Type: Primary  Bus: USB
	BD Address: FF:55:44:33:22:FE  ACL MTU: 27:7  SCO MTU: 0:0
	UP RUNNING 
	RX bytes:233 acl:0 sco:0 events:19 errors:0
	TX bytes:120 acl:0 sco:0 commands:19 errors:0
root@zyy-virtual-machine:~# hciconfig -a hci0 down
root@zyy-virtual-machine:~# hciconfig 
hci0:	Type: Primary  Bus: USB
	BD Address: FF:55:44:33:22:FE  ACL MTU: 27:7  SCO MTU: 0:0
	DOWN 
	RX bytes:239 acl:0 sco:0 events:20 errors:0
	TX bytes:123 acl:0 sco:0 commands:20 errors:0
root@zyy-virtual-machine:~# hciconfig -a hci0 up
root@zyy-virtual-machine:~# hciconfig 
hci0:	Type: Primary  Bus: USB
	BD Address: FF:55:44:33:22:FE  ACL MTU: 27:7  SCO MTU: 0:0
	UP RUNNING 
	RX bytes:478 acl:0 sco:0 events:40 errors:0
	TX bytes:278 acl:0 sco:0 commands:40 errors:0

root@zyy-virtual-machine:~# bluetoothctl 
[NEW] Controller FF:55:44:33:22:FE zyy-virtual-machine [default]
Agent registered
[bluetooth]# 
advertise       exit            power           system-alias
agent           help            quit            trust
block           info            remove          unblock
connect         list            reset-alias     untrust
default-agent   menu            scan            version
devices         pair            select          
disconnect      pairable        set-alias       
discoverable    paired-devices  show            
[bluetooth]# agent on
Agent is already registered
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# default-agent 
Default agent request successful
[bluetooth]# discoverable on
Changing discoverable on succeeded
[CHG] Controller FF:55:44:33:22:FE Discoverable: yes
[bluetooth]# show
Controller FF:55:44:33:22:FE (public)
	Name: zyy-virtual-machine
	Alias: zyy-virtual-machine
	Class: 0x00000000
	Powered: yes
	Discoverable: yes
	Pairable: yes
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0530
	Discovering: no
[bluetooth]#
[bluetooth]# help
Menu main:
Available commands:
-------------------
advertise                                         Advertise Options Submenu
scan                                              Scan Options Submenu
gatt                                              Generic Attribute Submenu
list                                              List available controllers
show [ctrl]                                       Controller information
select <ctrl>                                     Select default controller
devices                                           List available devices
paired-devices                                    List paired devices
system-alias <name>                               Set controller alias
reset-alias                                       Reset controller alias
power <on/off>                                    Set controller power
pairable <on/off>                                 Set controller pairable mode
discoverable <on/off>                             Set controller discoverable mode
agent <on/off/capability>                         Enable/disable agent with given capability
default-agent                                     Set agent as the default one
advertise <on/off/type>                           Enable/disable advertising with given type
set-alias <alias>                                 Set device alias
scan <on/off>                                     Scan for devices
info [dev]                                        Device information
pair [dev]                                        Pair with device
trust [dev]                                       Trust device
untrust [dev]                                     Untrust device
block [dev]                                       Block device
unblock [dev]                                     Unblock device
remove <dev>                                      Remove device
connect <dev>                                     Connect device
disconnect [dev]                                  Disconnect device
menu <name>                                       Select submenu
version                                           Display version
quit                                              Quit program
exit                                              Quit program
help                                              Display help about this program