zephyrproject-rtos/zephyr #26637: Is there a way to identify sensor device?

timdesi, 2020-07-03:
Hi all, i need to identify/know which device is a sensor with API. Is there a way?

For example below are devices from reel_board listed from sensor shell. Last three of them are sensors.
```
devices:
- clk_k32src
- clk_m16src
- UART_0
- ENTROPY_0
- GPIO_0
- GPIO_1
- I2C_0
- APDS9960
- MMA8652FC
- HDC1008
```
DTS part of reel_board that sensors are defined.

```
arduino_i2c: &i2c0 {
	compatible = "nordic,nrf-twim";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
	sda-pin = <26>;
	scl-pin = <27>;

	mma8652fc@1d {
		compatible = "nxp,fxos8700","nxp,mma8652fc";
		reg = <0x1d>;
		label = "MMA8652FC";
		int1-gpios = <&gpio0 24 GPIO_ACTIVE_LOW>;
		int2-gpios = <&gpio0 25 GPIO_ACTIVE_LOW>;
	};

	ti_hdc@43 {
		compatible = "ti,hdc","ti,hdc1010";
		reg = <0x43>;
		label = "HDC1010";
		drdy-gpios = <&gpio0 22 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
	};

	apds9960@39 {
		compatible = "avago,apds9960";
		reg = <0x39>;
		label = "APDS9960";
		int-gpios = <&gpio0 23 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
	};
};
```

My problem is that calling **sensor_sample_fetch(dev);** function gives fatal error if device is not sensor.

```
starting test - test_sensor_list_devices                                                                                                 
D: OK UART_3                                                                                                                             
E: ***** USAGE FAULT *****                                                                                                               
E:   Illegal use of the EPSR                                                                                                             
E: r0/a1:  0x200021b0  r1/a2:  0x00000032  r2/a3:  0x00000080                                                                            
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00001a57                                                                            
E:  xpsr:  0x20000000                                                                                                                    
E: Faulting instruction address (r15/pc): 0x00000000                                                                                     
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0                                                                                      
E: Current thread: 0x200027e0 (unknown)                                                                                                  
E: Halting system 
```

MaureenHelm, 2020-07-07:
Have a look at how system calls validate the driver type. Or, if you have a known set of sensor drivers, you could do something like this: https://github.com/zephyrproject-rtos/zephyr/blob/41279af8a416e45b26a08550a3fe882d4aabba5a/tests/drivers/counter/counter_basic_api/src/test_counter.c#L29-L72

mbolivar-nordic, 2020-07-08:
> Hi all, i need to identify/know which device is a sensor with API. Is there a way?

In addition to what @MaureenHelm has said, I have to wonder why this is necessary. Would you be able to describe what you are trying to do in some more detail? Perhaps there is a better way than just looping over the devices.

timdesi, 2020-07-09:
First of all thx. @MaureenHelm and @mbolivar-nordic for your answers.

I am trying to develop generic library/firmware that identifies/find sensors and their channels attached to board. Very similar to https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/sensor/sensor_shell.c. After that would collect telemetry data from sensor devices and send it to IoT cloud.

mbolivar-nordic, 2020-07-09:
Thanks for the information.

Have you considered a devicetree-based approach where users can specify a phandle-array property containing references to all the devices they want your library to handle? That would be similar to how the sensor shell allows users to specify devices by name, except an arbitrary number of devices can be specified in a single property.

closed 2020-10-03 by github-actions[bot]