zephyrproject-rtos/zephyr #26643: Nucleo board Slow Code execution at power up - need to always reset

NearbySolutions, 2020-07-03:
I am developing a system based on STM32 SoC using Nucleo board and found that when powering on the board the code is executing slower than normal. When i press the RESET button on the board everything come back to normal and my code is executing normally.

To investigate the issue i decided to execute the Blinking LED sample app delivered with Zephyr code and figured out that the same issue is happening also. The main loop should blink the LED every second but it's blinking a lot slower until i press the reset button. After the reset the LED blink normally.

I decided to debug it and saw that it seems to be related to the irq_unlock code which takes longer. From what i saw so far, it seems to be related to context switching taking really long to execute.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the blinking LED sample folder in zephyr/samples/basic/blinky
2. west build -p auto -b nucleo_f303re .
3. west flash --openocd /usr/local/bin/openocd

**Expected behavior**
The LED on the board should blink every second.

**Impact**
showstopper since i always need to manually reset the board

**Environment (please complete the following information):**
 - OS: MacOS
 - GNU ARM Embedded
 - Last Master branch version

**Additional context**
Add any other context about the problem here.

tagunil, 2020-07-05:
Could you please re-test now with #26036 merged?

NearbySolutions, 2020-07-05:
Hi @tagunil it fixes the problem ! Thanks for your help on that !!
No more latency at the startup

closed 2020-07-05 by NearbySolutions