zephyrproject-rtos/zephyr #26647: build generates unaligned function reference in v2.3

sslupsky, 2020-07-04:
I appear to be observing an odd issue with v2.3 build.  I have encountered a situation where the pointer to an initialization function is wrong.

For example, here is one of the post kernel initialization functions taken from zephyr.lst:
```
static int fs_init(struct device *dev)
{
   10880:	b510      	push	{r4, lr}
	return z_impl_k_mutex_init(mutex);
   10882:	4804      	ldr	r0, [pc, #16]	; (10894 <fs_init+0x14>)
   10884:	f011 ffed 	bl	22862 <z_impl_k_mutex_init>
	list->head = (sys_dnode_t *)list;
   10888:	4b03      	ldr	r3, [pc, #12]	; (10898 <fs_init+0x18>)
   1088a:	601b      	str	r3, [r3, #0]
	list->tail = (sys_dnode_t *)list;
   1088c:	605b      	str	r3, [r3, #4]
	k_mutex_init(&mutex);
	sys_dlist_init(&fs_mnt_list);
	return 0;
}
```

As you can see, the function appears to be properly aligned to a word boundary 0x10880.  However, I noticed when rebasing my code to version 2.3 that it was not starting up.  I used the debugger to trace the execution to the post kernel initialization z_sys_init_run_level().  Here, I noticed that the debugger shows several functions appear to have incorrect pointers in the init_entry struct.  Using the example above, the struct has the following data:
```
init: 0x10881
dev: 0x0
```
Notice that the value of init, which is the pointer to the initialization function above, is incorrect (it has a one byte offset).  I went back and checked the pre_kernel initialization and I see a similar issue.

In many cases it doesn't cause a fault but eventually it does and the mcu locks up.

Here is a screenshot of the debug watch points taken just before the initialization call that causes the fault:
[image]
In this case, zephyr.lst shows the address of net_init() is 0x20568:
```
static int net_init(struct device *unused)
{
   20560:	b510      	push	{r4, lr}

	l3_init();

	net_mgmt_event_init();

	init_rx_queues();
   20562:	f7ff fff7 	bl	20554 <init_rx_queues>
```
I noticed some recent changes to the device structure.  Could there be some issue related to the recent changes causing this?

stephanosio, 2020-07-04:
> In many cases this issue does not cause a fault but eventually it does and the mcu locks up.

Are you sure unaligned address is actually the cause? (well it is not actually unaligned)

The LSB being set to 1 just means that the instruction at the 2-byte boundary of the corresponding address is to be executed in Thumb mode, which must be case for all Cortex-M cores.

sslupsky, 2020-07-04:
I was not aware of your statement regarding Thumb mode.  Indeed, this is a build for a sam0 mcu.  I have noticed situations where the pointer is offset by 2 bytes.  Is this another Thumb issue?

Notwithstanding the above Thumb mode, the last example I cited above is causing a fault so something is wrong.

sslupsky, 2020-07-04:
Thank you for the thumb info ... I looked this up not the ARM website and I understand now why this happens.

Must be some asynchronous event going on ... I'll dig deeper.

closed 2020-07-04 by sslupsky