zephyrproject-rtos/zephyr #26648: Data compression module integration

parthitce, 2020-07-04:
**Is your feature request related to a problem? Please describe.**
Currently Zephyr doesn't have any data compression support. Data compression interface enables the efficiency in communication with cloud platforms, OTA and various other use cases.

This issues is opened to discuss the possible ways to integrate data compression for Zephyr.

**Describe the solution you'd like**
Integrating available data compression library as external modules. But considering multiple compression algorithms and available open source packages with compatible licenses like lz4. zstd, should zephyr include one external module (e.g named as data compression/compression) and add all the compression modules as sub-modules or with custom directory structure?

**Sample**
We tried to compile and use lz4. Find the reference below,

1. https://github.com/linumiz/zephyr/commit/3c56fce5af166ffe6b52b96d9a43f822274fee48
2. https://github.com/linumiz/lz4/commit/22ecbb892dc05deec0d0858abb3cdbfc85e68ec0

Note:
- [lz4 library](https://github.com/lz4/lz4/blob/dev/LICENSE) is BSD 2-Clause license and remaining directory/content is GPLv2
- [zstd](https://github.com/facebook/zstd/blob/dev/LICENSE) is BSD License

carlescufi, 2020-07-08:
> Integrating available data compression library as external modules. But considering multiple compression algorithms and available open source packages with compatible licenses like lz4. zstd, should [...]

My feeling is that we should have one repo per upstream project approximately. Meaning that if lz4 and zstd are two different projects we probably want to have two different repositories, one for each. But the modules policies are currently undergoing rework, so there's no definite answer to this.

stephanosio, 2020-07-09:
I wonder if it would make more sense to integrate them into a new "compression" subsystem with common API for using different compression algorithms; after all, most compression algorithm implementations are usually just one or two files, and they can be considered an integral part of the operating system in terms of IoT.

parthitce, 2020-07-09:
> I wonder if it would make more sense to integrate them into a new "compression" subsystem with common API for using different compression algorithms; after all, most compression algorithm implementations are usually just one or two files, and they can be considered an integral part of the operating system in terms of IoT (cc @pabigot).

@stephanosio Does that mean to implement from scratch or copy the existing implementation into Zephyr? In the later case, can we copy BSD licensed code into Zephyr, AFAIK it's not possible?

pabigot, 2020-07-09:
Based on existing practice I believe libraries of this type should be added as external modules via west via a PR for each such library.  I assume they'd have to be exposed as a CMake library to be used in Zephyr.  This is consistent with @carlescufi's comment above except I don't know anything about how module policy is being revised.

It may be that a generic compression abstraction API is useful, but we should delay creating one until there's evidence that people want to be able to select a compression algorithm as a configurable feature rather than based on specific needs.  I have doubts of its value.  If abstraction has a clear use case then use the APIs of the algorithms involved will inform the design of the generic API.

IMO code developed and maintained externally in a self-contained project should not be copied into Zephyr, even if it's one or two files of implementation (what about tests? documentation? support contact information?).  The maintenance burden is too high.

parthitce, 2020-09-06:
> But the modules policies are currently undergoing rework, so there's no definite answer to this.

@carlescufi Do we have new/updated standardization/process in place for the external modules? Does it makes sense to open an RFC with external module request?

carlescufi, 2020-09-18:
> > But the modules policies are currently undergoing rework, so there's no definite answer to this.
> 
> @carlescufi Do we have new/updated standardization/process in place for the external modules? Does it makes sense to open an RFC with external module request?

@parthitce 
It's almost complete, but I'd say it does make sense to open an RFC at this point, yes. Most of the process to actually create a new module repo is unchanged.

parthitce, 2021-06-12:
Feasibility of zstd is planned next. Re-opening to address zstd.

carlescufi, 2021-12-20:
Closing this for now @parthitce. Please create a new one if you plan to add a new module.

PR #28536 (2021-06-12): samples: compression: lz4: lz4 sample application
This PR add sample application for minimal lz4 library.

lz4 library RFC requested here: #28535

Fixes: #26648
commit c9aa260f (2021-06-12): samples: compression: lz4: lz4 sample application

This PR add sample application for minimal lz4 library.

lz4 library RFC requested here: #28535

Fixes: #26648
closed 2021-12-20 by carlescufi