zephyrproject-rtos/zephyr #26651: Updatehub: frdm_k64f resets in a loop

parthitce, 2020-07-05:
**Describe the bug**
Updatehub with latest mainline master resets in loop on FRDM K64F target. 

**To Reproduce**
Steps to reproduce the behavior:
1. Flashed mcuboot with ```west build -b frdm_k64f -s ./boot/zephyr && west flash```
2. Adjust overlay-prj.conf.example for docker instance running locally
3. ```west build -b frdm_k64f -d ./2.0 -- -DCONF_FILE="prj.conf overlay-prj.conf.example"```
3. Sign with ```west sign -t imgtool -- --key ../../../../mcuboot/root-rsa-2048.pem```
4. Change updatehub code with additional print ```========================>> Updated version```
5. Create package as described in [here](https://docs.zephyrproject.org/2.1.0/samples/net/updatehub/README.html).
6. Upload and test with ```updatehub run```
7. Observe target resets in a loop. Mcuboot didn't jump into revert mode as well

**Expected behavior**
Updatehub works and cloud instance updates the status

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.3.0-809-g2cc2d1e5511a  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.007,000] <inf> mcuboot: Swap type: none
[00:00:00.007,000] <err> mcuboot: Unable to find bootable image
*** Booting Zephyr OS build zephyr-v2.3.0-809-g2cc2d1e5511a  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.007,000] <inf> mcuboot: Swap type: none
*** Booting Zephyr OS build zephyr-v2.3.0-809-g2cc2d1e5511a  ***

[00:00:03.001,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:11.010,000] <inf> main: UpdateHub sample app started
[00:00:11.010,000] <inf> main: Confirming the boot image
[00:00:11.011,000] <inf> main: Starting UpdateHub polling mode
[00:00:11.031,000] <inf> updatehub: No update available
[00:01:11.048,000] <inf> updatehub: No update available
[00:02:11.064,000] <inf> updatehub: No update available
uart:~$ 
uart:~$ updatehub run 
Starting UpdateHub run...
[00:02:42.185,000] <inf> updatehub: Probe metadata received
[00:02:43.776,000] <inf> updatehub: Firmware downloaded successfully
[00:02:43.805,000] <inf> updatehub: Image flashed successfully, you can reboot now
uart:~$ kernel reboot cold *** Booting Zephyr OS build zephyr-v2.3.0-809-g2cc2d1e5511a  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.007,000] <inf> mcuboot: Swap type: test
*** Booting Zephyr OS build zephyr-v2.3.0-809-g2cc2d1e5511a  ***

[00:00:04.001,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:06.009,000] <inf> main: ========================>> Updated version
[00:00:06.009,000] <inf> main: Confirming the boot image
[00:00:06.009,000] <inf> main: Starting UpdateHub polling mode
[00:00:06.016,000] <inf> updatehub: Probe metadata received
uart:~$ *** Booting Zephyr OS build zephyr-v2.3.0-809-g2cc2d1e5511a  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=good, swap_type=0x2, copy_done=0x1, image_ok=0x1
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: none
[00:00:00.006,000] <inf> mcuboot: Swap type: test
*** Booting Zephyr OS build zephyr-v2.3.0-809-g2cc2d1e5511a  ***

[00:00:04.001,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:09.007,000] <inf> main: ========================>> Updated version
[00:00:09.007,000] <inf> main: Confirming the boot image
[00:00:09.007,000] <inf> main: Starting UpdateHub polling mode
[00:00:09.016,000] <inf> updatehub: Probe metadata received
uart:~$ *** Booting Zephyr OS build zephyr-v2.3.0-809-g2cc2d1e5511a  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=good, swap_type=0x2, copy_done=0x1, image_ok=0x1
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: none
[00:00:00.006,000] <inf> mcuboot: Swap type: test
*** Booting Zephyr OS build zephyr-v2.3.0-809-g2cc2d1e5511a  ***
```

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: Zephyr HEAD: c001baafc8e408829adf745e9f28efebf25bdaa5 and mcuboot HEAD: [d854bb6659755e4d550513f8df3f212beb5338cb](https://github.com/JuulLabs-OSS/mcuboot/commit/d854bb6659755e4d550513f8df3f212beb5338cb)

- Stopping the updatehub cloud UI, stops the reset loop.

nandojve, 2020-07-05:
CC: @otavio

nandojve, 2020-07-05:
Hi @parthitce,

Is the new image confirming itself as an good image, first time it boots?
This is an MCUboot procedure that you currently need implement.
You can validate using below snip.
```
#include <zephyr.h>
#include <sys/printk.h>

#include <dfu/mcuboot.h>

void main(void)
{
	int ret = -1;

	printk("Hello World! 2 %s\n", CONFIG_BOARD);

	/* The image of application needed be confirmed */
	printk("Confirming the boot image\n");
	ret = boot_write_img_confirmed();
	if (ret < 0) {
		printk("Error to confirm the image\n");
	} else {
		printk("Image is OK\n");
	}
}
```

parthitce, 2020-07-05:
@nandojve Yes, that does work. But Updatehub UI waits for the device to confirm the update. So I used the same updatehub sample (with minor diff) again to check the status, which result in reset loop.

nandojve, 2020-07-07:
Hi @parthitce, I made a clear MCUboot + UpdateHub (v: 1.0.0) and flash frdm-k64f.
After that, create an package with same UpdateHub (v: 2.0.0) to test Update and it is working without any problem.

```
*** Booting Zephyr OS build zephyr-v2.3.0-836-g2eb633d12ed9  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.007,000] <inf> mcuboot: Swap type: none
*** Booting Zephyr OS build zephyr-v2.3.0-836-g2eb633d12ed9  ***

[00:00:03.001,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:07.261,000] <inf> main: UpdateHub sample app started
[00:00:07.261,000] <inf> main: Confirming the boot image
[00:00:07.261,000] <inf> main: Starting UpdateHub polling mode
[00:00:09.262,000] <err> updatehub: Could not receive data
uart:~$ updatehub info 
Unique device id: ffffffff4e454130000b000f
Firmware Version: 1.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
UpdateHub Server: 192.168.10.199
[00:01:11.264,000] <err> updatehub: Could not receive data
# Register device on UpdateHub-CE
uart:~$ updatehub run 
Starting UpdateHub run...
No update found
[00:01:23.202,000] <inf> updatehub: No update available
[00:02:11.277,000] <inf> updatehub: No update available
[00:03:11.282,000] <inf> updatehub: No update available
# Run UpdateHub-CE test for new image
uart:~$ updatehub run 
Starting UpdateHub run...
[00:04:06.822,000] <inf> updatehub: Probe metadata received
[00:04:08.442,000] <inf> updatehub: Firmware downloaded successfully
[00:04:08.461,000] <inf> updatehub: Image flashed successfully, you can reboot now
[00:04:11.285,000] <inf> updatehub: Probe metadata received
uart:~$ *** Booting Zephyr OS build zephyr-v2.3.0-836-g2eb633d12ed9  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.007,000] <inf> mcuboot: Swap type: test
*** Booting Zephyr OS build zephyr-v2.3.0-836-g2eb633d12ed9  ***

[00:00:03.001,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:06.005,000] <inf> main: UpdateHub sample app started
[00:00:06.005,000] <inf> main: Confirming the boot image
[00:00:06.006,000] <inf> main: Starting UpdateHub polling mode
[00:00:06.023,000] <inf> updatehub: No update available
uart:~$ updatehub info 
Unique device id: ffffffff4e454130000b000f
Firmware Version: 2.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
UpdateHub Server: 192.168.10.199
uart:~$ 
```
OS: Linux Debian-10
Toolchain: Zephyr SDK 0.11.3
Zephyr HEAD: 2eb633d12e - net: lwm2m: Reset only messages owned by lwm2m context (3 hours ago) <Robert Lubos>
MCUBoot HEAD: d52aff5065ab5995f785877a834112461ff93526

Procedure:
```
git fetch upstream
git rebase upstream/master
west update
git diff
diff --git a/samples/net/updatehub/prj.conf b/samples/net/updatehub/prj.conf
index 603ea45e41..1d5afa2259 100644
--- a/samples/net/updatehub/prj.conf
+++ b/samples/net/updatehub/prj.conf
@@ -18,3 +18,8 @@ CONFIG_UPDATEHUB_SHELL=y
 # Debug helpers
 CONFIG_LOG=y
 CONFIG_UPDATEHUB_LOG_LEVEL_INF=y
+
+CONFIG_UPDATEHUB_PRODUCT_UID="e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f"
+CONFIG_UPDATEHUB_POLL_INTERVAL=1
+CONFIG_UPDATEHUB_CE=y
+CONFIG_UPDATEHUB_SERVER="<your UpdateHub-CE local host IP"
```

How sign image 1:
`/home/gfbudke/zephyros/mainline/bootloader/mcuboot/scripts/imgtool.py sign --key /home/gfbudke/zephyros/mainline/bootloader/mcuboot/root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.0.0 --slot-size 0x60000 /home/gfbudke/zephyros/mainline//zephyr/build/zephyr/zephyr.bin /home/gfbudke/zephyros/mainline/zephyr/build/zephyr/zephyr-1.0.0.bin`

How sign image 2:
`/home/gfbudke/zephyros/mainline/bootloader/mcuboot/scripts/imgtool.py sign --key /home/gfbudke/zephyros/mainline/bootloader/mcuboot/root-rsa-2048.pem --header-size 0x200 --align 8 --version 2.0.0 --slot-size 0x60000 /home/gfbudke/zephyros/mainline//zephyr/build/zephyr/zephyr.bin /home/gfbudke/zephyros/mainline/zephyr/build/zephyr/zephyr-2.0.0.bin`

Step 8: Create a package with UpdateHub Utilities (uhu):
```
pip3 install --user uhu
uhu product use "e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f"
uhu package add build/zephyr/zephyr-2.0.0.bin -m zephyr
uhu package version 2.0.0
uhu package archive --output build/zephyr/zephyr-2.0.0.pkg
```
Could you give another try. I'm thinking you may missed some step.

parthitce, 2020-07-12:
@nandojve I missed the versioning when using `west sign` and it looks mandatory. Sorry for the noise. Closing this issue.

closed 2020-07-12 by parthitce