zephyrproject-rtos/zephyr #26656: SAM0 USB transfer slot leak

stephanosio, 2020-07-06:
**Describe the bug**
"No transfer slot available" error message is displayed and USB transfer stops working when repeatedly unplugging and re-plugging the USB cable on the SAM0-family SoCs.

This bug has been confirmed on the SAM D21 and E54 SoCs using `atsamd21_xpro` and `atsame54_xpro`, respectively.

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash `samples/subsys/usb/cdc_acm`
2. Unplug and re-plug the USB device cable for 4 times.

**Expected behavior**
USB transfer slots are properly cleaned up.

**Impact**
USB is unusable.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.3.0-809-g897b5bf60a12  ***
[00:00:00.005,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.431,000] <inf> usb_cdc_acm: USB device configured
<REPLUG>
[00:00:13.350,000] <inf> usb_cdc_acm: USB device configured
<REPLUG>
[00:00:16.301,000] <inf> usb_cdc_acm: USB device configured
<REPLUG>
[00:00:19.575,000] <inf> usb_cdc_acm: USB device configured
<REPLUG>
[00:00:26.076,000] <err> usb_transfer: No transfer slot available
[00:00:26.076,000] <inf> usb_cdc_acm: USB device configured
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.3
 - Commit SHA: 897b5bf60a12e71976df6885817aff04b09a33ea

stephanosio, 2020-07-06:
cc @jfischer-phytec-iot @nzmichaelh @nandojve

jfischer-phytec-iot, 2020-07-06:
The bug is also reported here  #21234 and has an PR #24402

MaureenHelm, 2020-07-07:
Duplicate of #21234

stephanosio, 2020-07-07:
This issue is clearly not a duplicate of #21234; at most, it is related.

carlescufi, 2020-09-09:
@jfischer-phytec-iot can you take a look?

closed 2021-02-07 by github-actions[bot]