zephyrproject-rtos/zephyr #26658: Need Support for AoA/AoD Bluetooth feature in Zephyr

danielkc26, 2020-07-06:
**Is your feature request related to a problem? Please describe.**
I would like to develop AoA/AoD application using nordic nrf52833 DK board in Zephyr framework

**Describe the solution you'd like**
Support for this feature is expected from Zephyr 

@zephyrbot

aftabahmed59, 2020-11-23:
Any updates on when is it expected to be released ?

jhedberg, 2024-12-16:
@alwa-nordic this is already supported, isn't it?

alwa-nordic, 2024-12-18:
Yes, I think so. We have the following samples:

 - https://docs.zephyrproject.org/latest/samples/bluetooth/direction_finding_peripheral/README.html
 - https://docs.zephyrproject.org/latest/samples/bluetooth/direction_finding_central/README.html
 - https://docs.zephyrproject.org/latest/samples/bluetooth/direction_finding_connectionless_tx/README.html
 - https://docs.zephyrproject.org/latest/samples/bluetooth/direction_finding_connectionless_rx/README.html

I will close this ticket as completed.

closed 2024-12-18 by alwa-nordic