zephyrproject-rtos/zephyr #26659: unexpected behavior of ARCv2 memset in NEWLIB_LIBC with -Os

wentongwu, 2020-07-06:
```
        printk("-- PWRCTL = 0x%08x\n", sys_read32(PWRCTL));
50030c70:       40c3 5004 31b0          mov_s   r0,0x500431b0
50030c76:       45cb f000 1d10          mov_s   r13,0xf0001d10
50030c7c:       0d1e f94f               bl      -53988  ;50023998 <printk>
50030c80:       1500 1001               ld      r1,[r13]
        printk("++ CLKCTL = 0x%08x\n", sys_read32(CLKCTL));
50030c84:       40c3 5004 31d0          mov_s   r0,0x500431d0
50030c8a:       0d12 f94f               bl      -54000  ;50023998 <printk>
        __asm__ volatile("ld    %1, %0\n"
50030c8e:       1500 1010               ld      r16,[r13]
50030c92:       2050 2050               bclr    r16,r16,0x1
50030c96:       1d00 1400               st      r16,[r13]
        __asm__ volatile("ld%U1 %0, %1;\n\t"
50030c9a:       1500 1001               ld      r1,[r13]
        printk("-- CLKCTL = 0x%08x\n", sys_read32(CLKCTL));
50030c9e:       40c3 5004 31f0          mov_s   r0,0x500431f0
50030ca4:       0cf6 f94f               bl      -54028  ;50023998 <printk>
        msg_req->msg.type = LARGE_CONFIG_GET;
50030ca8:       d843                    mov_s   r0,0x43
        memset(msg_req, 0, sizeof(msg_req_hdr_t));
50030caa:       1c08 3400               st      r16,[sp,8]
50030cae:       1c0c 3400               st      r16,[sp,12]
        msg_req->msg.type = LARGE_CONFIG_GET;
50030cb2:       1c0b 3002               stb     r0,[sp,11]
        msg_req->msg.final_block = 1 ;
50030cb6:       d830                    mov_s   r0,0x30
50030cb8:       0a32 ffef               bl.d    -1488   ;500306e8 <z_arch_is_user_context>
50030cbc:       1c0f 3002               stb     r0,[sp,15]
        LOG_INF("%s", __func__);
```
this is a fragment of my code, intends to do memset for a local variable which locates at current stack [sp,8] and [sp,12] as instruction `50030caa` and `50030cae` show, and it seems compiler think the `r16` is `0` on instruction `50030caa` and `50030cae`, but before that, the value of `r16` is first loaded from `[r13]` with `r13 = 0xf0001d10`, and then do `bclr    r16,r16,0x1`, I don't know why compiler assumes `r16` is `0` at current situation and it caused wrong behavior in my code. 

@vonhust could you please help take a look at this? Did I miss something?

wentongwu, 2020-07-06:
@andrewboie

wentongwu, 2020-07-06:
@abrodkin any suggestion? Thanks

abrodkin, 2020-07-06:
@wentongwu could you please provide instructions on how to reproduce that problem? I.e. which application and how should I compile to get that assembly? From what we currently have here it's not clear why there's no call to `memset()` via `bl` or `bl.d` instruction but instead in-place `st`.

wentongwu, 2020-07-07:
@abrodkin I have no idea about how to reproduce it although I tried a lot with self developed similar code. It can only be reproduced when built along with many other files which I can't share(sorry for that). And BTW sizeof(msg_req_hdr_t) = 8.

stephanosio, 2020-07-07:
> From what we currently have here it's not clear why there's no call to `memset()` via `bl` or `bl.d` instruction but instead in-place `st`.

likely an inline loop expansion optimisation, but I am completely lost as to why you get that with `-Os` (optimise for size) enabled because that is one thing you never want when you are optimising for size.

wentongwu, 2020-07-07:
> likely an inline loop expansion optimisation, but I am completely lost as to why you get that with `-Os` (optimise for size) enabled because that is one thing you never want when you are optimising [...]

you mean there is another flag to enable inline loop expansion optimization?

abrodkin, 2020-07-07:
@wentongwu may I have some more data here, in particular:
1. C source of that snippet. I understand you cannot share everything but at least this function.
2. Full command-line used to compile that source into object file
3. Compiler version (whatever you get with `arc-zephyr-elf-gcc -v`)

wentongwu, 2020-07-07:
> Full command-line used to compile that source into object file

FLAGS = -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros output/zephyr_1.14.0/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-common -mcpu=arcem -Werror -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -g3 -gdwarf-2 -fno-delete-null-pointer-checks -mlittle-endian -mno-swap -mnorm -mbarrel-shifter -mno-sdata -std=c99

> Compiler version (whatever you get with arc-zephyr-elf-gcc -v)

arc-zephyr-elf-gcc (crosstool-NG 1.24.0-rc2-dirty) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

> C source of that snippet. I understand you cannot share everything but at least this function.

thanks, but need some time...

stephanosio, 2020-07-07:
> arc-zephyr-elf-gcc (crosstool-NG 1.24.0-rc2-dirty) 8.3.0
> Copyright (C) 2018 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions. There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Can you check if this issue is reproducible with the latest version of Zephyr SDK (that has GCC 9.2.0)?

wentongwu, 2020-07-07:
> Can you check if this issue is reproducible with the latest version of Zephyr SDK (that has GCC 9.2.0)?

will try.

stephanosio, 2020-07-07:
>         memset(msg_req, 0, sizeof(msg_req_hdr_t));
> 50030caa:       1c08 3400               st      r16,[sp,8]
> 50030cae:       1c0c 3400               st      r16,[sp,12]

This makes even less sense given that `st 0,[sp,imm]` can be encoded in the ARC ISA.

I wrote a test code and can see that the compiler generates valid instructions (tested with GCC 9.2.0):
```
void main(void)
{
80400a08:       c1a2                    sub_s   sp,sp,0x8
        struct rubbish_struct rubbish;

        __builtin_memset(&rubbish, 0, sizeof(rubbish));
80400a0a:       1c00 3001               st      0,[sp]
80400a0e:       1c04 3001               st      0,[sp,4]
```

claziss, 2020-07-07:
If u use upstream gcc8.x to build ARC compiler, it may be a problem as it uses an old register allocation algorithm which may lead to unexpected behaviour. This may explain or not the observed behaviour, however, my recommendation is to upgrade your compiler to gcc9.x or gcc10.x for improved stability and performance.

wentongwu, 2020-07-07:
> If u use upstream gcc8.x to build ARC compiler, it may be a problem as it uses an old register allocation algorithm which may lead to unexpected behaviour. This may explain or not the observed [...]

Tool chain is from Zephyr SDK (version 0.10.3), on which not sure there is the problem you described, but anyway I will try latest version of Zephyr SDK.

wentongwu, 2020-07-08:
```
        printk("-- PWRCTL = 0x%08x\n", sys_read32(PWRCTL));
50030d12:       40c3 5004 36e8          mov_s   r0,0x500436e8
50030d18:       45cb f000 1d10          mov_s   r13,0xf0001d10
50030d1e:       0d9a f94f               bl      -53864  ;50023ab4 <printk>
50030d22:       1500 1001               ld      r1,[r13]
        printk("++ CLKCTL = 0x%08x\n", sys_read32(CLKCTL));
50030d26:       40c3 5004 3708          mov_s   r0,0x50043708
50030d2c:       0d8a f94f               bl      -53880  ;50023ab4 <printk>
        __asm__ volatile("ld    %1, %0\n"
50030d30:       1500 1010               ld      r16,[r13]
50030d34:       2050 2050               bclr    r16,r16,0x1
50030d38:       1d00 1400               st      r16,[r13]
        __asm__ volatile("ld%U1 %0, %1;\n\t"
50030d3c:       1500 1001               ld      r1,[r13]
        printk("-- CLKCTL = 0x%08x\n", sys_read32(CLKCTL));
50030d40:       40c3 5004 3728          mov_s   r0,0x50043728
50030d46:       0d72 f94f               bl      -53904  ;50023ab4 <printk>
        msg_req->msg.type = LARGE_CONFIG_GET;
50030d4a:       d843                    mov_s   r0,0x43
        memset(msg_req, 0, sizeof(msg_req_hdr_t));
50030d4c:       1c08 3001               st      0,[sp,8]
50030d50:       1c0c 3001               st      0,[sp,12]
        msg_req->msg.type = LARGE_CONFIG_GET;
50030d54:       1c0b 3002               stb     r0,[sp,11]
        msg_req->msg.final_block = 1 ;
50030d58:       d830                    mov_s   r0,0x30
50030d5a:       1c0f 3002               stb     r0,[sp,15]
50030d5e:       0a86 ffcf               bl      -1404   ;500307e0 <z_arch_is_user_context>
        LOG_INF("%s", __func__);
```

great, it seems right with 9.2.0, thanks everyone's(@abrodkin @stephanosio @claziss ) help.  

@claziss is there any link about `it may be a problem as it uses an old register allocation algorithm which may lead to unexpected behaviour`

claziss, 2020-07-08:
@wentongwu no, but u can consider the above statement as the link, as I am the ARC GNU compiler maintainer.

wentongwu, 2020-07-09:
> @wentongwu no, but u can consider the above statement as the link, as I am the ARC GNU compiler maintainer.

@claziss Thanks! And the mentioned algorithm `old register allocation algorithm` is dedicated for arc arch or it's generic for any platform? Thanks