zephyrproject-rtos/zephyr #26676: MDB runner is not capturing real board's output

abrodkin, 2020-07-06:
With addition of MDB runner (see https://github.com/zephyrproject-rtos/zephyr/pull/24806) it's now possible to run Zephyr binaries on both ARC boards and nSIM simulator via MetaWare debugger. But while it works OK-ish (see https://github.com/zephyrproject-rtos/zephyr/issues/26272) with manual execution and to some extent with `sanitycheck` script with simulated `nsim_*` targets, attempt to use `sanitycheck` script with ARC boards results on all failed tests.

That's because MDB runner is based on `ZephyrBinaryRunner` (see https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/west_commands/runners/mdb.py#L13) which listens on `stdout` of the spawned process. But in case of real board connected via UART we need to act more like OpenOCD runner listening on the specified serial port.

Obviously there're 2 options on how to solve this:
1. Restrict usage/disable MDB runner for real boards keeping it only for simulation (especially `nsim_hs_smp`). 
2. Add missing back-end to the runner for connection to the serial port

But looking forward with addition of MetaWare toolchain support (see https://github.com/zephyrproject-rtos/zephyr/pull/22668) we'd like to have an ability to use MetaWare tools for everything (both building and running) w/o any extra tools, so having MDB runner capable of running Zephyr on real targets in `sanitycheck` could be a very nice addition. So I'd prefer the second option to be implemented.

IRISZZW, 2020-09-07:
I believe the root cause is in the sanity check script, not in MDB runner. 
MDB and openocd runner both not listening on the specified serial port.

There are two basic Handlers in sanity check script:[BinaryHandler](https://github.com/zephyrproject-rtos/zephyr/blob/016b5422d90e9d3405eda22da4fda23d6a13a415/scripts/sanity_chk/sanitylib.py#L330) and [DeviceHandler](https://github.com/zephyrproject-rtos/zephyr/blob/016b5422d90e9d3405eda22da4fda23d6a13a415/scripts/sanity_chk/sanitylib.py#L471).

DeviceHandler will listen on the specified serial port, while BinaryHandler will listen on the STD.

https://github.com/zephyrproject-rtos/zephyr/blob/016b5422d90e9d3405eda22da4fda23d6a13a415/scripts/sanity_chk/sanitylib.py#L2053-L2063

How to run sanitycheck script with ARC boards with mdb runner:

` ./scripts/sanitycheck -p iotdk -O build.sanity.iotdk  --device-testing --device-serial /dev/serial-board-251642517567 -T tests/kernel/common --west-runner=mdb --west-flash=--dig-device=IoTDK`

How to run sanitycheck script with nsim with mdb runner:

comment these lines and run sanitycheck as usual.
https://github.com/zephyrproject-rtos/zephyr/blob/016b5422d90e9d3405eda22da4fda23d6a13a415/boards/arc/nsim/board.cmake#L4-L7

abrodkin, 2020-09-07:
@evgeniy-paltsev this piece of info should be merged in a greater MetaWare-related documentation section in say https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html as suggested by @tejlmand.

closed 2020-09-07 by abrodkin