zephyrproject-rtos/zephyr #26685: sanitycheck "--only-failed" is broken

andrewboie, 2020-07-06:
**Describe the bug**
```
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/net/utils/net.util/handler.log
INFO    - Total complete:  374/ 374  100%  skipped:   20, failed:   18
INFO    - 336 of 354 tests passed (94.92%), 18 failed, 20 skipped with 0 warnings in 319.59 seconds
INFO    - In total 856 test cases were executed on 1 out of total 260 platforms (0.38%)
INFO    - 231 tests executed on platforms, 143 tests were only built.
apboie@apboie-mobl4:~/projects/zephyr/zephyr (1) mmu-x86-work /home/apboie/projects/zephyr/zephyr
$ scr -p qemu_x86 --only-failed
Keeping artifacts untouched
INFO    - JOBS: 48
INFO    - 3 test configurations selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:    1/   3  33%  skipped:    0, failed:    0
ERROR   - qemu_x86                  tests/kernel/mem_protect/protection/kernel.memory_protection.protection FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/mem_protect/protection/kernel.memory_protection.protection/handler.log
INFO    - Total complete:    3/   3  100%  skipped:    0, failed:    1
INFO    - 2 of 3 tests passed (66.67%), 1 failed, 0 skipped with 0 warnings in 12.30 seconds
INFO    - In total 856 test cases were executed on 1 out of total 260 platforms (0.38%)
INFO    - 3 tests executed on platforms, 0 tests were only built.
```

Above see how 18 tests failed, but when I called again with "--only-failed" only 3 tests were selected!

**Expected behavior**
All failing scenarios re-built/re-run

**Impact**
This is seriously annoying for anyone who is using sanitycheck to validate their code, the workaround is a full build which takes a lot longer

andrewboie, 2020-07-06:
It looks like any test that failed the *build* and not the emulator is not included.

nashif, 2020-07-07:
> It looks like any test that failed the _build_ and not the emulator is not included.

yes, we stopped rebuilding failed tests (those with build failures) because it does not make sense in CI but I see where this can still be useful for developers. Will see how we can continue to support this with --only-failed and maintain new behavior with --retry-failed.

PR #26775 (2020-07-09): sanitycheck: do not retry build errors with --retry-failed
restore how --only-failed works by allow rebuilds in case of build
errors, however, do not rebuild when --retry-failed is used, which is a
CI usecase and nothing would change in the second iteration.

Fixes #26685
commit dc43c298 (2020-07-09): sanitycheck: do not retry build errors with --retry-failed

restore how --only-failed works by allow rebuilds in case of build
errors, however, do not rebuild when --retry-failed is used, which is a
CI usecase and nothing would change in the second iteration.

Fixes #26685
closed 2020-07-09 by nashif