zephyrproject-rtos/zephyr #26689: Couldn't get test result from serial on up_squared board.

chen-png, 2020-07-07:
**Describe the bug**
running some tests on up_squared, **the console log will stop at the "Booting Zephyr OS" step**, and couldn't get test result.

git bisect and found the first bad commit is  928d3112.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/common/
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**

***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v2.3.0-809-g897b5bf60a12  (delayed boot 500ms) ***

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.4
 - Commit ID: 897b5bf60a

wentongwu, 2020-07-07:
seems x86_64 is broken, see also https://buildkite.com/zephyr/zephyr/builds/2229

dcpleung, 2020-07-07:
This should be fixed by #26687.

chen-png, 2020-07-08:
fixed by #26687

closed 2020-07-08 by chen-png