zephyrproject-rtos/zephyr #26695: net: TCP2: connect() returns 0 without waiting for handshake completion

a8961713, 2020-07-07:
**Describe the bug**
When using `CONFIG_NET_TCP2=y`:
A call to BSD `connect()` or `zsock_connect()` returns 0 immediately after sending SYN, without waiting for TCP handshake completion. Specifically it does not wait for [SYN,ACK] reply from server.
That leads to client assuming connection is established. Immediately sending data after `connect()`, results with an error -128 (ENOTCONN).

**To Reproduce**
The bug can be reproduced on MQTT sample at path: samples/net/mqtt_publisher.
The sample uses `mqtt_connect()`, which calls `connect()`  followed immediately by writing to the socket (`send()`).
This works with `CONFIG_NET_TCP1=y`, but fails with `CONFIG_NET_TCP2=y`.

The bug was reproduced on the following boards:
- stm32f746g_disco (using the on board ethernet)
- nucleo_l432kc (using SLIP over usart1)

**Expected behavior**
`connect()` should return with 0 only after completing the TCP handshake, i.e. receiving [SYN,ACK].

**Impact**
This is a showstopper. You should be able to send immediately after `connect()` returns successfully.

**Environment (please complete the following information):**
 - OS: Linux, Ubuntu 18.04
 - Toolchain: Zephyr SDK

a8961713, 2020-07-07:
**Additional remark**
`connect()` succeeds (returns 0) even when there is no listening server.

rveerama1, 2020-07-07:
Check with this PR https://github.com/zephyrproject-rtos/zephyr/pull/26098

a8961713, 2020-07-07:
> Check with this PR #26098

Great! This is what we need.

MaureenHelm, 2020-07-07:
> Great! This is what we need.

Can you confirm it fixes this issue?

a8961713, 2020-07-08:
> > Great! This is what we need.
> 
> Can you confirm it fixes this issue?

Yes, it does.

PR #26098 (2020-07-20): tcp2: Implement a blocking connect
In order to implement a blocking connect, add a semaphore
and block on it in net_tcp_connect().

The semaphore is released when ESTABLISHED state is reached.

In case tcp_conn_unref() is called while waiting on the semaphore,
defer the unreference, tcp_conn_unref() will be called from
net_tcp_connect().

Fixes #26695
closed 2020-07-20 by jukkar