zephyrproject-rtos/zephyr #26700: waveshare_open103z board can't build tests/mem_protect

maksimmasalski, 2020-07-07:
**Describe the bug**
I find out that waveshare_open103z can't build some tests from `zephyr/tests/mem_protect`. There is list of build results:
1. zephyr/tests/mem_protect/futex Build error
2. zephyr/tests/mem_protect/mem_protect Build error
3. zephyr/tests/mem_protect/obj_validation Build error
4. zephyr/tests/mem_protect/protection Build success
5. zephyr/tests/mem_protect/stackprot Build success
6. zephyr/tests/mem_protect/stack_random Build success
7. zephyr/tests/mem_protect/syscalls Build error
8. zephyr/tests/mem_protect/sys_sem Build success
9. zephyr/tests/mem_protect/userspace Build error

**To Reproduce**
Steps to reproduce the behavior:
1. Use the latest master, in my case it was 41279af8a416e45b26a08550a3fe882d4aabba5a
2. $west update
3. Use SDK 0.11.4 or 0.11.3 I tested on both the same effect.
4. $west build -p -b waveshare_open103z tests/kernel/mem_protect/futex

**Expected behavior**
All tests should be build successfully.

**Impact**
waveshare_open103z can't have proper QA testing due to lack of the tests.

**Environment (please complete the following information):**
 - OS: Linux Fedora
 - Toolchain Zephyr SDK 0.11.4 or 0.11.3
 - Commit SHA 41279af8a416e45b26a08550a3fe882d4aabba5a

maksimmasalski, 2020-07-07:
@smanni Could you reproduce the same behaviour? Please test it on your machine.

smanni, 2020-07-07:
@maksimmasalski these tests fails also on other boards (e.g. nucleo_f103rb) with:

`error: array type has incomplete element type 'struct k_futex'`

smanni, 2020-07-07:
every board with CONFIG_ARM_MPU=n won't compile

erwango, 2020-07-08:
@maksimmasalski tests/kernel/mem_protect/futex/testcase.yaml declares the following:
```
tests:
  kernel.futex:
    filter: CONFIG_ARCH_HAS_USERSPACE
    tags: kernel userspace
```
So it requires CONFIG_ARM_MPU=y, which may not be the case for various reasons (no MPU, MPU available but userspaec requesting too much resources vs memory constraints).
So in case of STM32F103, MPU is not present..

maksimmasalski, 2020-07-08:
@erwango Does that means buildkite/zephyr tried to build that test for the unsupported board by mistake? https://github.com/zephyrproject-rtos/zephyr/pull/25842

erwango, 2020-07-08:
@maksimmasalski Ok, I hadn't seen this. This is indeed unexpected, will have a check..

erwango, 2020-07-08:
@maksimmasalski Build issue is due to 
```
/workdir/zephyr/boards/arm/waveshare_open103z/waveshare_open103z_defconfig:27: warning: attempt to assign the value 'n' to the undefined symbol CLOCK_STM32_PLL_XTPRE
```
This is a unfortunate merge conflict issue that has been fixed few commits after.
Can you rebase and have a new try ?

maksimmasalski, 2020-07-08:
> @maksimmasalski Build issue is due to
>
> ```
> /workdir/zephyr/boards/arm/waveshare_open103z/waveshare_open103z_defconfig:27: warning: attempt to assign the value 'n' to the undefined symbol [...]

After rebasing everything is fine. Close bug issue.

closed 2020-07-08 by maksimmasalski