zephyrproject-rtos/zephyr #26708: RFC: API Change: watchdog: wdt_feed error codes

pabigot, 2020-07-07:
## Problem Description

The Atmel SAMx2x and SAMx5x watchdog peripherals implemented in `wdt_sam0.c` require synchronization between the peripheral and system clocks.  Attempts to feed the watchdog while a previous operation is in progress will stall the CPU for a duration measured in ticks of the global and peripheral clock.  In the worst case a feed operation may stall the CPU for 6 ms.  The conditions in which a stall would occur are changes to the peripheral configuration and issuing a feed operation.

That a feed would cause a stall can be detected.  Proposed behavior of `wdt_feed()` in that case is to not stall, but instead return immediately with an error code `-EBUSY` indicating that completing the feed would cause a stall.

Motivating situation is #23115

## Proposed Change

See PR #26707

## Detailed RFC

See PR #26707

## Dependencies

Application code that invokes `watchdog_feed()` and reacts to a non-zero return value may need to be modified to accept and possibly react to the new non-fatal `-EBUSY` response.

henrikbrixandersen, 2020-07-07:
> That a feed would cause a stall can be detected. Proposed behavior of `wdt_feed()` in that case is to not stall, but instead return immediately with an error code `-EBUSY` indicating that completing the feed would cause a stall.

Perhaps `-EAGAIN` would be better here as it implies that the busy state is temporary and should go away without user intervention?

pabigot, 2020-07-07:
Agreed.

pabigot, 2020-07-09:
Change was fast-tracked so this is done.

closed 2020-07-09 by pabigot