zephyrproject-rtos/zephyr #26717: Big HTTP Download - Upgrade

Nukersson, 2020-07-07:
**Is your enhancement proposal related to a problem? Please describe.**

- [Big HTTP Download](https://docs.zephyrproject.org/latest/samples/net/sockets/big_http_download/README.html) example does not prints useful metrics (KiB, MiB, KiB/sec.) of download progress. It leads to time waste at development.

- [Big HTTP Download](https://docs.zephyrproject.org/latest/samples/net/sockets/big_http_download/README.html) example does not supports [Zephyr logging API](https://docs.zephyrproject.org/latest/reference/logging/index.html) properly. It leads to lost debug messages at dev time.

**Describe the solution you'd like**
The example project should become above upgrades.

pfalcon, 2020-07-08:
> does not prints useful metrics (KiB, MiB, KiB/sec.) of download progress. It leads to time waste at development.

Can you elaborate, what you mean by this? (The last sentence specifically).

Nukersson, 2020-07-08:
> > does not prints useful metrics (KiB, MiB, KiB/sec.) of download progress. It leads to time waste at development.
> 
> Can you elaborate, what you mean by this? (The last sentence specifically).

I mean downloaded [`Kibibytes aka KiB`](https://en.wikipedia.org/wiki/Kibibyte), `Mibibytes aka MiB`. Currently only Bytes will be printed - it is kind a hard to read.

PR #26734 (2020-07-11): Big http download upgrade#26717
PR for Big HTTP Download - Upgrade 

The PR #26719 must be closed. All updates to code reviews were moved here also.

Fixes #26717
closed 2020-07-11 by jukkar