zephyrproject-rtos/zephyr #26722: uarte_nrfx_poll_out() in NRF UARTE driver does not work with hardware flow control

plskeggs, 2020-07-08:
In Zephyr 2.1.99, the uarte_nrfx_poll_out() function worked correctly with hardware flow control, in this case at 1000000 baud to carry the Zephyr Bluetooth HCI traffic between an nRF9160 and an nRF52840.  When porting this app to Zephyr 2.3.0 (part of the Nordic NCS 1.3.0 release and associated Zephyr fork), the Bluetooth HCI interface stopped working -- it would timeout and assert in hci_core.c:bt_hci_cmd_send_sync() on line 370 in the HCI host device (the nRF9160):
```
	err = k_sem_take(&sync_sem, HCI_CMD_TIMEOUT);
	BT_ASSERT_MSG(err == 0, "k_sem_take failed with err %d", err);
```
With all logging enabled, it appeared that 1 to 2 bytes sent by the nRF52840 never arrived at the nRF9160.  After hooking up a logic analyzer to TX, RX, CTS, and RTS, it was clear that, for some reason, the nRF52840 never sent those 2 bytes.

After exhausting all other avenues for the source of the problem (bad wiring, bad device tree, etc.), it was noticed that the uarte_nrfx_poll_out() function had new changes in it since NCS 1.2.0 which allowed it to timeout and give up: specifically, the safety_cnt loop at line 952 and the NRFX_WAIT_FOR() macro on line 970, added in https://github.com/zephyrproject-rtos/zephyr/pull/22355/commits/a44e722959dea382e312da092875c04f199c8ec5.  As an experiment I modified the limits on the counter to much larger values, and now the 2 bytes were correctly sent.  I also tried going back to the old limits but adding logging; the timeout logging messages appear (see below).

I assume this can also be reproduced with the zephyr/samples/bluetooth/hci_uart sample on an nRF52840, and something compatible with it, as explained in the README.rst, but which is very slow on its processing of received data.  This sample was the basis for the app I'm working on (which was working well on top of Zephyr 2.1.99 as used in the Nordic NCS 1.2.0 release and associated Zephyr fork).  However, I have not done so.

The expected behavior for hardware flow control used with uarte_nrfx_poll_out() is that it would block forever until the other side asserts CTS.  Timeouts should not be used given the signature of the function, which returns no value.  Alternatively, the timeouts should be configurable in Kconfig, defaulting to infinite when hardware flow control is enabled, if they must really be provided to fix some race condition.  Perhaps that race condition could be isolated to only, for example, when CONFIG_UART_ASYNC_API is enabled, and revert to the old (no-timeout) behavior otherwise.  Either way, if timeouts are needed, the signature for poll_out() should be changed to return an error (which is a major UART driver API change, since it is a stable API).  A a minimum, it would have helped save a week of debugging if the safety loop change had simply logged a timeout error rather than just drop the data silently.

The attached screenshots show the UART signals:
1. *overview.png*: shows TX request and RX / RTS (labeling in traces is from the remote side's perspective; RX is the signal being sent from the device exhibiting the improper handling of hardware handshaking; RTS of the one side connects to CTS of the other, TX of one side connects to RX of the other, etc.)
2. *HCIResetCmd.png*: zoom in with decoding of the TX line (it sends an HCI Reset command, which is prefixed by HCI type 1 indicating a command): 0x01 0x03 0x0C 0x00
3. *First3Bytes.png*: zoom in on the first 3 bytes of response ended by deassert of RTS (an HCI Command Complete event, prefixed by HCI type 4 indicating an event):
0x04 0x0E 0x04
4. *Next2Bytes.png*: zoom in on the next 2 bytes with assert / deassert of RTS:
0x01 0x03
5. *Last1Byte.png*: zoom in on last 1 byte, prematurely ending the response:
0x0C

Log on remote device:
```
[00:00:01.537,353] <dbg> bt_hci_core.send_cmd: Sending command 0x0c03 (buf 0x2003e178) to driver
[00:00:01.538,940] <dbg> bt_hci_core.bt_send: buf 0x2003e178 len 3 type 0
[00:00:01.540,191] <dbg> bt_driver.h4_send: buf 0x2003e178 type 0 len 3
[00:00:01.592,315] <dbg> bt_driver.process_rx: remaining 0 discard 0 have_hdr 0 rx.buf 0x00000000 len 0
[00:00:01.593,994] <dbg> bt_driver.process_rx: remaining 2 discard 0 have_hdr 0 rx.buf 0x00000000 len 0
[00:00:01.595,672] <dbg> bt_driver.process_rx: remaining 1 discard 0 have_hdr 0 rx.buf 0x00000000 len 0
[00:00:01.597,320] <dbg> bt_driver.get_evt_hdr: Got event header. Payload 4 bytes
[00:00:01.598,663] <dbg> bt_driver.process_rx: remaining 4 discard 0 have_hdr 1 rx.buf 0x00000000 len 0
[00:00:01.600,311] <dbg> bt_driver.get_rx: type 0x04, evt 0x0e
[00:00:01.601,409] <dbg> bt_hci_core.bt_buf_get_cmd_complete: sent_cmd 0x2003e178
[00:00:01.602,752] <dbg> bt_driver.read_payload: Allocated rx.buf 0x2003e178
[00:00:01.604,034] <dbg> bt_driver.read_payload: got 1 bytes, remaining 3
[00:00:01.605,255] <dbg> bt_driver.read_payload: Payload (len 3): 0e0401
[00:00:01.606,475] <dbg> bt_driver.process_rx: remaining 3 discard 0 have_hdr 1 rx.buf 0x2003e178 len 3
[00:00:01.608,123] <dbg> bt_driver.read_payload: got 1 bytes, remaining 2
[00:00:01.609,375] <dbg> bt_driver.read_payload: Payload (len 4): 0e040103
[00:00:01.610,595] <dbg> bt_driver.process_rx: remaining 2 discard 0 have_hdr 1 rx.buf 0x2003e178 len 4
[00:00:01.612,274] <dbg> bt_driver.read_payload: got 1 bytes, remaining 1
[00:00:01.613,494] <dbg> bt_driver.read_payload: Payload (len 5): 0e0401030c
ASSE*** Booting Zephyr OS build v2.3.0-rc1-ncs1-3-ga92097f1f2b6  ***
```
Note above -- the ASSE(RT...) from giving up on receiving the last byte was a result of timing out after 10 seconds.

Log on device involved in this bug report:
```
00> [00002277] dbg> bt_ctlr_hci_driver.cmd_handle: Replying with event of 6 bytes
00> [00286614] <err> uart_nrfx_uarte: Timeout waiting for ENDTX
00> [00310223] <err> lte_gateway_ble: Timeout waiting for h4_send() to complete!
00> GPIO reset line asserted, device reset.
```
Note above -- the uart_nrfx_uarte log entry is one I added for debugging to uart_nrfx_uarte.c:
```
	NRFX_WAIT_FOR(nrf_uarte_event_check(uarte, NRF_UARTE_EVENT_ENDTX),
		      1000, 1, res);
	if (!res) {
		LOG_PANIC();
		LOG_ERR("Timeout waiting for ENDTX");
	}
```
- OS: Windows 10
 - Toolchain:  west v0.7.2, GNU Tools ARM Embedded 9-2019-q4-major
-  Version: Nordic NCS 1.3.0 tagged release, containing Zephyr 2.3.0

plskeggs, 2020-07-08:
Note that there was a similar commit in the same PR which may cause the same problem for the uart_nrfx_uart.c driver:
https://github.com/zephyrproject-rtos/zephyr/pull/22355/commits/2512b7333cbdeb1badd12198d4d61a2dbff17d72

mbolivar-nordic, 2020-07-08:
@carlescufi assigned to you for triage; since the issue appears to have been added in https://github.com/zephyrproject-rtos/zephyr/commit/a44e722959dea382e312da092875c04f199c8ec5, perhaps @Mierunski can take a look at this?

carlescufi, 2020-07-08:
@mbolivar-nordic thanks.

@plskeggs we currently use `uart_poll_out()` with hardware flow control enabled in several of the samples and configurations. I will have to wait until @nordic-krch, @Mierunski or @anangl have a chance to dig into the issue.

plskeggs, 2020-07-22:
It's important that tests cover situations where CTS / RTS elongate a transfer, and not just behave in a nominal, no-delay manner.

plskeggs, 2020-08-26:
Any progress on this?

nordic-krch, 2020-09-03:
#27917 reworked hci_uart to use `uart_fifo_fill` instead of `poll_out` that will solve the issue for hci case. I will take a look at how `poll_out` could be fixed.

carlescufi, 2020-09-14:
> #27917 reworked hci_uart to use `uart_fifo_fill` instead of `poll_out` that will solve the issue for hci case. I will take a look at how `poll_out` could be fixed.

@nordic-krch is this something you could try to fix for Zephyr 2.4.0?

plskeggs, 2020-09-28:
I tried https://github.com/zephyrproject-rtos/zephyr/pull/27917 @nordic-krch and that did help my particular use case with hci.  Seems a more general solution is needed though.

closed 2020-10-14 by carlescufi