zephyrproject-rtos/zephyr #26723: NULL handler in work queue entry can be called resulting in silent reboot

plskeggs, 2020-07-08:
Due to some debugging changes and delays added to an application, an application thread inadvertently called `k_delayed_work_submit()` with a pointer to a `struct k_delayed_work` variable which had not yet been initialized.  Since it was not yet initialized, the member variable `struct k_work work` contained a `k_work_handler_t handler` member variable which was still NULL.  But, because Zephyr's lib/os/work_q.c function `z_work_q_main()` did not check to see if the handler was NULL before calling, it called the NULL function pointer.  This did not result in a processor exception or any other indication as to why the application restarted.

Because none of the usual assert handlers, reboot handlers, fatal error handlers, or exception handlers were triggered, it was quite difficult to figure out what was happening (breakpoints were set using gdb but to no avail).  It took stepping through code line by line to find the problem.

This can be reproduced with a simple addition to zephyr/sample/basic/minimal/src/main.c:
```
/*
 * Copyright (c) 2019 Nordic Semiconductor ASA
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr.h>
#include <kernel.h>
#include <sys/printk.h>

struct k_work work;

void work_handler(struct k_work *work)
{
	printk("In work_handler()\n");
}

void main(void)
{
	printk("start\n");

	// trigger bug: submit work before initializing it...
	k_work_submit(&work);

	k_work_init(&work, work_handler);

	printk("end\n");
}
```
And change prj.conf to:
```
CONFIG_ASSERT=y
```

Build and run this on a board, such as a Nordic nRF9160DK:
```
C:\wrk\zephyr>west build -b nrf9160dk_nrf9160 samples/basic/minimal
C:\wrk\zephyr>west flash --hex-file build/zephyr/zephyr.hex
```
Serial terminal output shows this:
```
start
*** Booting Zephyr OS build v2.3.0-rc1-ncs1-3-ga92097f1f2b6  ***
start
*** Booting Zephyr OS build v2.3.0-rc1-ncs1-3-ga92097f1f2b6  ***
...
```

Expected behavior would be that, at a minimum, somewhere in the call tree the NULL handler would be noticed with an assert.  For example, if this line were added to `z_work_q_main()` prior to the line containing `handler(work);`, it would have caused logging which would make it easier to diagnose:
```
		__ASSERT(handler != NULL, "handler must be provided");
...
			handler(work);
```

With this change, the terminal output shows:
```
*** Booting Zephyr OS build v2.3.0-rc1-ncs1-3-ga92097f1f2b6  ***
start
ASSERTION FAIL [handler != ((void *)0)] @ C:/wrk/zephyr/lib/os/work_q.c:28
        handler must be provided
*** Booting Zephyr OS build v2.3.0-rc1-ncs1-3-ga92097f1f2b6  ***
start
ASSERTION FAIL [handler != ((void *)0)] @ C:/wrk/zephyr/lib/os/work_q.c:28
        handler must be provided
...
```

The impact was that it was very hard to debug when it happened in a much more complex application.

OS: Windows 10
Toolchain: west v0.7.2, GNU Tools ARM Embedded 9-2019-q4-major
Version: Nordic NCS 1.3.0 tagged release, containing Zephyr 2.3.0

mbolivar-nordic, 2020-07-08:
@andyross any advice on where the ASSERT might belong?

andyross, 2020-07-08:
Assertions are best used to identify "should never happen" cases internal to the Zephyr code.  They're for early detection of bugs in Zephyr, not protection against bugs in user code.  I wouldn't reject a patch that added some, but I'm not sure I see the value.  There are an effectively infinite numbers of ways that bad user data can cause a crash, we can't possibly detect all of them.

Is a corrupt work queue item (and, I guess, corrupt in exactly such a way that the pointer is valid but the callback is NULL) really such a common mistake that we want protection against it, vs. other mistakes?  I dunno.  I guess my feeling is that's a case that has to be made with evidence.

plskeggs, 2020-07-08:
Well, whether the solution is an assert or not, I do feel that, just like dereferencing a NULL pointer is a bad idea, jumping to one is as well.  In z_work_q_main(), it already checks whether the `struct k_work *work` pulled from the queue is NULL, and if so, ignores it.  Seems to me at a minimum it should do the same if `work->handler` is NULL.

PR #27107 (2020-07-24): lib/os/work_q: sanity check work_q handler prior to calling it
Just as NULL pointers should not be dereferenced, they should
not be called either.

Fixes #26723