zephyrproject-rtos/zephyr #26725: USB suspend-resume process is not properly handled

stephanosio, 2020-07-08:
**Describe the bug**
The USB subsystem does not properly handle the suspend-resume process.

When a USB device is suspended, all on-going transfers are cancelled:

https://github.com/zephyrproject-rtos/zephyr/blob/0651688dd350851ab5e00265d1170f3f015e77c0/subsys/usb/usb_device.c#L1017-L1025

The CDC ACM device class implementation, for instance, does not restore the cancelled USB transfer upon the resume event and this results in the USB transfer remaining in cancelled state:

https://github.com/zephyrproject-rtos/zephyr/blob/0651688dd350851ab5e00265d1170f3f015e77c0/subsys/usb/class/cdc_acm.c#L437-L440

This causes the USB device to become unresponsive.

**To Reproduce**
1. Build and flash `samples/subsys/usb/cdc_acm`
2. Open the board USB device serial port and see that everything is echoed.
3. Close the board USB device serial port
3. Suspend the board USB device.
4. Resume the board USB device.
5. Open the board USB device serial port and see that the device does not echo anything back.

**Expected behavior**
USB endpoint transfers are in working state after device is resumed.

**Impact**
USB suspend-resume process does not work, and any attempt by the host to suspend the device will result in the device being unusable until the device is re-enumerated (and configured).

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.3
 - Commit SHA: cbb252adef08ba482b14ef2ab562baf879956c93

**Additional context**
Note that this issue is not specific to the CDC ACM device class; a quick glance over other device class implementations suggest that they will be affected by this issue as long as the USB transfers are cancelled when a suspend event is received.

Also note that the `USB_DC_DISCONNECTED` event is not implemented or, as far as I can tell, does not work on any platforms.

stephanosio, 2020-07-08:
cc @finikorg @jfischer-phytec-iot

stephanosio, 2020-07-08:
To briefly summarise the kind of mess we are in:
1. USB drivers do not report `USB_DC_DISCONNECTED` event because most USB device cores do not report such a thing (and there is a good reason for that), and we must rely on the `USB_DC_SUSPEND` event occurring when the device is disconnected (and, of course, suspended).
2. #16193 made the USB subsystem cancel/free all on-going transfer slots upon `USB_DC_SUSPEND` (in this case, disconnection) in order to make sure that re-connection does not result in USB transfer slot leak.
3. In case of a genuine suspend condition and in case the device is resumed thereafter, the device is not re-enumerated (and configured), so the USB transfer is not re-initiated.

jfischer-phytec-iot, 2020-07-08:
> **To Reproduce**
> 
>     1. Build and flash `samples/subsys/usb/cdc_acm`
> 
>     2. Open the board USB device serial port and see that everything is echoed.
> 
>     3. Close the board USB device serial port
> 
>     4. Suspend the board USB device.
> 
>     5. Resume the board USB device.
> 
>     6. Open the board USB device serial port and see that the device does not echo anything back.

Can not reproduce, works fine on nRF52840. It works even with #26657, but only few times.

stephanosio, 2020-07-08:
@jfischer-phytec-iot I can reproduce this issue on nRF52840 (`nrf52840dk_nrf52840`). Please re-check.

jfischer-phytec-iot, 2020-07-08:
> The CDC ACM device class implementation, for instance, does not restore the cancelled USB transfer upon the resume event and this results in the USB transfer remaining in cancelled state:

Expected transition is SUSPEND->RESUME, this behavior is correct. Everything after USB_DC_DISCONNECTED will reset the device and reconfigure usb_transfer. If you platform does not support detached or disconnected events suspend can for sure used instead but you must implement resume too.

stephanosio, 2020-07-08:
> Expected transition is SUSPEND->RESUME, this behavior is correct. Everything after USB_DC_DISCONNECTED will reset the device and reconfigure usb_transfer. If you platform does not support detached or disconnected events suspend can for sure used instead but you must implement resume too.

@jfischer-phytec-iot I think you are still thinking about the other issue (#26656). This is different.

I am not talking about the disconnect/detach event here, but an actual suspend-resume event; and yes, you need to implement both suspend and resume events to get suspend and resume features working -- that was clear enough.

Try turning on autosuspend on the host and you will see that nRF52840 stops responding on the ACM serial port after the device goes through suspend and resume.

stephanosio, 2020-07-08:
Also I forgot to mention, this issue is seen on all platforms I tested (sam0, sam, kinetis, nrf).

This is a logic issue, not a platform driver issue.

stephanosio, 2020-07-08:
More detailed steps on how to reproduce this:
1. Turn on autosuspend
```
echo 'auto' > '/sys/bus/usb/devices/(Zephyr USB Device)/power/control'
```

2. Wait for `usb_cdc_acm: USB device suspended` to be printed on the UART console

3. Open the board USB ACM serial port (`usb_cdc_acm: USB device resumed` will be printed on the UART console)

Now you will see that the USB ACM serial port does not echo anything and became completely unresponsive for the reasons described above.

jfischer-phytec-iot, 2020-07-08:
> This is a logic issue, not a platform driver issue.

Where is the issue? What you describe is correct behavior:

>The CDC ACM device class implementation, for instance, does not restore the cancelled USB transfer upon the resume event and this results in the USB transfer remaining in cancelled state:

stephanosio, 2020-07-08:
> Where is the issue? What you describe is correct behavior:

So you are saying that the ACM device class implementation not restoring the USB transfer after resume and being unable to communicate with the host is the correct behaviour?

I don't think it takes an expert to see that is a problem unless, of course, your goal is to maintain the status quo that is an unusable substandard implementation.

jfischer-phytec-iot, 2020-07-08:
> So you are saying that the ACM device class implementation not restoring the USB transfer after resume and being unable to communicate with the host is the correct behaviour?

No, I read your sentence too quickly. The transfers are cancelled not disabled. On resume event CDC ACM class sets internal state to `dev_data->usb_status = USB_DC_CONFIGURED;` and allows to fill the fifo (tx_work_handler -> usb_transfer will pass the data to IN endpoint buffer), callbacks for the OUT endpoint are still configured for the usb_transfer and will be triggered.

jfischer-phytec-iot, 2020-07-08:
I can not reproduce it, I should try an other host. Probably regression or was not really fixed in https://github.com/zephyrproject-rtos/zephyr/issues/17488

stephanosio, 2020-07-08:
The log indicates that `cdc_acm_read_cb` is not called after resume.

```
<Suspending>
[00:00:30.384,155] <dbg> usb_transfer.usb_cancel_transfers: Cancel transfer
[00:00:30.384,185] <dbg> usb_cdc_acm.cdc_acm_dev_status_cb: cfg 0x200002d4 status 5
[00:00:30.384,185] <inf> usb_cdc_acm: USB device suspended
[00:00:30.384,216] <dbg> usb_transfer.usb_transfer_work: Transfer cancelled or completed, ep 0x01
[00:00:30.384,216] <dbg> usb_transfer.usb_transfer_work: Transfer done, ep 0x01, status -72, size 0
<Resuming>
[00:00:53.035,644] <dbg> usb_cdc_acm.cdc_acm_dev_status_cb: cfg 0x200002d4 status 6
[00:00:53.035,675] <inf> usb_cdc_acm: USB device resumed
[00:00:53.102,294] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:53.102,325] <dbg> usb_device.custom_handler: bRequest 0x00, wIndex 0x0000
[00:00:53.102,325] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_STATUS
[00:00:53.102,416] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:53.102,905] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:53.102,935] <dbg> usb_device.class_handler: bRequest 0x22, wIndex 0x0000
[00:00:53.102,935] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x3
[00:00:53.103,363] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:53.103,546] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:53.103,576] <dbg> usb_device.class_handler: bRequest 0x20, wIndex 0x0000
[00:00:53.103,607] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 115200 0 0 8
[00:00:53.103,607] <dbg> usb_device.usb_handle_control_transfer: >> usb_data_to_host(2)
<After sending a char on the ttyACM>
[00:01:07.453,948] <dbg> usb_transfer.usb_transfer_work: Transfer cancelled or completed, ep 0x01
```

stephanosio, 2020-07-08:
> I can not reproduce it, I should try an other host. Probably regression or was not really fixed in #17488

Is your host configuring (`REQ_SET_CONFIGURATION`) the device after resume? Because if it does, this problem will not be reproducible since `USB_DC_CONFIGURED` calls `cdc_acm_read_cb`.

It is my understanding that `REQ_SET_CONFIGURATION` should not be sent after resume and that would indeed be your host's problem.

jfischer-phytec-iot, 2020-07-08:
I can reproduce it on PC, sorry, I think it is a regression as I tested it to fix #17488

jfischer-phytec-iot, 2020-07-08:
At fist glance the fix for cdc acm class could be to re-trigger the transfer for OUT endpoint:

```diff
diff --git a/subsys/usb/class/cdc_acm.c b/subsys/usb/class/cdc_acm.c
index 54ebcc6bd9..036a47f7e0 100644
--- a/subsys/usb/class/cdc_acm.c
+++ b/subsys/usb/class/cdc_acm.c
@@ -436,6 +436,8 @@ static void cdc_acm_do_cb(struct cdc_acm_dev_data_t *dev_data,
                break;
        case USB_DC_RESUME:
                dev_data->usb_status = USB_DC_CONFIGURED;
+               cdc_acm_read_cb(cfg->endpoint[ACM_OUT_EP_IDX].ep_addr, 0,
+                               dev_data);
                LOG_INF("USB device resumed");
                break;
        case USB_DC_SOF:
```

stephanosio, 2020-07-09:
@jfischer-phytec-iot Yes, that works for the most part.

The only problem is that some platforms trigger the `USB_DC_RESUME` event multiple times when resuming and that may cause multiple transfer slots to be allocated for single endpoint. For instance, on `frdm_k64f`, I see `usb_cdc_acm: USB device resumed` happening twice after the device is resumed (cc @MaureenHelm). I suppose we call that a bug with the Kinetis USB driver and address it separately; though, I am also seeing something similar happening on the SAM0 platform (still investigating).

On a side note, I am not an expert on the USB stuff, but does it make sense to cancel all on-going transfers when suspending? I think @anangl raised the same question here too: https://github.com/zephyrproject-rtos/zephyr/issues/17488#issuecomment-516800469

I wonder if it would be better to free all transfer slots (i.e. call `usb_cancel_transfers`) before configuring the device (i.e. as a first thing in `REQ_SET_CONFIGURATION` handler) since that is the place where a new set of endpoints are being activated and it makes sense for all existing transfers to be terminated.

jfischer-phytec-iot, 2020-07-09:
> @jfischer-phytec-iot Yes, that works for the most part.
> 
> The only problem is that some platforms trigger the `USB_DC_RESUME` event multiple times when resuming and that may cause multiple transfer slots to be allocated for single endpoint. For instance, on `frdm_k64f`, I see `usb_cdc_acm: USB device resumed` happening twice after the device is resumed (cc @MaureenHelm). I suppose we can call that a bug with the Kinetis USB driver and address it separately; though, I am also seeing something similar happening on the SAM0 platform (still investigating).

I change the patch to filter spurious resume events which **may** happen during device detection. I had not time to test on frdm_k64f but will do. Not all the drivers are aligned in the behavior.

> 
> On a side note, I am not an expert on the USB stuff, but does it make sense to cancel all on-going transfers when suspending? I think @anangl raised the same question here too: [#17488 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/17488#issuecomment-516800469) [#17492 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/17492#discussion_r309145878)

We do that on suspend and disconnect events, https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_device.c#L1017

> I wonder if it would be better to free all transfer slots (i.e. call `usb_cancel_transfers`) before configuring the device (i.e. as a first thing in `REQ_SET_CONFIGURATION` handler) since that is [...]

That will be done, if necessary, at the line https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_device.c#L537

closed 2020-07-17 by carlescufi