zephyrproject-rtos/zephyr #26729: FCB flash_area_write fails on nRF52840DK when using mx25r64 storage

ball-hayden, 2020-07-08:
**Describe the bug**

Calls to `flash_area_write(fcb->fap, FCB_ENTRY_FA_DATA_OFF(loc), test_data, i);` fail when using mx25r64 as a storage device.

I suspect, more generally, that this is an issue for any flash area using the `nrf_qspi_nor` driver.

**To Reproduce**
Steps to reproduce the behavior:
1. Apply the attached patch, which reconfigures fcb tests to use `mx25r64` storage available on nRF52840DK
2. `west build -t flash -p auto -b nrf52840dk_nrf52840 zephyr/tests/subsys/fs/fcb`
3. Observe failing tests

**Expected behavior**

FCB should be usable regardless of the underlying driver.

**Impact**

This is blocking our use of FCB.

**Logs and console output**

<details>
<summary>Test Log</summary>

<pre>
*** Booting Zephyr OS build v2.3.0-rc1-ncs1  ***
Running test suite test_fcb
===================================================================
starting test - test_fcb_len
PASS - test_fcb_len
===================================================================
starting test - test_fcb_init
PASS - test_fcb_init
===================================================================
starting test - test_fcb_empty_walk
PASS - test_fcb_empty_walk
===================================================================
starting test - test_fcb_append

    Assertion failed at ../src/fcb_test_append.c:30: test_fcb_append: (rc not equal to 0)
flash_area_write call failure (-22) (iteration: 1)
FAIL - test_fcb_append
===================================================================
starting test - test_fcb_append_too_big

    Assertion failed at ../src/fcb_test_append_too_big.c:45: test_fcb_append_too_big: (rc == 0 is false)
fcb_append call failure
FAIL - test_fcb_append_too_big
===================================================================
starting test - test_fcb_append_fill

    Assertion failed at ../src/fcb_test_append_fill.c:49: test_fcb_append_fill: (rc == 0 is false)
flash_area_write call failure
FAIL - test_fcb_append_fill
===================================================================
starting test - test_fcb_rotate

    Assertion failed at ../src/fcb_test_rotate.c:50: test_fcb_rotate: (rc == 0 is false)
flash_area_write call failure
FAIL - test_fcb_rotate
===================================================================
starting test - test_fcb_multi_scratch

    Assertion failed at ../src/fcb_test_multiple_scratch.c:41: test_fcb_multi_scratch: (rc == 0 is false)
flash_area_write call failure
FAIL - test_fcb_multi_scratch
===================================================================
starting test - test_fcb_last_of_n

    Assertion failed at ../src/fcb_test_last_of_n.c:38: test_fcb_last_of_n: (rc == 0 is false)
flash_area_write call failure
FAIL - test_fcb_last_of_n
===================================================================
starting test - test_fcb_reset

    Assertion failed at ../src/fcb_test_reset.c:44: test_fcb_reset: (rc == 0 is false)
flash_area_write call failure
FAIL - test_fcb_reset
===================================================================
Test suite test_fcb failed.
===================================================================
PROJECT EXECUTION FAILED
</pre>

</details>

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: `gcc-arm-embedded 9-2020-q2-update`
 - Commit SHA or Version used: https://github.com/nrfconnect/sdk-zephyr/tree/bbd71e23a2d56f05cefc9dca7bef609c22c1966c

**Additional context**

The issue here appears to be related to address alignment.

The `nrf_qspi_nor` driver reports that any write block size is supported:

https://github.com/nrfconnect/sdk-zephyr/blob/bbd71e23a2d56f05cefc9dca7bef609c22c1966c/drivers/flash/nrf_qspi_nor.c#L750

This leads to `fcb_append` setting an `fe_data_off` that is 1-byte aligned (as `cnt` is 1 or 2):

https://github.com/nrfconnect/sdk-zephyr/blob/bbd71e23a2d56f05cefc9dca7bef609c22c1966c/subsys/fs/fcb/fcb_append.c#L105

However, the `nrf_qspi_nor` driver also requires that addresses are 4-byte aligned:

https://github.com/nrfconnect/sdk-zephyr/blob/bbd71e23a2d56f05cefc9dca7bef609c22c1966c/drivers/flash/nrf_qspi_nor.c#L554-L557

This leads to calls to `flash_area_write` failing with `EINVAL`.

ball-hayden, 2020-07-08:
Further investigation has shown than increasing the `nrf_qspi_nor` write block size to 4 solves this problem (see above commit).

I'm not currently proposing that as a fix, unless someone with more understanding than I have can confirm that the driver is reporting the wrong block size there?

carlescufi, 2020-07-21:
@pabigot states that this should've been fixed by the recent PRs that modified the nRF QSPI NOR driver. @nvlsianpu and @de-nordic can you double-check that?

pabigot, 2020-07-21:
Specifically I believe that if the QSPI driver no longer has size or alignment requirements, there should be no need to adjust the write block size.

Also note that the original report is relative to the Nordic fork, not upstream Zephyr.

ball-hayden, 2020-07-22:
Thanks @pabigot / @carlescufi. I'll hunt out those PRs and merge them into our fork to check that solves the issue.

> Also note that the original report is relative to the Nordic fork, not upstream Zephyr.

Apologies - I should have made this clearer in the report (it would be helpful if Nordic still allowed issue reports on their fork!)

ball-hayden, 2020-07-22:
I've cherry-picked ce7d3f3437213f1e63850db2222d022be4aa571b (which I believe is the only related commit?) onto the nordic sdk-zephyr fork `master`, and with the patch in the description applied to make FCB tests run against the mx25r64 (and therefore use qSPI) the tests are still failing so I believe this issue is still unresolved.

<details>
<summary>Test Output</summary>

```
*** Booting Zephyr OS build zephyr-v2.3.0-1087-gce7d3f343721  ***
Running test suite test_fcb
===================================================================
START - test_fcb_len
 PASS - test_fcb_len
===================================================================
START - test_fcb_init
 PASS - test_fcb_init
===================================================================
START - test_fcb_empty_walk
 PASS - test_fcb_empty_walk
===================================================================
START - test_fcb_append

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_append.c:30: test_fcb_append: (rc not equal to 0)
flash_area_write call failure (-22) (iteration: 4)
 FAIL - test_fcb_append
===================================================================
START - test_fcb_append_too_big

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_append_too_big.c:45: test_fcb_append_too_big: (rc == 0 is false)
fcb_append call failure
 FAIL - test_fcb_append_too_big
===================================================================
START - test_fcb_append_fill

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_append_fill.c:49: test_fcb_append_fill: (rc == 0 is false)
flash_area_write call failure
 FAIL - test_fcb_append_fill
===================================================================
START - test_fcb_rotate

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_rotate.c:50: test_fcb_rotate: (rc == 0 is false)
flash_area_write call failure
 FAIL - test_fcb_rotate
===================================================================
START - test_fcb_multi_scratch

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_multiple_scratch.c:41: test_fcb_multi_scratch: (rc == 0 is false)
flash_area_write call failure
 FAIL - test_fcb_multi_scratch
===================================================================
START - test_fcb_last_of_n

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_last_of_n.c:38: test_fcb_last_of_n: (rc == 0 is false)
flash_area_write call failure
 FAIL - test_fcb_last_of_n
===================================================================
START - test_fcb_reset

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_reset.c:44: test_fcb_reset: (rc == 0 is false)
flash_area_write call failure
 FAIL - test_fcb_reset
===================================================================
Test suite test_fcb failed.
===================================================================
PROJECT EXECUTION FAILED
```

</details>

ball-hayden, 2020-07-22:
Ah. [ce7d3f3](https://github.com/zephyrproject-rtos/zephyr/commit/c3cc0c8787a79dd775a298ce04cdced3125a2177) allows unaligned read.

Fixing this would require unaligned write, which it looks like is still not possible?

nvlsianpu, 2020-09-08:
The nrf qspi_nor driver is able to write with 32-bits word alignment of size and offset, apart form sub-word wrtes when it can write part of word to alligned offset (which is a hack for mcumgr.
I will change write-block-size to 4.
FYI @sigvartmh

PR #28189 (2020-09-10): drivers/flash/nrf_qspi_nor: corrects write-block-size to 4B
The driver is able to write with 32-bits word alignment of size
and offset, apart form sub-word writes for which it can write
part of word to aligned offset (which is a hack).

fixes #26729
commit 285d8786 (2020-09-10): drivers/flash/nrf_qspi_nor: corrects write-block-size to 4B

The driver is able to write with 32-bits word alignment of size
and offset, apart form sub-word writes for which it can write
part of word to aligned offset (which is a hack).

fixes #26729
closed 2020-09-10 by MaureenHelm