zephyrproject-rtos/zephyr #26732: Scan and advertise only on single bluetooth channel - Zephyr, Bluetooth

EdytaBosacka, 2020-07-08:
I would like to advertise and scan ONLY on one bluetooth channel (for example channel 37). Is there any way to do this in Zephyr? I tried to search parametr responsibles for it in this structure:
struct bt_le_scan_param scan_param = {
.type = BT_HCI_LE_SCAN_PASSIVE,
.options = BT_LE_SCAN_OPT_NONE,
.interval = 0x0010,
.window = 0x0010,}
but i dont think that this is solution.

I also found a file lll_chan in zephyrproject\zephyr\subsys\bluetooth\controller\ll_sw\nordic\lll. TI includes channel selection algorithms, but i dont see where can i select just one advertising channel?

carlescufi, 2020-07-08:
@EdytaBosacka You need to use extended advertising, which is supported in Zephyr but currently only in the Host. Once this is supported in the Controller (@andrzej-kaczmarek is working on it) you will be able to use this. If you are only using the Host and have a separate controller then you should be able to do it already.
However, Zephyr doesn't currently make this configurable by the app:
`cp->prim_channel_map  = 0x07;`

So I will treat this issue as an enhancement request to make this controllable by the app.

EdytaBosacka, 2020-07-09:
Hello :)
So do i need set up this: BT_LE_ADV_OPT_SCANNABLE = BIT(9), ?
Because I have also found this function:
int bt_le_set_chan_map(uint8_t chan_map[5]);
/** @brief Helper for parsing advertising (or EIR or OOB) data.
If I want just SCAN one bluetooth channel do i need use this function, set
up  BT_LE_ADV_OPT_SCANNABLE = BIT(9), (if yes, where should be this set up,
in plik .CONF ?) , or  should I do this both?

Thank you for your help :)

Z wyrazami szacunku,
*Edyta Bosacka*

EdytaBosacka, 2020-07-09:
@carlescufi 
Hello :) 
I dont really know what "separate controller" means.  i have 2 boards and i want to advertise on channel 37 on the first board, and receive  on this 37 channel by the second board.

So do i need set up this: BT_LE_ADV_OPT_SCANNABLE = BIT(9), ?
Because I have also found this function:
**int bt_le_set_chan_map(uint8_t chan_map[5]);****
/** @brief Helper for parsing advertising (or EIR or OOB) data.

If I want just SCAN one bluetooth channel do i need use this function or set
up  BT_LE_ADV_OPT_SCANNABLE = BIT(9), (if yes, where should be this set up,
in plik .CONF ?) , or  should I do this both?

carlescufi, 2020-07-09:
@EdytaBosacka what you want to do is not possible right now. You can subscribe to this issue for updates when we get to implementing it.

EdytaBosacka, 2020-07-09:
So to make it work should i extend (write) A function
Le_ext_adv_param_set for controller?

EdytaBosacka, 2020-07-09:
@carlescufi  So to make it work should i extend (write) A function ? (I
would like to implement this funcionality by myself)
Le_ext_adv_param_set for controller?

carlescufi, 2020-07-10:
> @carlescufi So to make it work should i extend (write) A function Le_ext_adv_param_set for controller? (I would like to write extended advertising funcionality by myself)

You would need to extend the structure that it takes to include the channel set, and then pass it down to the controller.

joerchan, 2020-07-14:
@EdytaBosacka Adding it for the advertiser would be easy, AE support is not needed, it can be added for both advertising APIs.
However setting scan channel map is not supported by HCI (host-controller interface specified by the Bluetooth core Specification).

joerchan, 2020-09-29:
Striking scan channel selection from this feature request, since that is not supported by HCI.

PR #28759 (2020-10-15): Bluetooth: host: Add advertising options to disable channel
Add advertising options to disable the individual advertising
channel 37, 38 and 39 per advertising instance.

Fixes: #26732
commit ea273100 (2020-10-15): Bluetooth: host: Add advertising options to disable channel

Add advertising options to disable the individual advertising
channel 37, 38 and 39 per advertising instance.

Fixes: #26732
closed 2020-10-15 by jhedberg