zephyrproject-rtos/zephyr #26740: Add support for userspace on RISC-V [seeking feedback on partial implementation]

lindemer, 2020-07-08:
The RISC-V privileged specification defines a hardware extension called Physical Memory Protection (PMP), which is conceptually very similar to an Arm MPU. Zephyr already supports boards that contain these extensions (e.g., SiFive HiFive), but this hardware is unused. My colleagues and I have created a working (partial) implementation of userspace support for the HiFive Rev B, which can be found on the master branch of [my Zephyr fork](https://github.com/lindemer/zephyr). 

We'd like to complete this implementation and get it to pass the tests in `tests/kernel/mem_protect/userspace`. For now, we've just included a very basic sample in `samples/userspace/riscv_pmp`, which demonstrates context switching and system calls from userspace threads. For an overview of the memory protection configuration and how it differs from other architectures, check out the ASCII art in `arch/riscv/core/thread.c`.

This is our first contribution to the Zephyr project, and we'd like some feedback on how best to proceed before we submit a pull request. Is anyone else in the Zephyr community interested in this feature?

carlescufi, 2020-07-08:
@mgielda @PiotrZierhoffer would you be up for reviewing this?

stephanosio, 2020-07-09:
It would be nice if you could submit a draft pull request (even if it is not in mergeable state/finalised) so that the reviews can be tracked more sanely.

andrewboie, 2020-07-09:
Hi, there! Appreciate the contribution, I designed much of Zephyr's userspace feature and would like to review this. If the code is still WIP but you'd like early feedback, a draft pull request is probably appropriate.

tests/kernel/mem_protect/userspace is a good place to start for checking completeness. Eventually your implementation should eventually pass all tests in tests/kernel/mem_protect and any test tagged with "userspace".

lindemer, 2020-07-09:
Thanks for the replies. We'll continue to develop it over the coming weeks, but feedback on the work so far in [#26755](https://github.com/zephyrproject-rtos/zephyr/pull/26755) would be greatly appreciated! We're also planning to add support for LiteX VexRiscv in the near future.

carlescufi, 2024-02-22:
This is complete

closed 2024-02-22 by carlescufi