zephyrproject-rtos/zephyr #26759: Build error -  Nothing found at GNUARMEMB_TOOLCHAIN_PATH

Satkarni, 2020-07-09:
I have just installed Zephyr and am trying to build hello world. I get the following error:
<pre><font color="#8AE234"><b>rp@jiodev</b></font>:<font color="#729FCF"><b>~/zephyrproject/zephyr</b></font>$ west build -p auto -b nucleo_f401re samples/basic/blinky
<font color="#8AE234">-- west build: generating a build system</font>
Including boilerplate (Zephyr base (cached)): /home/rp/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/rp/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.3.99 (/home/rp/zephyrproject/zephyr)
-- Board: nucleo_f401re
-- Found toolchain: zephyr (/home/rp/zephyr-sdk-0.11.3)
-- Found west: /home/rp/.local/bin/west (found suitable version &quot;0.7.2&quot;, minimum required is &quot;0.7.1&quot;)
-- Found dtc: /usr/bin/dtc (found suitable version &quot;1.4.7&quot;, minimum required is &quot;1.4.6&quot;)
CMake Error at /home/rp/zephyrproject/zephyr/cmake/toolchain/gnuarmemb/generic.cmake:16 (message):
  Nothing found at GNUARMEMB_TOOLCHAIN_PATH:
  &apos;/home/rp/xpack-arm-none-gcc-eabi-9.2.1-1.1&apos;
Call Stack (most recent call first):
  /home/rp/zephyrproject/zephyr/cmake/generic_toolchain.cmake:64 (include)
  /home/rp/zephyrproject/zephyr/cmake/app/boilerplate.cmake:509 (include)
  /home/rp/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/rp/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:4 (find_package)
</pre>

I have checked that the path to the cross compiler is ok, the compiler is installed ok and also successfully compiled a small c file with the cross compiler. So it does not seem an issue with the cross compiler.

carlescufi, 2020-07-09:
Are you using GNU Arm Embedded as downloaded from: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm? `GNUARMEMB_TOOLCHAIN_PATH` is unlikely to work with other distributions of it.

Satkarni, 2020-07-10:
I am using the following cross compiler:
https://xpack.github.io/arm-none-eabi-gcc/
The xPack GNU Arm Embedded GCC is an alternate binary distribution that complements the official GNU Arm Embedded Toolchain maintained by Arm.

Will this work ?

carlescufi, 2020-07-23:
> I am using the following cross compiler:
> https://xpack.github.io/arm-none-eabi-gcc/
> The xPack GNU Arm Embedded GCC is an alternate binary distribution that complements the official GNU Arm Embedded Toolchain maintained by Arm.
> 
> Will this work ?

Probably not. Our `gnuarmemb` toolchain option requires a certain layout of the  `GNUARMEMB_TOOLCHAIN_PATH`, which your version is likely not respecting. I suggest you either try with the original distribution of GNU Arm Embedded or you use the [crosscompile](https://docs.zephyrproject.org/latest/getting_started/toolchain_other_x_compilers.html#other-cross-compilers) toolchain mechanism instead.

closed 2020-07-23 by carlescufi

Satkarni, 2020-07-24:
Ok, I got it to build with the standard zephyr sdk, so my issue is solved. 
Maybe I'll try with the cross-compile option in the future if any issues arise for the zephyr sdk.