zephyrproject-rtos/zephyr #26769: "west flash -r openocd --serial <serial_num>" ignores serial_num and flashes wrong board

wojciiii, 2020-07-09:
**Describe the bug**

I have two "nucleo_wb55rg" boards attached.

# pyocd list

  #   Probe          Unique ID
-----------------------------------------------
  0   STM32 STLink   0670FF363931594E43053025
  1   STM32 STLink   0670FF3837334D4E43065316

# west flash -r openocd --serial 0670FF363931594E43053025

flashes the board with serial 0670FF3837334D4E43065316. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Build a sample for nucleo_wb55rg board.
2. Attach two nucleo_wb55rg boards.
3. Try to flash both boards using openocd providing the --serial option - one board at a time
4. Only one board gets flashed.

**Expected behaviour**
When I provide the --serial <serial_num> option to west/openocd command I expect only the board with the provided <serial_num> to be flashed.

**Impact**
I have no way of choosing which board to flash when multiple boards are flashing when working remote. This is annoying.

**Logs and console output**

Board #1:
wojci@wifibuildbox:~/zephyrproject/zephyr$ west flash -r openocd --serial 0670FF3837334D4E43065316
-- west flash: rebuilding
[0/1] cd /home/wojci/zephyrproject/zephyr/build/zephyr/cmake/flash && /usr/bin/cmake -E echo

-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/wojci/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-05-16-15:41)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
0670FF3837334D4E43065316
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : clock speed 500 kHz
Info : STLINK V2J36M26 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.259396
Info : stm32wbx.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32wbx.cpu       hla_target little stm32wbx.cpu       running

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08002274 msp: 0x20001f20
Info : device idcode = 0x20016495 (STM32WB5x - Rev: 2.1)
Info : flash size = 1024kbytes
Info : flash mode : single-bank
Info : Padding image section 0 at 0x0800e79c with 4 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x0800e7a0 .. 0x0800efff
auto erase enabled
wrote 59296 bytes from file /home/wojci/zephyrproject/zephyr/build/zephyr/zephyr.hex in 2.378626s (24.344 KiB/s)

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08002274 msp: 0x20001f20
verified 59292 bytes in 1.809028s (32.007 KiB/s)

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
shutdown command invoked

Board #2:

wojci@wifibuildbox:~/zephyrproject/zephyr$ west flash -r openocd --serial 0670FF363931594E43053025
-- west flash: rebuilding
[0/1] cd /home/wojci/zephyrproject/zephyr/build/zephyr/cmake/flash && /usr/bin/cmake -E echo

-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/wojci/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-05-16-15:41)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
0670FF363931594E43053025
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : clock speed 500 kHz
Info : STLINK V2J36M26 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.256242
Info : stm32wbx.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32wbx.cpu       hla_target little stm32wbx.cpu       running

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08002274 msp: 0x20001f20
Info : device idcode = 0x20016495 (STM32WB5x - Rev: 2.1)
Info : flash size = 1024kbytes
Info : flash mode : single-bank
Info : Padding image section 0 at 0x0800e79c with 4 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x0800e7a0 .. 0x0800efff
auto erase enabled
wrote 59296 bytes from file /home/wojci/zephyrproject/zephyr/build/zephyr/zephyr.hex in 2.367292s (24.461 KiB/s)

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08002274 msp: 0x20001f20
verified 59292 bytes in 1.809163s (32.005 KiB/s)

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
shutdown command invoked

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 LTS
 - Toolchain: zephyr-sdk-0.11.3/
 - Commit SHA: 7da5f15e1577755b4d01dc05b8fac86f2da8cba8

**Additional context**
N/A

erwango, 2020-07-09:
So we're generating:
```
/local/mcu/zephyrproject/sdk/zephyr-sdk-0.11.3-beta/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /local/mcu/zephyrproject/zephyr/boards/arm/nucleo_wb55rg/support -s /local/mcu/zephyrproject/sdk/zephyr-sdk-0.11.3-beta/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts '-c set _ZEPHYR_BOARD_SERIAL 066DFF545057717867195937' -f /local/mcu/zephyrproject/zephyr/boards/arm/nucleo_wb55rg/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /local/mcu/zephyrproject/zephyr/samples/hello_world/build/nucleo_wb55rg/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /local/mcu/zephyrproject/zephyr/samples/hello_world/build/nucleo_wb55rg/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

I'm not sure if `set _ZEPHYR_BOARD_SERIAL 066DFF545057717867195937` is expected, but `hla_serial 066DFF545057717867195937` should work..
We should check how to get that.

erwango, 2020-07-10:
So for some reason, the following command doesn't work:
```
 west flash -d build/nucleo_wb55rg/ -r openocd --serial 066EFF3832394D3457191438
```
But this one does:
```
west flash -d build/nucleo_wb55rg/ --r openocd -- --cmd-pre-init 'hla_serial 066DFF545057717867195937'
```

erwango, 2020-07-10:
Reading `--serial` helper: `selects FTDI instance by its serial number, defaults to empty`.
So it looks like `--serial` is specific to some probe manufacturer and this makes this option restricted to ARC devices.
For STM32, the above working command is advised.

@wojciiii can you have a test and close the point if satisfied?

wojciiii, 2020-07-10:
Thank you. This works as expected and solves my current problem.

Perhaps this should be noted somewhere in the documentation. I was clearly mislead by reading the documentation since I assumed that --serial would be enough to identify the device that I wanted to flash. :)

wojciiii, 2020-07-10:
See my previous comment.

closed 2020-07-10 by wojciiii