zephyrproject-rtos/zephyr #26788: cmake build system works wrong with cmake version 3.15.5

IRISZZW, 2020-07-10:
**Describe the bug**
When I update to the newest master branch of zephyr, the build system seems work wrong.
There must be some changes in build system these days, when I using an other branch which hundreds
commits behind the newest master branch, it works fine.

And When I update my cmake to version 3.17.3, and the problem fixed.

as zephyr project required `cmake_minimum_required(VERSION 3.13.1)`, so I think this is a bug.

https://github.com/zephyrproject-rtos/zephyr/blob/a043d48c54726082521f3e335d807dc649cc02f3/cmake/app/boilerplate.cmake#L20-L29

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nsim_em ..
3. See error

**Logs and console output**
```
cmake -GNinja -DBOARD=nsim_em ..
Including boilerplate (Zephyr base): C:/Users/zhiwei/Desktop/module/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/Users/zhiwei/Desktop/module/zephyr/samples/hello_world
-- Zephyr version: 2.3.99 (C:/Users/zhiwei/Desktop/module/zephyr)
-- Found Python3: C:/Python/Python37/python3.exe (found suitable exact version "3.7.1") found components:  Interpreter
-- Board: nsim_em
-- Found west: C:/Python/Python37/Scripts/west.exe (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: cross-compile (C:/arc_gnu/bin/arc-elf32-)
-- Found BOARD.dts: C:/Users/zhiwei/Desktop/module/zephyr/boards/arc/nsim/nsim_em.dts
-- Generated zephyr.dts: C:/Users/zhiwei/Desktop/module/zephyr/samples/hello_world/build.nsim_em/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Users/zhiwei/Desktop/module/zephyr/samples/hello_world/build.nsim_em/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/Users/zhiwei/Desktop/module/zephyr/Kconfig
Loaded configuration 'C:/Users/zhiwei/Desktop/module/zephyr/boards/arc/nsim/nsim_em_defconfig'
Merged configuration 'C:/Users/zhiwei/Desktop/module/zephyr/samples/hello_world/prj.conf'
Configuration saved to 'C:/Users/zhiwei/Desktop/module/zephyr/samples/hello_world/build.nsim_em/zephyr/.config'
Kconfig header saved to 'C:/Users/zhiwei/Desktop/module/zephyr/samples/hello_world/build.nsim_em/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/arc_gnu/bin/arc-elf32-gcc.exe
-- Cache files will be written to: C:\Users\zhiwei\AppData\Local/.cache/zephyr
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/CheckCCompilerFlag.cmake:41 (set):
  Syntax error in cmake code at
    C:/Program Files/CMake/share/cmake-3.15/Modules/CheckCCompilerFlag.cmake:41
  when parsing string
    -fmacro-prefix-map=C:\Users\zhiwei\Desktop\module=WEST_TOPDIR
  Invalid character escape '\U'.
Call Stack (most recent call first):
  ../../cmake/extensions.cmake:1340 (check_c_compiler_flag)
  ../../cmake/extensions.cmake:810 (check_compiler_flag)
  ../../cmake/extensions.cmake:1379 (zephyr_check_compiler_flag)
  ../../cmake/extensions.cmake:1355 (target_cc_option_fallback)
  ../../cmake/extensions.cmake:118 (target_cc_option)
  ../../CMakeLists.txt:343 (zephyr_cc_option)

-- Configuring incomplete, errors occurred!
See also "C:/Users/zhiwei/Desktop/module/zephyr/samples/hello_world/build.nsim_em/CMakeFiles/CMakeOutput.log".
```

**Environment (please complete the following information):**
 - OS: (Windows)
 - Toolchain (cross-compile)

tejlmand, 2020-07-10:
I noticed this:
```
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/CheckCCompilerFlag.cmake:41 (set):
  Syntax error in cmake code at
    C:/Program Files/CMake/share/cmake-3.15/Modules/CheckCCompilerFlag.cmake:41
  when parsing string
    -fmacro-prefix-map=C:\Users\zhiwei\Desktop\module=WEST_TOPDIR
  Invalid character escape '\U'.
```
which is the root cause for your error, and has been fixed here:
https://github.com/zephyrproject-rtos/zephyr/pull/26783

Please fetch latest master.

tejlmand, 2020-07-10:
Closed as already fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/26783

closed 2020-07-10 by tejlmand