zephyrproject-rtos/zephyr #26803: Cortex-M7 Thumb-2 Instructions Alignment

kkiyota, 2020-07-10:
**Describe the bug**
Since I got a bus fault, I analyzed my code by Trace32 debugger.  It seemed that the bus fault occurred at the code like  "cmp r6,#0x40000" which is 32 bits instruction.  This one is allocated at address 0x622A where is not a word aligned.   

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=myownboard
3. ninja
4. run by trace32

**Expected behavior**
No bus fault occurred.  32 bits instructions should be word aligned.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain (gcc)
 - Zephyr 2.2

**Additional context**
Add any other context about the problem here.

kkiyota, 2020-07-14:
I found the assembler directive, ".align 2" to make the word alignment.  Although the first Thumb-2 instruction, "cmp r6, #262144", was word aligned by addition NOP at 0x542e as shown below.   However, second one at 0x5436 was not word aligned.  Since I have 3 cases, gcc seems to generate 3 cmp instructions at switch statement.  As a result, I can't put other ".align 2". 

[image]

As next step, I rewrite this portion to IF-THEN statements.  There is only one "cmp" instruction which is the word aligned properly at 0x5430 as shown below.

I couldn't run it on my hardware today.  I'll let you know when I can run it on my hardware.

I also tried several different gcc options thru CONFIG_*, but there was no solution yet.  I'm not sure if this is gcc limitation or whatever.  Any idea? 

[image]

stephanosio, 2020-07-15:
All Thumb instructions (including the 32-bit Thumb-2 instructions) are to supposed be half-word (i.e. 2-byte) aligned.

The bus fault must be due to something else.

kkiyota, 2020-07-15:
My CPU is ARMv7-M (Coretex-M7).  According to ARM document, some unaligned instructions seem to be supported, but my case is cmp.w.  I guess it may be out of support.

[image]

stephanosio, 2020-07-15:
The portion you quoted above only apply to the data accesses. See the section right below that:
[image]

ioannisg, 2020-07-15:
> All Thumb instructions (including the 32-bit Thumb-2 instructions) are supposed to be half-word (i.e. 2-byte) aligned.
> 
> The bus fault must be due to something else.

@stephanosio you mean, they are _required_ to be half-word aligned?

ioannisg, 2020-07-15:
@kkiyota could you post some more information on the fault you're getting, as @stephanosio suggested above?

kkiyota, 2020-07-15:
@stephanosio @ioannisg Sure I will do it later.  Unfortunately, I ca't access to my system due to a network problem.

kkiyota, 2020-07-17:
I ran step execution and change my program to figure out root causes.  Looks like it may be related to interrupt related rather than Thumb-2 instructions' alignment so far.  I'll let you know again when I figure it out.

kkiyota, 2020-08-06:
I found that my device setting had a typo.  As a result, my ISR structure made an incorrect address.  

Bus fault was gone after I fixed it.

Appreciated your help!

closed 2020-08-06 by kkiyota