zephyrproject-rtos/zephyr #26804: Bluetooth mesh repeated provision/gatt bearer connection crash

johnsmithecon, 2020-07-10:
**Describe the bug**
A clear and concise description of what the bug is.
If I use nrf mesh provision app and I provision then reset then provision then reset and continue for about 5 times in a row, the node will become unable to get composition data and I notice that i can’t connect to it as proxy. The debug messages are below:

**To Reproduce**
Steps to reproduce the behavior:
See above

**Expected behavior**
Not crash

**Impact**
Makes the node crash so if this does not get fixed, product can never be used in production setting.

**Logs and console output**
00> [00:01:35.087,677] <dbg> bt_mesh_proxy.proxy_connected: conn 0x20001478 err 0x00
00> [00:01:35.676,361] <dbg> bt_mesh_proxy.proxy_ccc_write: value: 0x0001
00> [00:01:35.676,483] <dbg> bt_mesh_proxy.proxy_ccc_changed: value 0x0001
00> [00:01:35.676,605] <dbg> bt_mesh_beacon.bt_mesh_beacon_create: net_idx 0x0000 flags 0x00 NetID 577fdc7551306fa9
00> [00:01:35.676,605] <dbg> bt_mesh_beacon.bt_mesh_beacon_create: IV Index 0x00000000 Auth 5d8268649d44f205
00> [00:01:35.676,635] <dbg> bt_mesh_proxy.proxy_segment_and_send: conn 0x20001478 type 0x01 len 22: 0100577fdc7551306fa9000000005d8~
00> [00:01:35.676,666] <dbg> bt_mesh_proxy.proxy_send: 23 bytes: 010100577fdc7551306fa9000000005~
00> [00:01:35.676,696] <err> bt_conn: Unable to allocate TX context
00> [00:01:35.736,236] <err> os: ***** BUS FAULT *****
00> [00:01:35.736,236] <err> os:   Imprecise data bus error
00> [00:01:35.736,236] <err> os: r0/a1:  0xfffffffc  r1/a2:  0x00000022  r2/a3:  0x0000001e
00> [00:01:35.736,236] <err> os: r3/a4:  0x20006cd8 r12/ip:  0x00000000 r14/lr:  0x000282cb
00> [00:01:35.736,267] <err> os:  xpsr:  0x21000000
00> [00:01:35.736,267] <err> os: Faulting instruction address (r15/pc): 0x000282d0
00> [00:01:35.736,267] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
00> [00:01:35.736,267] <err> os: Current thread: 0x20001b18 (unknown)
00> [00:01:36.194,793] <err> os: Halting system

**Environment (please complete the following information):**
 - OS: Mac

carlescufi, 2020-07-10:
>00> [00:01:35.676,696] bt_conn: Unable to allocate TX context

This looks familiar, and is probably unrelated to mesh. Which SHA are you using to reproduce this? We recently merged a PR that solved a similar problem (https://github.com/zephyrproject-rtos/zephyr/pull/25954), though this feels like it's a variant of it, not exactly the same symptoms.

johnsmithecon, 2020-07-10:
I am using the newest code from master branch. How do I check what SHA I’m using?

carlescufi, 2020-07-13:
@joerchan @trond-snekvik any ideas on this one?

joerchan, 2020-07-15:
@johnsmithecon Can you check if https://github.com/zephyrproject-rtos/zephyr/pull/26872 fixes your issue?

johnsmithecon, 2020-07-15:
@joerchan @trond-snekvik I can confirm that I am still seeing this issue. #26872 did not fix this.

joerchan, 2020-07-16:
@johnsmithecon Thank you for checking.

carlescufi, 2020-07-16:
I think we should start by checking the code in `bt_mesh_proxy.proxy_send()` when sending the GATT packet actually fails.

johnsmithecon, 2020-07-17:
My proj.conf file:

# J-Link Log Viewer
#CONFIG_UART_CONSOLE=n
#CONFIG_HAS_SEGGER_RTT=y
#CONFIG_USE_SEGGER_RTT=y
#CONFIG_RTT_CONSOLE=y
#CONFIG_LOG_BACKEND_RTT=y
#CONFIG_LOG_STRDUP_BUF_COUNT=20
#CONFIG_LOG_STRDUP_MAX_STRING=80

#CONFIG_MAIN_STACK_SIZE=512
#CONFIG_BT_RX_STACK_SIZE=2048
#CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

#CONFIG_BT=y
#CONFIG_BT_OBSERVER=y
#CONFIG_BT_BROADCASTER=y
#CONFIG_BT_DEBUG_LOG=n
#CONFIG_BT_MESH_GATT_PROXY=y
#CONFIG_BT_PERIPHERAL=y
#CONFIG_BT_MESH_PB_GATT=y
#CONFIG_BT_MESH_PB_ADV=y
#CONFIG_BT_DEVICE_NAME="LIGHT"

#CONFIG_BT_MESH=y
#CONFIG_BT_MESH_APP_KEY_COUNT=1

#CONFIG_BT_SETTINGS=y
#CONFIG_FLASH=y
#CONFIG_FLASH_PAGE_LAYOUT=y
#CONFIG_FLASH_MAP=y
#CONFIG_FCB=y
#CONFIG_SETTINGS=y
#CONFIG_SETTINGS_FCB=y

#CONFIG_BT_DEBUG_LOG=y
#CONFIG_BT_MESH_DEBUG=y
#CONFIG_BT_MESH_DEBUG_ACCESS=y
#CONFIG_BT_MESH_DEBUG_MODEL=y
#CONFIG_BT_MESH_DEBUG_TRANS=y
#CONFIG_BT_MESH_DEBUG_NET=y
#CONFIG_BT_MESH_DEBUG_CRYPTO=n

# PWM START
#CONFIG_PRINTK=y
CONFIG_PWM=y
#CONFIG_PWM_LOG_LEVEL_DBG=y
# PWM END

#CONFIG_SYS_CLOCK_TICKS_PER_SEC=1000

#BT_CTLR=BT_CTLR_TX_PWR_PLUS_8

# from prj.conf of mesh
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

CONFIG_BT_SETTINGS=y
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_NVS=y
CONFIG_SETTINGS=y

CONFIG_BT_OBSERVER=y
CONFIG_BT_CTLR_DUP_FILTER_LEN=0
CONFIG_BT_CTLR_LE_ENC=n
CONFIG_BT_DATA_LEN_UPDATE=n
CONFIG_BT_PHY_UPDATE=n
CONFIG_BT_CTLR_CHAN_SEL_2=n
CONFIG_BT_CTLR_MIN_USED_CHAN=n
CONFIG_BT_CTLR_PRIVACY=n

CONFIG_BT_PERIPHERAL=y

CONFIG_BT=y
CONFIG_BT_TINYCRYPT_ECC=y
CONFIG_BT_L2CAP_RX_MTU=69
CONFIG_BT_L2CAP_TX_MTU=69
CONFIG_BT_L2CAP_TX_BUF_COUNT=5

CONFIG_BT_MESH=y
CONFIG_BT_MESH_MODEL_EXTENSIONS=y
CONFIG_BT_MESH_RELAY=y
CONFIG_BT_MESH_FRIEND=y
CONFIG_BT_MESH_PB_GATT=y
CONFIG_BT_MESH_PB_ADV=y
CONFIG_BT_MESH_GATT_PROXY=y

#CONFIG_BT_MESH_LOW_POWER=y
#CONFIG_BT_MESH_LPN_SCAN_LATENCY=30
#CONFIG_BT_MESH_LPN_RECV_DELAY=40
#CONFIG_BT_MESH_LPN_POLL_TIMEOUT=300

CONFIG_BT_MESH_SUBNET_COUNT=2
CONFIG_BT_MESH_APP_KEY_COUNT=2
CONFIG_BT_MESH_MODEL_GROUP_COUNT=2
CONFIG_BT_MESH_LABEL_COUNT=3

CONFIG_BT_DEBUG_LOG=y

CONFIG_BT_MESH_DEBUG=y
#CONFIG_BT_MESH_DEBUG_PROV=y
CONFIG_BT_MESH_DEBUG_PROXY=y
#CONFIG_BT_MESH_DEBUG_BEACON=y
#CONFIG_BT_MESH_DEBUG_NET=y
#CONFIG_BT_MESH_DEBUG_TRANS=y
#CONFIG_BT_MESH_DEBUG_SETTINGS=y
#CONFIG_BT_MESH_DEBUG_LOW_POWER=y
#CONFIG_BT_MESH_DEBUG_FRIEND=y
#CONFIG_BT_MESH_DEBUG_MODEL=y
#CONFIG_BT_MESH_DEBUG_ACCESS=y
#CONFIG_BT_MESH_DEBUG_CRYPTO=y
#CONFIG_BT_MESH_DEBUG_ADV=y

# UART STUFF FOR SENSOR
CONFIG_UART_CONSOLE=n
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_LOG_BACKEND_SWO=n
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_RING_BUFFER=y

#CONFIG_BT_MAX_PAIRED=1

johnsmithecon, 2020-07-17:
I reset node 1 and try to provision the sensor node and it gets stuck at trying to get composition data and my logs for node 2 tells me no space to store  ccc cfg. And literally node 2 is broken now so I need to turn off and back on and now I can get composition data. And now I can’t reset node 2 with my app. But if I reset the node 1 first then try to reset the node 2, I am able to reset.

this is in the context of bluetooth mesh

carlescufi, 2020-09-10:
@johnsmithecon is this still an issue? Could you please try with the latest master and close the issue if it's not longer reproducible?

carlescufi, 2020-09-17:
Closing this, @johnsmithecon please reopen this if it's still an issue.

closed 2020-09-17 by carlescufi