zephyrproject-rtos/zephyr #26805: test: drivers: i2c: i2c_slave_api:

ABOSTM, 2020-07-10:
**Describe the bug**
We introduced new board to our test bench: nucleo_f091rc 
And we get status failed on test: tests/drivers/i2c/i2c_slave_api/

Looking at original commit introducing this test:
(sha1: 94d3156540d27392236044eebcb4c3a6e6065185)

>     test: i2c: Add i2c_slave_api test
>     
>     Add a new test to make I2C transfers between an I2C controller and
>     an I2C slave driver, either by:
>     - Using a virtual I2C driver
>     - Using a physical link and 2 physical controllers
>     
>     The Physical test uses :
>     - Two EEPROM Slave drivers each connected to one I2C controller
>     - Both I2C controllers also act as master and take turns writing
>     to the opposite EEPROM Slave.
>     All of this connected physically on the board.

First issue : specifically for nucleo_f091rc and also for stm32f072b_disco, virtual I2C is disabled
`CONFIG_I2C_VIRTUAL=n`
Thus Physical EEPROM should be required, whereas no eeprom is connected to the board.

I could either:
* exclude both board from this test, but the whitelist in yaml contains only those boards !! Thus test would become useless.
  This is probably a 2nd issue, I don't understand why only those 2 boards are whitelisted. 
  Other board (not only stm32) could be tested either with physical eeprom or with virtual I2C.
Note: if external hardware is required maybe a 'harness' or 'fixture' should be added

* use virtual I2C for those boards.
Unfortunately it doesn't work.
I don't think issue is specific to STM32, but simply not tested on other boards. 
Possibly this test has not been run with virtual i2C for ages ...
@henrikbrixandersen it seem you use this test ?

**To Reproduce**
Steps to reproduce the behavior:
1. either remove `CONFIG_I2C_VIRTUAL=n` from file nucleo_f091rc.conf, or not
2. west build -p auto -b nucleo_f091rc tests/drivers/i2c/i2c_slave_api/
3. west flash
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs and console output**
*** Booting Zephyr OS build zephyr-v2.3.0-980-gb08309e9924c  ***
```
Running test suite test_eeprom_slave
===================================================================
START - test_eeprom_slave
I: Found I2C Master device I2C_1
I: Found I2C Master device I2C_2
I: Found EEPROM device EEPROM_SLAVE_0
I: Found EEPROM device EEPROM_SLAVE_1
D: i2c: slave registered
I: EEPROM EEPROM_SLAVE_0 Attached !
D: i2c: slave registered
I: EEPROM EEPROM_SLAVE_1 Attached !
I: Start full read. Master: I2C_1, address: 0x56
D: stm32_i2c_msg_write: NACK
```

**Environment (please complete the following information):**
 - OS:  linux
 - Toolchain Zephyr SDK
 - Commit SHA: dc43c298a2

henrikbrixandersen, 2020-07-17:
@ABOSTM Sorry, no I have never used this test not the I2C slave functionality. I may have fixed a few things in the test case when introducing I2C EEPROM support.

pabigot, 2020-07-21:
> I don't understand why only those 2 boards are whitelisted.

I suspect because they were the ones originally used to develop the I2C follower interface in Zephyr.  The test pre-dates the `testcast.yaml` infrastructure and has never reflected the need to cross-connect I2C lines for two controllers to verify behavior (suggested in the original commit messages).

#26003 will support running this test on native posix with the existing simulator.  It may be best to simply remove the whitelist for the current test programs and replace them with native_posix.

Alternatively the test specification would have to be updated to note the need to wire two controllers together, then verify that it actually works.

There's [evidence](https://github.com/zephyrproject-rtos/zephyr/pull/26003#issuecomment-661968444) in the discussion of #26003 that the [driver used to simulate](https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/i2c/slave) follower behavior has bugs.  That should also be investigated.

PR #27387 (2020-08-13): cleanup of i2c_slave driver and test implementation and documentation
Closes #27303 by moving the simulated EEPROM I2C follower device into the test that uses it.

Closes #26805 by documenting how the test is intended to be used.

See individual commit messages for more details.

This commit includes material and concepts that derive from changes originally proposed by @sjg20 in  #26003.
closed 2020-08-13 by carlescufi