zephyrproject-rtos/zephyr #26807: Bluetooth HCI USB sample is not working

brooksprumo, 2020-07-10:
**Describe the bug**
I'm trying to get the Bluetooth HCI USB sample working. I have Zephyr/NCS version v2.3.0-rc1-ncs1. I am running Ubuntu Linux in a VM on Windows 10. I am using an NRF 52840 DK board.

After building the sample and flashing the board, my Windows machine sees there's a new Bluetooth device, but something is wrong and Windows does not load the driver for it.

[image]

Since Windows does not load the driver, my Linux VM does not see the Bluetooth device either. I wonder if there's something I didn't configure properly thought?

I *am* able to use the Bluetooth HCI UART sample and have the nrf52840 successfully show up as a Bluetooth device in my Linux VM. This won't work long-term for me because I'm only using the nrf52840 DK for prototyping, and my final product does not have a UART connection back to the Host, just a USB connection.

**To Reproduce**
1. Get Zephyr/NCS version v2.3.0-rc1-ncs1
2. Navigate to `zephyr/samples/bluetooth/hci_usb`
3. Build with `west build -b nrf52840dk_nrf52840`
4. Flash the board with `west flash`

**Expected behavior**
I expect that my Windows machine enumerates a new Bluetooth device, and then my Linux VM also enumerates a new Bluetooth device. After that, I would expect to use `hciconfig` to see the Bluetooth device and use `bluetoothctl` to interact with it.

**Impact**
This may be a showstopper for us, as that's why we're using an nrf52840 and Zephyr, just for Bluetooth HCI over USB.

**Logs and console output**
I'm not sure what else is useful to add. Should additional information be needed, please let me know and I'll happily add it!

**Environment (please complete the following information):**
 - OS: Windows host, Linux VM
 - Toolchain: Zephyr/NCS version v2.3.0-rc1-ncs1, https://github.com/nrfconnect/sdk-zephyr
 - Commit SHA or Version used: `bbd71e23a2d56f05cefc9dca7bef609c22c1966c`

**Additional context**
I was able to use Apache Mynewt to do Bluetooth HCI USB with the same machine and board using this example: 

https://github.com/hathach/mynewt-tinyusb-example

carlescufi, 2020-07-10:
This should not be a problem, and in fact I have used a very similar setup in the past with no issues.

>Since Windows does not load the driver, my Linux VM does not see the Bluetooth device either. I wonder if there's something I didn't configure properly thought?

At least with VMWare, you don't need Windows to recognize your device in order for the VM to be able to use it, since it redirects the whole USB device.
I don't think Windows supports single-mode, BLE-only controllers like Zephyr implements, so I am not surprised that Windows doesn't find drivers for it.

That said, the fact that the Mynewt example you linked works with the same setup makes me think that something in our USB configuration can be improved.

Could you please dump the contents of `lsusb` from the VM, specifically the full set of interfaces and descriptors for the Mynewt USB device? then perhaps @emob-nordic can take a look at ours and see if there's any differences.

brooksprumo, 2020-07-10:
I agree, sounds strange! Below are dumps from `lsusb --verbose` from Zephyr and Mynewt. They are long...

Also, I just saw the EDIT too. I'll go back and check if there's anything strange about my Zephyr VM. I'll try switching the USB to USB 3 and see if that changes anything. I'll reply back to let you know.

## Output from Zephyr

```
$ lsusb --verbose

Bus 001 Device 007: ID 1366:1015 SEGGER 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1366 SEGGER
  idProduct          0x1015 
  bcdDevice            1.00
  iManufacturer           1 SEGGER
  iProduct                2 J-Link
  iSerial                 3 000683002044
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          121
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 Configuration
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0 None
      iFunction               5 CDC
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              5 CDC
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x03
          call management
          use DataInterface
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
[... 801 lines cut ...]
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval             255
```

carlescufi, 2020-07-10:
@brooksprumo can you edit your [message](https://github.com/zephyrproject-rtos/zephyr/issues/26807#issuecomment-656903647) and use the `-d vid:pid` option in `lsusb` like I posted in mine to restrict the dump to the device in question?

carlescufi, 2020-07-10:
Hmm I think that Mynewt sets the following fields in the Device Descriptor:

```
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
```

whereas Zephyr leaves them at 0s:

```
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
```

brooksprumo, 2020-07-10:
> @brooksprumo can you edit your [message](https://github.com/zephyrproject-rtos/zephyr/issues/26807#issuecomment-656903647) and use the `-d vid:pid` option in `lsusb` like I posted in mine to restrict the dump to the device in question?

Done! Let me know if there's more info that would be helpful.

carlescufi, 2020-07-10:
> > @brooksprumo can you edit your [message](https://github.com/zephyrproject-rtos/zephyr/issues/26807#issuecomment-656903647) and use the `-d vid:pid` option in `lsusb` like I posted in mine to restrict the dump to the device in question?
> 
> Done! Let me know if there's more info that would be helpful.

Does the mynewt controller actually work with Windows? if you try say searching for devices

brooksprumo, 2020-07-10:
> Does the mynewt controller actually work with Windows? if you try say searching for devices

I do not know how to check 🤕

But, it does look like Windows has the same issue with the Mynewt firmware as it does with Zephyr. So, I opened a different VM, still with my board running the Zephyr firmware, and the Bluetooth device was present!

```
lsusb -vd 2fe3:000b

Bus 003 Device 023: ID 2fe3:000b  
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2fe3 
  idProduct          0x000b 
  bcdDevice            2.03
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
```

I was also able to successfully scan via `bluetoothctl`. So I think this means it's something about my VM. Sorry for the run-around on this. Thank you for helping me!

carlescufi, 2020-07-10:
Ok, glad to hear it! I will close this then, feel free to reopen if applicable.

closed 2020-07-10 by carlescufi

jfischer-phytec-iot, 2020-07-10:
> Hmm I think that Mynewt sets the following fields in the Device Descriptor:
> 
> ```
>   bDeviceClass          224 Wireless
>   bDeviceSubClass         1 Radio Frequency
>   bDeviceProtocol         1 Bluetooth
> ```
> 
> whereas Zephyr leaves them at 0s:
> 
> ```
>   bDeviceClass            0 

"If this field is reset to zero, each interface
within a configuration specifies its own
class information and the various
interfaces operate independently." (USB2.0 spec)

>   bDeviceSubClass         0 

"If the bDeviceClass field is reset to zero,
this field must also be reset to zero." (USB2.0 spec)

>   bDeviceProtocol         0 

"If this field is reset to zero, the device
does not use class-specific protocols on a
device basis. However, it may use class-
specific protocols on an interface basis." (USB2.0 spec)

> ```
> 
> And they seem to have 2 interfaces instead of one like Zephyr, wonder why.

Zephyr:
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth

mirko, 2020-10-11:
I'm having the same issue when passing through the ZEPHYR dongle (nrf52840 USB dongle) into my qemu/KVM VM.
lsusb shows device with empty identifier whereas on the hypervisor and my notebook i can use it just fine.