zephyrproject-rtos/zephyr #26812: NXP: tests/drivers/dma/loop_transfer fails on FRDM-K64F

henrikbrixandersen, 2020-07-11:
**Describe the bug**

The `tests/drivers/dma/loop_transfer` test case fails on the NXP FRDM-K64F. Not all DMA RX buffers contain the DMA TX buffer string.

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/drivers/dma/loop_transfer; mkdir build; cd build
2. cmake -DBOARD=frdm_k64f
3. make flash
4. See the following output on the board console:

```
*** Booting Zephyr OS build zephyr-v2.3.0-898-g668e359d8558  ***
DMA memory to memory transfer started on DMA_0
Preparing DMA Controller
Starting the transfer and waiting for 1 second
TX data: The quick brown fox jumps over the lazy dog ....
block_size 48
I: channel 0 error status is 0x2
ERROR: unfinished transfer
Each RX buffer should contain the full TX buffer string.
TX data: The quick brown fox jumps over the lazy dog ....
RX data Loop 0: The quick brown fox jumps over the lazy dog ....
RX data Loop 1: The quick brown fox jups
RX data Loop 2: 
RX data Loop 3: 
RX data Loop 4: 
Finished: DMA
```
**Expected behavior**
All RX buffers contain the TX buffer content.

**Impact**
Not sure if the EDMA driver works correctly. The `tests/drivers/dma/chan_blen_transfer` test case passes, so something is working...

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA: 668e359d85585e18df5678199cae8f5be25a8b9a

**Additional context**
I found the issue while enabling the eDMA driver on the NXP KE1xF SoC.

hakehuang, 2020-07-14:
@henrikbrixandersen this is a racing condtion which cause the dma access flash to tcm  which will cause error. and it is not happen always, but with high possibilities. so the solution is to change the test cases as below:
```
#if CONFIG_NOCACHE_MEMORY
- static const char TX_DATA[] = "The quick brown fox jumps over the lazy dog";
+ static char TX_DATA[] = "The quick brown fox jumps over the lazy dog";
static __aligned(16) char tx_data[64] __used  
```

henrikbrixandersen, 2020-07-19:
@hakehuang Thank you. I can confirm that your commit mentioned above fixes the race condition.

closed 2020-07-20 by MaureenHelm