zephyrproject-rtos/zephyr #26818: drivers: uart_console.c: usb_enable() broken

sslupsky, 2020-07-11:
This commit 6da7e4159216faf835345480ca92b3c33e29d932 broke usb_enable().

usb_enable() takes a user_status_callback parameter.  This no longer works properly when uart_console_init() is called because this function now calls usb_enable() with a null pointer.  When the application calls usb_enable(), usb_enable() immediately returns because it has already been enabled.

sslupsky, 2020-07-11:
@jfischer-phytec-iot I believe you made the commit.  You may have overlooked this issue?

jfischer-phytec-iot, 2020-07-14:
The commit 6da7e41 fixed a bug and described it in the commit message.
The design does not provide a way to manually initialize the USB stack when using the console.

closed 2020-07-14 by jfischer-phytec-iot

sslupsky, 2020-07-18:
@jfischer-phytec-iot The changes in 2.2 **require** the application to Initialize the usb stack.

You have introduced **breaking change** to a published API, please do not close this issue.

@carlescufi @pabigot

jfischer-phytec-iot, 2020-07-19:
> @jfischer-phytec-iot The changes in 2.2 **require** the application to Initialize the usb stack.

Yes, the console driver is the application in this case. And again the commit fixed the issue regarding "USB_UART_DTR_WAIT".

> 
> You have introduced a **breaking change** to a published API, please do not close this issue.

I did not.

pabigot, 2020-07-19:
I don't agree that the console driver is itself an application.  From the 2.2 release notes:
```
  * The usb_enable() function, which was previously invoked automatically
    by the USB stack, now needs to be explicitly called by the application
    in order to enable the USB subsystem.
  * The usb_enable() function now takes a parameter, usb_dc_status_callback
    which can be set by the application to a callback to receive status events
    from the USB stack. The parameter can also be set to NULL if no callback is required.
```

I have a use case where I want a console on USB *when USB is connected*, but I also want to be able to make use of the status callbacks.  Which means the console becomes a device that can come and go.

I don't know exactly how that can be supported in Zephyr, but having the console driver take over the USB subsystem before the application using it can take advantage of a USB feature is a problem.  So is having the application block during initialization because the console isn't available.

I agree the referenced commit did not break the API, but it did expose an existing behavior that can be said to break the promise of how the USB subsystem works from an application's perspective.  We need to come up with a way to support both use cases.

sslupsky, 2020-07-21:
@pabigot The use case you describe is exactly what I am doing ... use the console on USB when USB is connected ... and I want to minimize power consumption when USB is disconnected.

Could this issue be resolved by adding a Kconfig option for "early usb console start" or maybe using the existing "early console start" option?

sslupsky, 2020-07-21:
@carlescufi Could you please  re-open this issue?

jfischer-phytec-iot, 2020-07-21:
This bug report is not true and will stay closed. You have to open an enhancement request if you need.

carlescufi, 2020-07-22:
@sslupsky could you please open a new Enhancement describing the functionality that you would like implemented?

sslupsky, 2020-07-22:
@carlescufi It is clearly stated here in this report what the broken functionality is.  The consequence of the referenced commit is that when the application calls usb_enable(), it fails to set the user callback because USB has already been enabled.  This capability was implemented as part of the changes to the USB API in version 2.2.  Specifically, the release notes describe the USB API on this matter as follows:
```
* USB

  * The usb_enable() function, which was previously invoked automatically
    by the USB stack, now needs to be explicitly called by the application
    in order to enable the USB subsystem.
  * The usb_enable() function now takes a parameter, usb_dc_status_callback
    which can be set by the application to a callback to receive status events
    from the USB stack. The parameter can also be set to NULL if no callback is required.
```
Clearly, the API states USB is enabled **explicitly** by the application.  The referenced commit broke that requirement by enabling USB before the application.  Moreover, the referenced commit has broken a number of existing applications in the tree including:
samples/subsys/usb/cdc_acm_composite
samples/subsys/usb/hid
samples/subsys/usb/dfu
samples/subsys/usb/hid-mouse

Consequently, the statement "this bug report is not true" is clearly false.  This report is in fact not requesting an enhancement, this report documents what is clearly a bug that arose as a result of the referenced commit.

pabigot, 2020-07-22:
I've opened #27071 to explore this, though I'm still working my way through the consequences.  I don't believe any of the above listed examples break due to this change, though I suspect they would if `CONFIG_USB_UART_CONSOLE=y` were enabled alongside; I'll be testing that as part of my follow-up.

sslupsky, 2020-07-22:
@pabigot Yeah, that CONFIG flag would need to be set to trigger the issue.

carlescufi, 2020-07-23:
@sslupsky I think you are mostly right in that the enable was documented as having to be called by the application, and that is not quite the case if you enable certain classes. That said, this is a wider problem than just the console, so thanks @pabigot for opening #27071. I have marked that as a USB bug now, and we can continue the discussion there.