zephyrproject-rtos/zephyr #26819: drivers: modem: SARA modem driver leaks sockets

aport, 2020-07-11:
**Describe the bug**
When calling `close` on a socket, the ublox sara driver only issues the `+USOCL` command if the socket is UDP, or if it's TCP *and* connected.

I am porting existing POSIX software to Zephyr. This code calls `close()` on a socket descriptor if `connect()` fails. The result on Zephyr is that the sockets allocated internally by the module are never released. On Linux the file descriptor will be released.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the SARA-R4 driver
2. Create a TCP socket with `socket(AF_INET, SOCK_STREAM, IPPROTO_TCP)`
3. Note the ID of the socket (modem logging helps)
4. Call `close` on the socket
5. See that the socket is not released, no `+USOCL` is issued to the module
6. Repeat step 2
7. See that a new, different socket is allocated by the module. There are now two existing sockets inside the SARA.

**Expected behavior**
Calling `close` on a socket should release its resources, independent on whether it is connected or not.

**Impact**
Eventually the module runs out of internal sockets and new `socket` calls fail. The R4 only supports 7 sockets.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain gnuarmemb
 -  a18a73c19b (master as of the creation of this ticket)

carlescufi, 2020-09-08:
@aport did you work around this or find a fix? if so a Pull Request would be appreciated since there's been no activity on this issue.
@mike-scott any chance you'll be able to look at this?

mike-scott, 2020-09-08:
@carlescufi I have not looked at this.  @aport a PR would be greatly appreciated.

aport, 2020-09-09:
Hi, sorry I have not set up a pull request. My tree is a bit out of date and I've been sidetracked by other projects.

My change was minimal, simply the removal of the if statement at https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/ublox-sara-r4.c#L1219

When I get some time I'll try to do a PR for this

PR #28972 (2021-01-07): modem: fix socket descriptor leak
`z_free_fd()` is called twice then you `close()`. For example in ublox sara r4 driver `offload_close()` calls  `modem_socket_put()` where `z_free_fd()` is called first time. Then this same function is called another time in socket.c in `z_impl_zsock_close()` after this function has called `offload_close()` in ublox sara r4 driver. This causes socket descriptor leak. 

Fixes #26819
commit 3bcb1efc (2021-01-07): modem: fix socket descriptor leak

z_free_fd() is called twice then you close(). For example in ublox sara
r4 driver offload_close() calls modem_socket_put() where z_free_fd() is
called first time. Then this same function is called another time in
socket.c in z_impl_zsock_close() after this function has called
offload_close() in ublox sara r4 driver. This causes socket
descriptor leak.

Fixes #26819
closed 2020-11-23 by github-actions[bot]