zephyrproject-rtos/zephyr #26826: i2c_nrfx_twi_transfer hangs when SDA/SCL are set to pins 0,1

tetious, 2020-07-12:
**Describe the bug**
When using a custom board, `i2c_nrfx_twi_transfer` hangs when SDA/SCL are set to pins 0,1. 
(seems like a few others too, but high pins, like the DK defaults of 26,27 do not hang)

The hang happens here, and it seems like the peripheral just never responds. 
`k_sem_take(&(get_dev_data(dev)->completion_sync), K_FOREVER);`

I've confirmed the following: 
1. `PSELSCL` and `PSELSDA` are set correctly in `twi_control_block` when `nrfx_twi_enable` is called.
2. This occurs with nordic's custom "fork" in NRF Connect SDK as well as Zephyr 2.3.0.
3. The peripheral works perfectly when using either Arduino cores. (Adafruit's or the older one from sandeepmistry)
4. When set to pins 26/27, the call returns an error as expected. (as on my board, those pins are not connected)
5. When set to pins 26/27, I see normal i2c activity on the oscilloscope. 
6. When set to pins 0/1, I see **no activity** on the oscilloscope.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a board with the i2c0 stanza like this:
```
&i2c0 {
	compatible = "nordic,nrf-twi";
	status = "okay";
	sda-pin = <0>;
	scl-pin = <1>;
};
```
2. Attempt to get an ACK from a device. (doesn't matter if it exists or not)
```
    uint8_t buf[] = {42};
    int ret = i2c_write(dev, buf, sizeof(buf), 0x1c);
```
3. Hang occurs as described above.

**Expected behavior**
i2c_write should return.

**Impact**
This prevents me using Zephyr for my current project.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Platform IO for "vanilla 2.3.0", 9-2019-q4-major for nrf connect

**Additional context**
I'm pretty new to working with Zephyr, but I've done LOTS of troubleshooting on this so far, and I likely forgot a detail or two, so please let me know how I can help with trying to figure this out.

anangl, 2020-07-14:
@tetious I assume that on your custom board you don't have the 32.768 kHz crystal (LFXO), which would be connected to pins 0/1. But maybe you've got the `CLOCK_CONTROL_NRF_K32SRC_XTAL` option enabled in your application configuration (it is so by default)? Try adding `CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y` in your application's `prj.conf` or your board's `*_defconfig` file.

tetious, 2020-07-14:
@anangl Thanks for this suggestion! I did indeed have that set wrong, and while it didn't solve the problem, I'm sure it would have caused confusion later on. :)

While puttering around after fixing this, I discovered another point:
The hang only occurs with pins 0-3. While I didn't check ALL the pins, I tried many others and they error as expected. If I use pins 0-3 for either SDA or SCL, the hang occurs.

Maybe interesting?

anangl, 2020-07-14:
@tetious Which SoC are you using?

tetious, 2020-07-14:
I'm using the nRF52832.

tetious, 2020-07-14:
More specifically, I'm using Raytac's MDBT42Q-512KV2 module.

anangl, 2020-07-14:
> While puttering around after changing that config, I discovered another point:
> The hang only occurs with pins 0, 1, 2 and 3. While I didn't check ALL the pins, I tried many others and they error as expected. If I use pins 0-3 for either SDA or SCL, the hang occurs.

@tetious I tried with pins 0-3 on the nRF52DK and I saw correct I2C transactions. Obviously, in order to use pins 0 and 1 I had to cut SB1 and SB2 on the board (to disconnect the 32.768 kHz crystal), and solder SB3 and SB4 instead, plus set `CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y` in my configuration. For pins 2 and 3 no modifications were needed.
Therefore, I think there's rather something wrong in your hardware. Such behavior (the TWI peripheral unable to start a transaction) can be observed when SCL or SDA is tied low, and the pull-up cannot set it high. Are you sure it is not the case in your setup?

tetious, 2020-07-14:
@anangl I really appreciate all your efforts!

I have 4.7k pullups on both lines, and everything works fine with the Arduino core. As such, I think my hardware is at least partially working. :) Also, it was always high when I watched the lines on the scope.

However your clue about the pullup made me remember something I read in the code for how the Arduino is setting up the GPIOs before configuring the I2C peripheral.

Now I assumed the GPIO was being set up by Zephyr in the I2C config somewhere, and none of the samples led me to think otherwise, so I wasn't doing any of that. I decided to try configuring the pins manually like this:
```
	gpio_pin_configure(gpio, 0, GPIO_INPUT | GPIO_PULL_UP);
	gpio_pin_configure(gpio, 1, GPIO_INPUT | GPIO_PULL_UP);
```

It stopped hanging!

Here is the weird thing, and I'm clawing what remains of my hair out trying to understand this, but now it works even if I comment those lines out. Even if I revert back exactly to the git commit that was previously failing, wipe the flash, build with --pristine and power-cycle the chip. So. I dunno. Magic? Cosmic rays?

I hate when this happens, as I think there is some combination of things that I'm doing wrong that makes it get into a state where badness occurs, but at this point I can't get it to fail again, so perhaps it is time to call it and move on. :)

anangl, 2020-07-15:
Nevertheless, I'm glad that it works for you as well. And I guess this issue can be closed?

tetious, 2020-07-15:
Yep. And thanks again for your help. I really appreciate it!

closed 2020-07-15 by tetious