zephyrproject-rtos/zephyr #26828: Network communication between QEMU and Host

xicocana, 2020-07-13:
I have a running application (main.c) on zephyr inside QEMU (using qemu_cortex_a53) and I am trying to implement a socket communication with a python app on the host OS (Ubuntu).

To enable user networking on QEMU I have added the following options to my app Kconfig file (prj_qemu_cortex_a53.conf) :

```
CONFIG_NET_QEMU_USER=y
CONFIG_NET_QEMU_USER_EXTRA_ARGS="net=192.168.0.0/24,hostfwd=tcp::8080-:8080"
```

As explained here: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/guides/networking/qemu_user_setup.html?fbclid=IwAR1EWXrqnmPVctOtFEQVrhAY0J6E6R1s_69vsOxIDg5BRuZwAh9HPwaxUUQ#networking-with-user-qemu

Although I have the following build errors:

```
prj_qemu_cortex_a53.conf:4: warning: attempt to assign the value 'y' to the undefined symbol NET_QEMU_USER
prj_qemu_cortex_a53.conf:5: warning: attempt to assign the value 'y' to the undefined symbol CONFIG_NET_QEMU_USER_EXTRA_ARGS
```

Am I adding the options on the wrong file? Are this options deprecated? Is there public available documentation to replace this?
Sorry if this is not the place to ask this.
Thanks.

carlescufi, 2020-07-13:
@xicocana are you using upstream Zephyr or Nordic's fork?

xicocana, 2020-07-13:
@carlescufi I didn't realize there was a difference. I believe I must be using Nordic's fork due to the fact that I followed this start up guide: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/getting_started/index.html

carlescufi, 2020-07-13:
> @carlescufi I didn't realize there was a difference. I believe I must be using Nordic's fork due to the fact that I followed this start up guide: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/getting_started/index.html

Nordic has no ICs with Cortex A53 cores, so using Nordic's fork combined with that core is not recommended. Please try the same using the standard getting started guide from zephyr:
https://docs.zephyrproject.org/latest/getting_started/index.html

xicocana, 2020-07-13:
@carlescufi now that I see that guide I believe I used that one. Is there a way for me to detect which zephyr fork I'm currently using?

xicocana, 2020-07-13:
@carlescufi I tried with the standard zephyr but I have the same exact problem. Is there a way to run the net sample 
```
zephyr/samples/net/sockets/echo_server/
```
with qemu_cortex_a53 ?

stephanosio, 2020-07-14:
There is currently no networking support on `qemu_cortex_a53` (i.e. QEMU virt platform).

carlescufi, 2020-07-17:
> There is currently no networking support on `qemu_cortex_a53` (i.e. QEMU virt platform).

Thanks @stephanosio. Closing this issue, @xicocana please use another QEMU target if you want to use networking.

closed 2020-07-17 by carlescufi